FormerLurker/Octolapse #633: Bug?: Clean install Octolapse refuses to install

Prutsium, 2020-11-27:
Octopi: 18.0
Octoprint: 1.5.0 (RC3)
Octolapse: 0.4.1+u.5a4b744 (Dev version from Fonts solution)

As i get a new printer i decided to setup on my second Pi Octoprint from scratch so generally did:
Flash Octopi 0.18 Image > Updated to Octoprint 1.5.0 (RC3) > Install Octolapse > Error

2020-11-27 11:48:40,977 - octoprint.plugins.pluginmanager - INFO - Installing plugin from https://github.com/FormerLurker/Octolapse/archive/5a4b744dbf6105d426c35eb449076c9c101e9dc2.zip
2020-11-27 11:54:03,421 - octoprint.plugins.pluginmanager - ERROR - Could not install plugin from https://github.com/FormerLurker/Octolapse/archive/5a4b744dbf6105d426c35eb449076c9c101e9dc2.zip
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 717, in _command_install_archive
    returncode, stdout, stderr = self._call_pip(pip_args)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 1379, in _call_pip
    return self._pip_caller.execute(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/pip.py", line 209, in execute
    return self.call(command, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/commandline.py", line 243, in call
    all_stderr += process_stderr(p.stderr.readlines(timeout=0.5))
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/commandline.py", line 239, in process_stderr
    return process_lines(lines, self._log_stderr)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/commandline.py", line 230, in process_lines
    *map(lambda x: to_unicode(x, errors="replace"), lines)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/pip.py", line 511, in _preprocess_lines
    return list(map(self._preprocess, lines))
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/pip.py", line 530, in _preprocess
    return to_unicode(clean_ansi(text))
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/commandline.py", line 57, in clean_ansi
    return _ANSI_REGEX.sub(b"", line.encode("latin1")).decode("latin1")
UnicodeEncodeError: 'latin-1' codec can't encode character '\u2018' in position 75: ordinal not in range(256)
2020-11-27 12:01:52,238 - octoprint.server.heartbeat - INFO - Server heartbeat <3

And the log from Plugin manager:
2020-11-27 11:48:41,495   /home/pi/oprint/bin/python3 -m pip --disable-pip-version-check install file:///tmp/tmpd56kwica/Octolapse-5a4b744dbf6105d426c35eb449076c9c101e9dc2.zip --no-cache-dir
2020-11-27 11:48:43,725 > Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
2020-11-27 11:48:43,726 > Processing /tmp/tmpd56kwica/Octolapse-5a4b744dbf6105d426c35eb449076c9c101e9dc2.zip
2020-11-27 11:48:54,366 > Collecting pillow>=6.2.0<7.0.0
2020-11-27 11:48:54,366 > Downloading https://www.piwheels.org/simple/pillow/Pillow-8.0.1-cp37-cp37m-linux_armv7l.whl (1.3 MB)
2020-11-27 11:48:54,368 > Requirement already satisfied: sarge in /home/pi/oprint/lib/python3.7/site-packages (from Octolapse==0.4.1+u.5a4b744) (0.1.5.post0)
2020-11-27 11:48:54,369 > Requirement already satisfied: six in /home/pi/oprint/lib/python3.7/site-packages (from Octolapse==0.4.1+u.5a4b744) (1.15.0)
2020-11-27 11:48:54,369 > Requirement already satisfied: OctoPrint>1.3.8 in /home/pi/oprint/lib/python3.7/site-packages (from Octolapse==0.4.1+u.5a4b744) (1.5.0rc3)
2020-11-27 11:48:54,369 > Requirement already satisfied: psutil in /home/pi/oprint/lib/python3.7/site-packages (from Octolapse==0.4.1+u.5a4b744) (5.7.3)
2020-11-27 11:48:54,369 > Collecting file_read_backwards
2020-11-27 11:48:54,370 > Downloading file_read_backwards-2.0.0-py2.py3-none-any.whl (9.0 kB)
2020-11-27 11:48:54,370 > Requirement already satisfied: setuptools>=6.0 in /home/pi/oprint/lib/python3.7/site-packages (from Octolapse==0.4.1+u.5a4b744) (50.3.2)
2020-11-27 11:48:54,370 > Collecting awesome-slugify<1.7,>=1.6.5
2020-11-27 11:48:54,370 > Downloading https://www.piwheels.org/simple/awesome-slugify/awesome_slugify-1.6.5-py3-none-any.whl (9.3 kB)
2020-11-27 11:48:54,370 > Requirement already satisfied: markdown<3.2,>=3.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.5a4b744) (3.1.1)
2020-11-27 11:48:54,371 > Requirement already satisfied: cachelib<1,>=0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.5a4b744) (0.1.1)
2020-11-27 11:48:54,371 > Requirement already satisfied: watchdog<1,>=0.10.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.5a4b744) (0.10.4)
2020-11-27 11:48:54,371 > Requirement already satisfied: wrapt<2,>=1.12.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.5a4b744) (1.12.1)
2020-11-27 11:48:54,371 > Requirement already satisfied: pylru<2,>=1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.5a4b744) (1.2.0)
2020-11-27 11:48:54,371 > Requirement already satisfied: requests<3,>=2.23.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.5a4b744) (2.25.0)
2020-11-27 11:48:54,372 > Requirement already satisfied: flask<2,>=1.1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.5a4b744) (1.1.2)
2020-11-27 11:48:54,372 > Requirement already satisfied: zeroconf<0.25,>=0.24; python_version >= "3" in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.5a4b744) (0.24.5)
2020-11-27 11:48:54,372 > Requirement already satisfied: frozendict<2,>=1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.5a4b744) (1.2)
2020-11-27 11:48:54,372 > Requirement already satisfied: tornado==5.1.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.5a4b744) (5.1.1)
2020-11-27 11:48:54,373 > Requirement already satisfied: Jinja2<3,>=2.11.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.5a4b744) (2.11.2)
2020-11-27 11:48:54,373 > Requirement already satisfied: Flask-Babel<2,>=1.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.5a4b744) (1.0.0)
2020-11-27 11:48:54,373 > Requirement already satisfied: future<1,>=0.18.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.5a4b744) (0.18.2)
2020-11-27 11:48:54,373 > Requirement already satisfied: OctoPrint-FirmwareCheck>=2020.09.23 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.5a4b744) (2020.9.23)
2020-11-27 11:48:54,373 > Requirement already satisfied: OctoPrint-FileCheck>=2020.08.07 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.5a4b744) (2020.8.7)
2020-11-27 11:48:54,374 > Requirement already satisfied: itsdangerous<2,>=1.1.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.5a4b744) (1.1.0)
2020-11-27 11:48:54,374 > Requirement already satisfied: emoji<1,>=0.5.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.5a4b744) (0.6.0)
2020-11-27 11:48:54,374 > Requirement already satisfied: feedparser<7,>=6.0.2; python_version >= "3" in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.5a4b744) (6.0.2)
2020-11-27 11:48:54,374 > Requirement already satisfied: rsa==4.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.5a4b744) (4.0)
2020-11-27 11:48:54,375 > Requirement already satisfied: regex!=2018.11.6 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.5a4b744) (2020.11.13)
2020-11-27 11:48:54,375 > Requirement already satisfied: werkzeug<2,>=1.0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.5a4b744) (1.0.1)
2020-11-27 11:48:54,375 > Requirement already satisfied: netaddr<1,>=0.7.19 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.5a4b744) (0.8.0)
2020-11-27 11:48:54,375 > Requirement already satisfied: PyYAML<6,>=5.3.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.5a4b744) (5.3.1)
[... cut ...]
2020-11-27 11:48:54,381 > Requirement already satisfied: webassets>=2.0 in /home/pi/oprint/lib/python3.7/site-packages (from Flask-Assets<3,>=2.0->OctoPrint>1.3.8->Octolapse==0.4.1+u.5a4b744) (2.0)
2020-11-27 11:48:54,381 > Building wheels for collected packages: Octolapse
2020-11-27 11:48:54,381 > Building wheel for Octolapse (setup.py): started
2020-11-27 11:49:58,429 > Building wheel for Octolapse (setup.py): still running...
2020-11-27 11:50:59,476 > Building wheel for Octolapse (setup.py): still running...
2020-11-27 11:51:28,691 > Building wheel for Octolapse (setup.py): finished with status 'error'
2020-11-27 11:51:28,692 > Running setup.py clean for Octolapse

Prutsium, 2020-11-27:
Minor update:
Did the same thing but now on the 64Bit Octopi version and it works.
The Pi is a 3B+ btw

Prutsium, 2020-11-27:
Oh boy ....
[image]

FormerLurker, 2020-11-27:
I just received another similar unicode related bug report for arc welder.  I will be so happy once python 2 is retired.  Sigh..

Regarding the second issue (looks like we have 2 now in this thread), look in your browser console for errors.  That is usually the culprit when the UI wont bind.

Prutsium, 2020-11-27:
> 
> 
> I just received another similar unicode related bug report for arc welder. I will be so happy once python 2 is retired. Sigh..
> 
> Regarding the second issue (looks like we have 2 now in this thread), look in your browser console for errors. That is usually the culprit when the UI wont bind.

Oh yeah there are some errors :)
Could not bind view model <empty string> to target #octolapse_tab : attr@http://192.168.99.4/static/webassets/packed_libs.js?ab739628 line 3 > Function:3:65
update@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:5:2442
p/</<@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:4:380
yd@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:2:18286
zd@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:2:17972
ha@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:2:17808
a.$@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:2:16560
p/<@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:4:366
D@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:1:148545
p@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:3:4001
k@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:3:2747
m@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:3:2629
k@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:3:2805
a.vc@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:4:3232
c/<@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:5:16902
b@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:5:16553
c@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:5:16851
e@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:5:17498
a.dc/<@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:5:18096
yd@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:2:18286
zd@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:2:17972
ha@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:2:17808
a.$@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:2:16560
a.dc@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:5:17983
update@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:6:1527
p/</<@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:4:380
yd@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:2:18286
zd@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:2:17972
ha@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:2:17808
a.$@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:2:16560
p/<@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:4:366
D@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:1:148545
p@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:3:4001
k@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:3:2747
m@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:3:2629
k@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:3:2805
a.vc@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:4:3232
c/<@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:5:16902
b@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:5:16553
c@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:5:16851
e@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:5:17498
a.dc/<@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:5:18096
yd@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:2:18286
zd@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:2:17972
ha@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:2:17808
a.$@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:2:16560
a.dc@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:5:17983
update@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:6:1527
p/</<@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:4:380
yd@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:2:18286
zd@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:2:17972
ha@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:2:17808
a.$@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:2:16560
p/<@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:4:366
D@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:1:148545
p@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:3:4001
k@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:3:2747
m@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:3:2629
k@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:3:2805
m@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:3:2629
k@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:3:2805
m@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:3:2629
k@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:3:2805
[... 34 lines cut ...]
@http://192.168.99.4/static/webassets/packed_core.js?9ab43570:1025:1438
c@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:1:28294
fireWith@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:1:29039
Deferred/</s[t[0]]@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:1:30940
DataUpdater/self._onConnectMessage@http://192.168.99.4/static/webassets/packed_core.js?9ab43570:838:434
[... cut ...]
[14]</</w.prototype._transportMessage@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:1232:16356
[3]</i.prototype.emit@http://192.168.99.4/static/webassets/packed_libs.js?ab739628:1232:4669
packed_core.js:1021:5
GEThttp://192.168.99.4/plugin/octolapse/getSnapshot?file_type=thumbnail&camera_guid=354def78-9eea-409a-ad23-ee966dfff4ba&apikey=&time=1606491951346
[HTTP/1.1 404 Not Found 1088ms]

GEThttp://192.168.99.4/plugin/octolapse/getSnapshot?file_type=snapshot&camera_guid=354def78-9eea-409a-ad23-ee966dfff4ba&apikey=&time=1606491951350
[HTTP/1.1 404 Not Found 1120ms]

Prutsium, 2020-11-27:
I also see various webfont errors btw:
The resource at “http://192.168.99.4/static/vendor/font-awesome-5.15.1/webfonts/fa-brands-400.woff2?v=5.15.1” preloaded with link preload was not used within a few seconds. Make sure all attributes of the preload tag are set correctly. 192.168.99.4

FormerLurker, 2020-11-27:
That looks like a firefox console log.  Can you try the same thing in chrome (it outputs more specific errors usually).  Thanks!

FormerLurker, 2020-11-27:
Hmm. I don't think the font awesome error is related.  I don't load any custom fonts for the UI.  Something strange is going on.  Maybe related to the RC you are using?  I haven't tried octolapse on it yet, just been so busy with other development.

Prutsium, 2020-11-27:
packed_core.js?9ab43570:1021 Could not bind view model  to target #octolapse_tab : ReferenceError: Unable to process binding "template: function(){return 'snapshot-plan-status-template' }"
Message: Unable to process binding "template: function(){return 'snapshot-plan-template' }"
Message: Unable to process binding "attr: function(){return {'data-help-url':help_link,'data-help-title':name} }"
Message: help_link is not defined
    at attr (eval at parseBindingsString (http://192.168.99.4/static/webassets/packed_libs.js?ab739628:3:1741), <anonymous>:3:81)
    at update (http://192.168.99.4/static/webassets/packed_libs.js?ab739628:5:2442)
    at a.$.l (http://192.168.99.4/static/webassets/packed_libs.js?ab739628:4:379)
    at Function.yd (http://192.168.99.4/static/webassets/packed_libs.js?ab739628:2:18286)
    at Function.zd (http://192.168.99.4/static/webassets/packed_libs.js?ab739628:2:17972)
    at Function.ha (http://192.168.99.4/static/webassets/packed_libs.js?ab739628:2:17808)
    at Object.a.o.a.$ (http://192.168.99.4/static/webassets/packed_libs.js?ab739628:2:16560)
    at http://192.168.99.4/static/webassets/packed_libs.js?ab739628:4:366
    at Object.D (http://192.168.99.4/static/webassets/packed_libs.js?ab739628:1:148545)
    at p (http://192.168.99.4/static/webassets/packed_libs.js?ab739628:3:4001)
(anonymous) @ packed_core.js?9ab43570:1021
Pn @ packed_libs.js?ab739628:1
(anonymous) @ packed_libs.js?ab739628:1
(anonymous) @ packed_core.js?9ab43570:1015
Pn @ packed_libs.js?ab739628:1
(anonymous) @ packed_libs.js?ab739628:1
bindViewModels @ packed_core.js?9ab43570:1006
setTimeout (async)
(anonymous) @ packed_core.js?9ab43570:1025
c @ packed_libs.js?ab739628:1
fireWith @ packed_libs.js?ab739628:1
l @ packed_libs.js?ab739628:1
(anonymous) @ packed_libs.js?ab739628:1
load (async)
send @ packed_libs.js?ab739628:1
ajax @ packed_libs.js?ab739628:1
OctoPrintClient.ajax @ packed_client.js?8f609640:7
OctoPrintClient.get @ packed_client.js?8f609640:7
OctoPrintSettingsClient.get @ packed_client.js?8f609640:57
SettingsViewModel.self.requestData @ packed_core.js?9ab43570:245
fetchSettings @ packed_core.js?9ab43570:1025
setTimeout (async)
(anonymous) @ packed_core.js?9ab43570:1025
c @ packed_libs.js?ab739628:1
fireWith @ packed_libs.js?ab739628:1
l @ packed_libs.js?ab739628:1
(anonymous) @ packed_libs.js?ab739628:1
load (async)
send @ packed_libs.js?ab739628:1
ajax @ packed_libs.js?ab739628:1
OctoPrintClient.ajax @ packed_client.js?8f609640:7
OctoPrintClient.ajaxWithData @ packed_client.js?8f609640:7
OctoPrintClient.post @ packed_client.js?8f609640:7
OctoPrintClient.postJson @ packed_client.js?8f609640:7
OctoPrintBrowserClient.passiveLogin @ packed_client.js?8f609640:38
LoginStateViewModel.self.requestData @ packed_core.js?9ab43570:159
onServerConnect @ packed_core.js?9ab43570:1025
(anonymous) @ packed_core.js?9ab43570:1025
c @ packed_libs.js?ab739628:1
fireWith @ packed_libs.js?ab739628:1
s.<computed> @ packed_libs.js?ab739628:1
DataUpdater.self._onConnectMessage @ packed_core.js?9ab43570:838
(anonymous) @ packed_client.js?8f609640:14
Pn @ packed_libs.js?ab739628:1
(anonymous) @ packed_libs.js?ab739628:1
OctoPrintSocketClient.propagateMessage @ packed_client.js?8f609640:14
(anonymous) @ packed_client.js?8f609640:21
(anonymous) @ packed_libs.js?ab739628:1
_t @ packed_libs.js?ab739628:1
(anonymous) @ packed_libs.js?ab739628:1
(anonymous) @ packed_libs.js?ab739628:1
onMessage @ packed_client.js?8f609640:21
r.dispatchEvent @ packed_libs.js?ab739628:1232
(anonymous) @ packed_libs.js?ab739628:1232
w._transportMessage @ packed_libs.js?ab739628:1232
i.emit @ packed_libs.js?ab739628:1232
[... 64 lines cut ...]
s.<computed> @ packed_libs.js?ab739628:1
DataUpdater.self._onConnectMessage @ packed_core.js?9ab43570:838
(anonymous) @ packed_client.js?8f609640:14
Pn @ packed_libs.js?ab739628:1
(anonymous) @ packed_libs.js?ab739628:1
OctoPrintSocketClient.propagateMessage @ packed_client.js?8f609640:14
(anonymous) @ packed_client.js?8f609640:21
(anonymous) @ packed_libs.js?ab739628:1
_t @ packed_libs.js?ab739628:1
(anonymous) @ packed_libs.js?ab739628:1
(anonymous) @ packed_libs.js?ab739628:1
onMessage @ packed_client.js?8f609640:21
r.dispatchEvent @ packed_libs.js?ab739628:1232
(anonymous) @ packed_libs.js?ab739628:1232
w._transportMessage @ packed_libs.js?ab739628:1232
i.emit @ packed_libs.js?ab739628:1232
ws.onmessage @ packed_libs.js?ab739628:1232
packed_core.js?9ab43570:1018 Did not bind view model BackupViewModel to target #wizard_plugin_backup since it does not exist
packed_core.js?9ab43570:1018 Did not bind view model UsageViewModel to target #wizard_plugin_tracking since it does not exist
packed_core.js?9ab43570:1018 Did not bind view model SoftwareUpdateViewModel to target #softwareupdate_confirmation_dialog since it does not exist
packed_core.js?9ab43570:1018 Did not bind view model SoftwareUpdateViewModel to target #wizard_plugin_softwareupdate since it does not exist
packed_core.js?9ab43570:159 User St0fzuiger logged in
packed_core.js?9ab43570:164 Login mechanism: http
packed_core.js?9ab43570:1023 ... binding done
packed_core.js?9ab43570:1023 Application startup complete

FormerLurker, 2020-11-27:
Ok, thanks for that.  This is a duplicate issue I am still looking into.  Will link to it when I'm done with some household chores in a few hours.

Prutsium, 2020-11-27:
Thanks! and sorry for finding so many things but yeah running various setups here so find alot :)

FormerLurker, 2020-11-27:
Hey, keep em coming.  If you found an issue there are probably 100s of others who have it but aren't reporting it.  I have one printer and one pi set up, though my pi4 just arrived!  I wish I had a bunch of various hardware to test on!

Prutsium, 2020-11-27:
> 
> 
> Hey, keep em coming. If you found an issue there are probably 100s of others who have it but aren't reporting it. I have one printer and one pi set up, though my pi4 just arrived! I wish I had a bunch of various hardware to test on!

Thats where you have test dummy's for :) And yeah will keep them coming no worries

FormerLurker, 2020-11-29:
Ok, so this is very likely related to 1.5.0rcx.  A new version of knockout was included, and there were conflicts with some of the HTML.  Try installing from this link and let me know if your problems go away:

https://github.com/FormerLurker/Octolapse/archive/e5a1fba0ec5e667a681e6a4dbd15cff37a4d2226.zip

 It's quite possible that there are some unresolved UI issues still, so if you notice anything off let me know and I'll take a look.  a TON of files needed to be changed to make it work, and I likely missed a few.  Shouldn't affect your printer at all, this is all UI stuff.

Prutsium, 2020-11-29:
Wierd .... did the "update" and after restart plugin is gone .....
2020-11-29 16:17:12,091   /home/pi/oprint/bin/python3 -m pip --disable-pip-version-check install file:///tmp/tmp968um8e6/Octolapse-e5a1fba0ec5e667a681e6a4dbd15cff37a4d2226.zip --no-cache-dir
2020-11-29 16:17:15,356 > Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
2020-11-29 16:17:15,357 > Processing /tmp/tmp968um8e6/Octolapse-e5a1fba0ec5e667a681e6a4dbd15cff37a4d2226.zip
2020-11-29 16:17:22,847 > Requirement already satisfied: pillow>=6.2.0<7.0.0 in /home/pi/oprint/lib/python3.7/site-packages (from Octolapse==0.4.1+u.e5a1fba) (8.0.1)
2020-11-29 16:17:22,848 > Requirement already satisfied: sarge in /home/pi/oprint/lib/python3.7/site-packages (from Octolapse==0.4.1+u.e5a1fba) (0.1.5.post0)
2020-11-29 16:17:22,850 > Requirement already satisfied: six in /home/pi/oprint/lib/python3.7/site-packages (from Octolapse==0.4.1+u.e5a1fba) (1.15.0)
2020-11-29 16:17:22,851 > Requirement already satisfied: OctoPrint>1.3.8 in /home/pi/oprint/lib/python3.7/site-packages (from Octolapse==0.4.1+u.e5a1fba) (1.5.0rc3)
2020-11-29 16:17:22,851 > Requirement already satisfied: psutil in /home/pi/oprint/lib/python3.7/site-packages (from Octolapse==0.4.1+u.e5a1fba) (5.7.3)
2020-11-29 16:17:22,851 > Requirement already satisfied: file_read_backwards in /home/pi/oprint/lib/python3.7/site-packages (from Octolapse==0.4.1+u.e5a1fba) (2.0.0)
2020-11-29 16:17:22,852 > Requirement already satisfied: setuptools>=6.0 in /home/pi/oprint/lib/python3.7/site-packages (from Octolapse==0.4.1+u.e5a1fba) (50.3.2)
2020-11-29 16:17:22,852 > Requirement already satisfied: awesome-slugify<1.7,>=1.6.5 in /home/pi/oprint/lib/python3.7/site-packages (from Octolapse==0.4.1+u.e5a1fba) (1.6.5)
2020-11-29 16:17:22,852 > Requirement already satisfied: Jinja2<3,>=2.11.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.e5a1fba) (2.11.2)
2020-11-29 16:17:22,852 > Requirement already satisfied: PyYAML<6,>=5.3.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.e5a1fba) (5.3.1)
2020-11-29 16:17:22,853 > Requirement already satisfied: future<1,>=0.18.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.e5a1fba) (0.18.2)
2020-11-29 16:17:22,853 > Requirement already satisfied: Flask-Login<0.6,>=0.5 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.e5a1fba) (0.5.0)
2020-11-29 16:17:22,853 > Requirement already satisfied: tornado==5.1.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.e5a1fba) (5.1.1)
2020-11-29 16:17:22,853 > Requirement already satisfied: flask<2,>=1.1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.e5a1fba) (1.1.2)
2020-11-29 16:17:22,854 > Requirement already satisfied: pyserial<4,>=3.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.e5a1fba) (3.4)
2020-11-29 16:17:22,854 > Requirement already satisfied: pylru<2,>=1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.e5a1fba) (1.2.0)
2020-11-29 16:17:22,854 > Requirement already satisfied: frozendict<2,>=1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.e5a1fba) (1.2)
2020-11-29 16:17:22,854 > Requirement already satisfied: Flask-Assets<3,>=2.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.e5a1fba) (2.0)
2020-11-29 16:17:22,855 > Requirement already satisfied: cachelib<1,>=0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.e5a1fba) (0.1.1)
2020-11-29 16:17:22,855 > Requirement already satisfied: watchdog<1,>=0.10.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.e5a1fba) (0.10.3)
2020-11-29 16:17:22,855 > Requirement already satisfied: wrapt<2,>=1.12.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.e5a1fba) (1.12.1)
2020-11-29 16:17:22,855 > Requirement already satisfied: unidecode<0.05,>=0.04.14 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.e5a1fba) (0.4.21)
2020-11-29 16:17:22,856 > Requirement already satisfied: sentry-sdk<1,>=0.15.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.e5a1fba) (0.19.1)
2020-11-29 16:17:22,856 > Requirement already satisfied: Flask-Babel<2,>=1.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.e5a1fba) (1.0.0)
2020-11-29 16:17:22,856 > Requirement already satisfied: zeroconf<0.25,>=0.24; python_version >= "3" in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.e5a1fba) (0.24.5)
2020-11-29 16:17:22,856 > Requirement already satisfied: rsa==4.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.e5a1fba) (4.0)
2020-11-29 16:17:22,857 > Requirement already satisfied: regex!=2018.11.6 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.e5a1fba) (2020.10.28)
2020-11-29 16:17:22,857 > Requirement already satisfied: blinker<2,>=1.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1+u.e5a1fba) (1.4)
[... cut ...]
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/stabilization_gcode.py", line 24, in <module>
    from octoprint_octolapse.position import Pos
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/position.py", line 24, in <module>
    import octoprint_octolapse.utility as utility
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/utility.py", line 118, in <module>
    _SLUGIFY = Slugify.Slugify()
AttributeError: type object 'Slugify' has no attribute 'Slugify'

FormerLurker, 2020-11-29:
> SLUGIFY = Slugify.Slugify()
AttributeError: type object 'Slugify' has no attribute 'Slugify'

Drat, I fixed that for OctoPrint 1.5.x and broke it for 1.4.x  There are like a million test cases to deal with, so getting this right is difficult.  Has to work in python 2.7 and 3.x, has to work in old versions of Octoprint, has to work in multiple platforms.  Jeesh.. lol.  Anyway, back to the drawing board.  Will have something in a bit.

FormerLurker, 2020-11-29:
Apparently I did NOT fix it for 1.5??  That's SO strange because it is working for me..  Anyway, I wanted to mention I got your version wrong.  Been dealing with so many 1.5 issues my head is spinning.

Prutsium, 2020-11-29:
If you want i can create some remote access to this Pi as it runs standalone at the moment

FormerLurker, 2020-11-29:
That is very considerate.  I may take you up on that.  Currently I am in the middle of building a pi that matches your install more or less.  Will let you know if it comes to that.

Prutsium, 2020-11-29:
Ok no problem just let me know.
Best to shout out on Telegram to me then: @St0fzuiger

FormerLurker, 2021-01-03:
@Prutsium, have you tried installing this:  https://github.com/FormerLurker/Octolapse/releases/tag/v0.4.1

Been away for a bit, forgot to reference this issue in the release.  I believe it is solved, because nobody else has complained about it.  Let me know.

Prutsium, 2021-01-03:
> 
> 
> @Prutsium, have you tried installing this: https://github.com/FormerLurker/Octolapse/releases/tag/v0.4.1
> 
> Been away for a bit, forgot to reference this issue in the release. I believe it is solved, because nobody else has complained about it. Let me know.

Yeah it's working great actually had zero issues with it.
I think it's solved as i also dont see reactions from others (yet)