FormerLurker/Octolapse #634: Cura automatic slicer code update?

annovialb, 2020-11-29:
#### Version of Octolapse
 
Octolapse Version:  0.4.0

#### Version of OctoPrint

OctoPrint Version: 1.4.2

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

1. I was starting an Octolapse print.
2. I used gcode from cura 4.8.
3. I used this code: https://raw.githubusercontent.com/wiki/FormerLurker/Octolapse/version/0.4/assets/scripts/slicer/cura/4.2/settings_output.gcode
4. When starting print it reported that some values were missing and the print could not start
5. I have updated the cura code adding the following
; deretraction_speed = {retraction_speed}
; retraction_length = {retraction_distance}
; x_y_travel_speed = {travel_speed}
; z_lift_height = {z_hop_height}
; z_travel_speed = {z_hop_speed}

#### What should have happened?

#### What happened instead?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: ___REPLACE_THIS__OS_NAME_GOES_HERE
Os Version: ___REPLACE_THIS__OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: Ender 5 Pro
Printer Firmware Version: TH3D

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-11-29:
It would be helpful if you included the error message AND the gcode.  Also, the settings.json would be useful.  Please fill out the issue template more and i will take a look

annovialb, 2020-11-29:
I am not able to reproduce the error any longer. Is it possible that the settings are now loaded in Octolapse and thus not triggering the error any longer? Just wondering. Anyhow, the error was something such as “Setting missing: Retracion Length, Retraction Speed, Deretraction Speed, XY Travel Speed, Z Travel Speed, Z lift height cannot start print"

I am attaching the code in the version after the fix, and without the fix (I did regenerate it now, since the original is now lost).
How can I get the settings.json file?

Cheers,
Alberto

FormerLurker, 2020-11-29:
Here are instructions with pretty pictures : https://github.com/FormerLurker/Octolapse/wiki/V0.4---Reporting-An-Issue

:)

annovialb, 2020-11-29:
Here they are.

Cheers

FormerLurker, 2020-11-29:
Looks like github ate your link, lol.  Happens all the time.  Can you try again?  Thanks for being responsive!

annovialb, 2020-11-29:
Here it is attached again.
Thanks for your time.

FormerLurker, 2020-11-29:
Crazy, it's still not there.  Try posting to gist.github.com if you haven't already and paste in a link.

However, now that I'm on a PC I will respond in more detail.  If your printer profile is set to use 'automatic configuration' it will extract the settings from the gcode, and will always complain if any are missing (even if it worked previously).  Is it possible you tried to print an old gcode file that didn't have the settings in there?  I've done this more times than I can count while testing the software.  Even though it seems to be working for you now, I like getting to the bottom of this, but if you can't replicate the issue perhaps we can chalk it up to that?

annovialb, 2020-11-29:
Here it is on GitHub.
https://gist.github.com/annovialb/c5b67bdedec1d0f23a50a9fc9e28811e

> Crazy, it's still not there. Try posting to gist.github.com if you haven't already and paste in a link.
>
> However, now that I'm on a PC I will respond in more detail. If your printer profile is set [...]
I would say no. However it seems to be a plausible explanation, so maybe.
> I've done this more times than I can count while testing the software. Even though it seems to be working for you now, I like getting to the bottom of this, but if you can't replicate the issue perhaps we can chalk it up to that?
We can :)

FormerLurker, 2020-11-29:
Your settings look good, and I can see that they were eventually detected.  Not sure what happened there, but if it happens again, let me know!  Also, be sure to post again if you have any other issues.  Thanks!

closed 2020-11-29 by FormerLurker