FormerLurker/Octolapse #635: Octolapse caused a pole in the model

ilker-aktuna, 2020-11-29:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4.0

#### Version of OctoPrint

OctoPrint Version: 1.4.2

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

1. Printing any 3D model
2. Octolapse enabled and running

#### What should have happened?
Should print with no pole in the model body.

#### What happened instead?
A pole occured in the body of the model.

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.17.0

#### Printer model & used firmware incl. version

Printer Model: Ender 3 V2
Printer Firmware Version: 2.0.1

#### Browser and version of browser, operating system running browser

Browser: Chrome 
Browser OS: Windows 10

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: all gcode files I tried

#### Link to settings.json

Link to settings.json with all passwords removed: https://drive.google.com/file/d/1FNRpHjF3laCTMvIiKK7QW793tXk_EnI1/view?usp=sharing

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://drive.google.com/file/d/1iN2TOZsEJOJZyzigRmO3w2kF2EajvURo/view?usp=sharing

#### Link to octoprint.log

Link to octoprint.log:  https://drive.google.com/file/d/1aQMOjN3W2YGQXXPL9JFQntmaG5vUtBEb/view?usp=sharing

#### Link to contents of Javascript console in the browser

Link to javascript console output: 

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: https://drive.google.com/file/d/19riY63Oh-LStENGIHQV0K-zfALCDGTIg/view?usp=sharing

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-11-29:
Try uploading your gcode to gist.github.com and paste a link in here, thx.

FormerLurker, 2020-11-29:
Ok, without seeing your gcode files I can think of many reasons this might be happening.  All of them are explained [here](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Troubleshooting-Print-Quality-Issues).  Take a look at that wiki and let me know if any of the situations described there apply to you.  Also, if you haven't already done so, take a look at the [getting started guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Getting-Started), which walks you through some very important configuration steps, as well as explaining some of your options in detail.  Thanks!

ilker-aktuna, 2020-11-29:
well, I didn't provide a gcode file because it happens on all gcode files that I tried. (actually tried 3) 
as soon as I turn of Octolapse , it stops making that pole.

Actually, I just read the wiki you mentioned. But there are many suggestions there and trying each would cause many trial & error. 
I had thought that there would be a simple/direct answer to my question because this is not a print quality issue but a "pole" in the body of the model.
so it might be caused by the positioning of the extruder but I didnt change anything , it is on default settings. (Octolapse plugin)

I am using a USB webcam. so that might be an issue of speed ? (I am not sure)
do you see anything on the logs ?

any suggested settings for Ender 3 V2 printer ? since this is a very common printer I thought it would work with a standard setting...

Do NOT use a DLSR 
- I am using a webcam

Reduce your camera delay 
- I just set it to 0 , but I am not sure if it will work. Besides , if lowest is recommended, then why the default is 125ms _?

Increase your webcam framerate
- how to do that ? On Octoprint ?

Change your stabilization position 
- it is already in the middle of the model, so changing that would not matter much ? Any recommended position ?

Increase your travel speed 
- it is already set to 150mm/s , the fastest on my printer I guess..

Try the Smart - Snap To Print trigger type 
- this was set to "smart- compatibility" , so the recommended is "snap to print" ?

Turn Up Your Layer Height
- it is set to 0.2mm , so that's already recommended setting.
But I am using z-hop , would that matter ?

Consider getting a faster SD card
- I am using a Sandisk 100MB/s write speed card.

Consider getting a faster Pi -
- Using Pi3 B+

Check your power supply 
- my pi is powered through its original adapter.

Cool off your Pi 
- I am using a heatsink on the Pi3

Check your USB Cable
- using a good USB cable. Tested with high speed file transfer. no issues there.

Check your Gcode 
- Happened with 3 different gcode files created with my Cura

FormerLurker, 2020-11-29:
The gcode file is important because I can use it in my debugger to see what Octolapse is doing.  Since the gcode is tied to the way Octolapse determines how to take a snapshot, it is vitally important, especially since you are seeing this in all your models.

The 'pole' is actually a print artifact caused by oozing, and is a well known issue.  Since Octolapse can stabilize in the same position over and over (depneding on your settings), any oozing gets built up there and can create this tower like structure you are seeing.  Enders (and other bowden style setups) are particularly vulnerable to this issue since there is usually some extra pressure in the teflon tube that take a bit of time to release, which can cause extra oozing during travel.  Also, I can see your slicer settings in the gcode file, which are a big help.  Since you are printing small models, I'm guessing (but do not know for sure without the gcode file) that you are printing with a very small layer height, which compounds oozing issues (as is stated in the quality guide).  Octolapse has many features to mitigate this, but lots of things are out of my control (like your slicer settings, feature comments in the gcode file, etc).

Additionally, the 'settings.json' file you uploaded is not from octolapse.  Please see [this guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Reporting-An-Issue) which details the steps for reporting an issue, including how to find the proper settings.json file.

Keep in mind that Octolapse has to work on many different platforms, with many different firmwares, many different slicers, and on many different printers.  This all means I need a bunch of help to figure out what is going on (like the gcode file) to solve issues like this.  I don't want to discourage or frustrate you, but it takes so much time to review an issue like this (hours, possibly days), and without all the info it's difficult to get to the bottom of things.

Also, FYI, I know this can be made to work on an Ender 3 V2 with minimal artifacts.  See the 'Wild Rose Builds' videos for many examples.  It just takes some time and effort to configure your printer, your slicer, and Octolapse.  If you keep at it I'm sure you can get success!  I'm willing to help, just need all the necessary info.

ilker-aktuna, 2020-11-29:
thanks for your reply.
link to gcode file:
https://drive.google.com/file/d/10GOAyCV8ZqbY-oRPZdYK0LqLneO75-Qq/view?usp=sharing

link to correct octolapse settings.json:
https://drive.google.com/file/d/10GOAyCV8ZqbY-oRPZdYK0LqLneO75-Qq/view?usp=sharing

I didn't mean to be rude by not sending the gcode file initially. I just thought that the issue is not related to gcode file since it happened with all of the gcode files I created on Cura.

Of course, I am willing to help you solve an issue (if there's any) and I can supply you any required data.

FormerLurker, 2020-11-29:
> I didn't mean to be rude by not sending the gcode file initially.

@ilker-aktuna, I did not take it that way, just wanted to explain clearly why it was so important that I receive it.  I really do appreciate your post and want to your problem solved.  I'm trying to work on my communications, but it is a work in progress :)

> I just thought that the issue is not related to gcode file since it happened with all of the gcode files I created on Cura.

It's reasonable that you felt that way.  Octolapse is somewhat mysterious in how it operates :)

I'll dive into this as soon as I finish fixing some critical issues related to the upcoming OctoPrint 1.5 release.  Otherwise Octolapse will not work for anyone, lol.

ilker-aktuna, 2020-11-30:
thanks for your response.

while you are working on adapting Octolapse to Octoprint 1.5 , do you have any recommendation for me to test according to my answer in previous post ?
I tried to adjust according to the wiki not much I could change...

FormerLurker, 2020-11-30:
So, I took a few seconds to look at your slicer settings and your Octolapse configuration, and there are quite a few problems.  

1. Automatic slicer settings configuration is not enabled. This is marked in the print quality guide. but the link is broken (it's in the full wiki and the getting started guide, and those links work).  Perhaps you gave up because the link is broken, so [here is one that works too](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Automatic-Slicer-Configuration).  Follow that FOR SURE.  At the very least it will prevent problems and save time in the long run.

2.  Retraction AND z-lift appears to be disabled in your slicer!  This might work printing a single simple model, or if travel moves are REALLY close together, but it will mask issues that will start to appear on more complicated prints.  You will want to enable both if you want to reduce problems in Octolapse.  This must be done in the slicer.  If you follow the guide for enabling automatic slicer settings exactly (step 1) you won't have to do anything in Octolapse once you enable retraction and z lift.  I'm not sure what the max z axis speed is for your printer, but it is set to 300mm/min, which is very slow.  Retraction speed should probably be increased too, but I'm NOT an expert on your printer, so maybe not.  Retraction length/speed tuning is not easy, so I recommend reading about it if you are new to tuning.

3.  Print a stringing test WITHOUT Octolapse.  This is essential for tuning retraction/deretraction length and speed.  I have a feeling you will have some problems with this.  Octolapse doesn't have a chance at preventing artifacts if your printer has poor results with a stringing test.  I recommend running the test printing from the outside to the inside (I think this setting is called something like 'Infill prints after perimeters'), just make sure you disable it again when you are finished else you may have problems with steep overhangs failing.  Please also note that zhop may make it harder to pass a stringing test, but since it typically makes octolapse work better I recommend doing this with zhop enabled.  For our purposes you can set zhop (lift) to a low value,  say to your layer height of 0.2mm.

4.  Try fresh PLA - Old pla strings and can ooze more than fresh.  Might be some moisture in there.  Get PLA working before trying PETG or anything else.

5.  Print a dual temperature tower test (basically a temp tower test for stringing, two towers each segment printed at a different temp).  This can be a pain to setup, but I've seen this make a HUGE difference.  Sometimes 10-15C lower or higher can make all the difference.  It's good to do this after tuning in retraction, and better still to go back and re-tune retraction after you've dialed in the temp.

I also wanted to comment on something you said earlier:

>  I had thought that there would be a simple/direct answer to my question because this is not a print quality issue but a "pole" in the body of the model.

Unfortunately there is not a simple answer, but it is a quality issue, and it will take some time to fix. It's just a unique quality issue that's caused by oozing.  I know you'll never see that pole without Octolapse, but you would have other issues that aren't as obvious.   If you work on the tuning now, ALL of your prints will turn out better, even when you aren't using Octolapse!  However, because of the physics of 3d printers, Octolapse can always cause SOME quality problems, but it can be reduced almost to the point of vanishing in most cases.

Let me know how it goes, and sorry if some of this advice is obvious to you (not sure where you are in your 3d printing journey)!  Also, I wish this were less work, but bowden style printers just tend to ooze more than direct drive.

ilker-aktuna, 2020-12-03:
thanks.

I had seen the auto slicer config but since I use Cura 4.8 , it was easier to select "Cura 4.2 or above" so I went that path. Now that you recommend auto slicer configuration, I will do it.

>Retraction AND z-lift appears to be disabled in your slicer

How ? On Cura profile, I didn't make any change to Ender 3 settings and when I check I see, under "Travel" , "enable retraction" selected.
About z-lift , I could not find any profile configuration. How can I enable it ?

> I'm not sure what the max z axis speed is for your printer, but it is set to 300mm/min, which is very slow. Retraction speed should probably be increased too, but I'm NOT an expert on your printer, so maybe not.

Well, I am just using my printer with default settings and default Ender 3 profile for Cura.
So I have zero knowledge on these settings. And I can not find any clue on Cura profile settings. Where should I look for these settings ?
I know these are not something to be consulted to you but since you recommended them, I would like to take my chance and ask... 

"stringing test"
When I google it, I saw this:
https://www.thingiverse.com/thing:2080224

would it be fine to use it ?

FormerLurker, 2020-12-03:
> How ? On Cura profile, I didn't make any change to Ender 3 settings and when I check I see, under "Travel" , "enable retraction" selected.

Take a snapshot of these settings. I inferred your settings by manually reading gcode in absolute E mode, which is difficult and prone to errors:

[image]

Maybe I was wrong.

> About z-lift , I could not find any profile configuration. How can I enable it ?

[image]

Also
> So I have zero knowledge on these settings. And I can not find any clue on Cura profile settings. Where should I look for these settings ?

You can see that I underlined this setting in the above image (Z Hop Speed).  Please search online and find someone knowledgeable to tell you the max z axis speed.  I would say that if you are hopping only a small distance (between 0.2 and 0.6mm or so), the acceleration setting for the Z axis will prevent you from being able to move too fast, so you can probably leave that alone for now.  Start off with 0.6mm for z hop height when doing tests.

> would it be fine to use it ? (https://www.thingiverse.com/thing:2080224)

Yeah, should be fine.  I prefer this one myself (it is a bit more difficult): https://www.thingiverse.com/thing:2219103

> So I really don'T understand how you see retraction and z-hop disabled.

Because I didn't see any evidence in your gcode.  Keep in mind I was trying to read all your gcode by sight, and it's even harder to do since you are using absolute extrusion (no easy to read minus sign for retraction), so it's possible I made a mistake.  However, I know with 100% certainty that z-hop and retraction weren't enabled in the Octolapse profile.  You are supposed to copy those settings from your slicer to Octolapse every time they change UNLESS you use the automatic slicer settings, which I recommended you use in the last post.

Does that clear things up?

ilker-aktuna, 2020-12-03:
[image]

[image]

ilker-aktuna, 2020-12-03:
ok. so I have z-hop and retraction enabled.
You recommend using 0.6mm for z-hop height instead of 0.2mm

and you also recommend using a better retraction setting. I just watched this video : 
https://www.youtube.com/watch?v=GXSqZ68UdsE

I can copy from that.

Then you advise to use auto slicer settings. 

and a string test, I will do that.

ilker-aktuna, 2020-12-03:
btw, in fact now I am using a profile which is made for Ender 3 V2 (my printer) , so maybe they are good settings.
So the problem might be that I am not using Auto slicer config.
is that possible ? (that only problem is auto slicer setting)

FormerLurker, 2020-12-03:
@ilker-aktuna,

> You recommend using 0.6mm for z-hop height instead of 0.2mm

Yeah, I struggled with that.  It's hard to recommend settings for a printer you don't own or use....  0.2 is better than 0 though.  1-2mm is prob too much, but maybe not.  You want it as low as possible.  It should pull the filament up a bit and cut it off as it's traveling, and should NEVER bump against your printed part as it moves over it.

> I can copy from that.

Beware, the best settings to use are usually filament and temperature specific.  Worst case scenario you can just revert to the defaults.  But, only change one settings at a time, run a stringing test, and then change something else.  Make sure octolapse is disabled at first too.  Only enable Octolapse once you start to get good results with the test, then you can continue to tweak it.  

> btw, in fact now I am using a profile which is made for Ender 3 V2 (my printer) , so maybe they are good settings.

Good in that they are generally good for starting out, yes.  There is probably no magic number that will work for all conditions, filament types and brands, nozzles, temperatures, etc..  When people talk about 'dialing in' their printer, this is a small part of what they mean.

> So the problem might be that I am not using Auto slicer config.

I'd say that is the biggest problem.  The slicer settings in cura MUST match the slicer settings in Octolapse 100%.  It's very technical why that is true, but automatic slicer settings eliminate that problem completely, and is more convenient.  It's really really easy to do.  Just paste in a short script and reslice..

ilker-aktuna, 2020-12-03:
there's just one point that I don'T understand about auto slicer setting:
since I use Cura 4.8 ,  I have to add the script from here:
https://raw.githubusercontent.com/wiki/FormerLurker/Octolapse/version/0.4/assets/scripts/slicer/cura/4.2/settings_output.gcode
to the start code of my Cura.
But this script is empty (it is all commented out), it sets nothing. So is it an error, or is there any magic ?

FormerLurker, 2020-12-03:
>  is there any magic

Yes there is.  All of Octolapse is magic :)  

Basically cura will output the settings where it sees things like ```{speed_z_hop}```.  Octolapse reads these in, fills in your slicer settings, and then moves on.. It does this at the start of every print.  If these settings aren't found (like if you use old gcode that doesn't have the script), Octolapse will complain and won't let you start your print with Octolapse enabled.

Follow [this process](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Automatic-Slicer-Configuration#if-you-are-using-cura-follow-these-steps) and you will be fine.  There are videos about this too, but they just follow this guide.  Should take like 2 minutes.  You'll know if it worked or not, because Octolapse will throw an error and stop your print if something's not working.  All auto-magic.  If you want, look at your gcode file and you'll see all your settings got filled in after you re-slice.

ilker-aktuna, 2020-12-04:
[image]

I just printed this without any additional settings. Do you think it is ok ?
Now I will print same with Octolapse and making Auto slicer config.

FormerLurker, 2020-12-04:
Id like to see one done with octolapse.  I think you could Improve your string test eventually, but it would be interesting to see a side by side with the same settings

ilker-aktuna, 2020-12-04:
btw, about these recommendations in the faq:

>Reduce your camera delay

I just set it to 0 , but I am not sure if it will work. Besides , if lowest is recommended, then why the default is 125ms _?

>Increase your webcam framerate

how to do that ? On Octoprint ?

>Change your stabilization position

it is already in the middle of the model, so changing that would not matter much ? Any recommended position ?

>Try the Smart - Snap To Print trigger type

this was set to "smart- compatibility" , so the recommended is "snap to print" ?

ilker-aktuna, 2020-12-04:
[image]

this is done with Octolapse enabled.
so , with the Auto slicer settings, "pole" is gone.
But I have to work on those strings...
I'll watch that video again and try to improve it...

ilker-aktuna, 2020-12-04:
btw, how can I adjust Octolapse so that the extruder is not in view while snapshot takes place ?

FormerLurker, 2020-12-04:
> why the default is 125ms _?

Because that is the lowest delay that will produce stable images for most webcams.  Defaults are made to work well in the average case, they are not tuned for any particular use case.

> Increase your webcam framerate

You should Google that.  There are tons of guides out there for changing your resolution and framerate for octopi.

> it is already in the middle of the model, 

Actually, it was at the center of the bed.  The majority of the model was not on the exact center.  If you move the bulk of your model to the center of the bed so the the extruder is over infill when taking a snapshot, most of that tower you saw would have been hidden.  Make sense?

> this was set to "smart- compatibility" , so the recommended is "snap to print" ?

No, it is not the recommended option in the sense that it is the best one overall.  It is the best one if you are primarily concerned about print quality, but want a 'good enough' timelapse.  Read about what that trigger does, there are help bittons in the UI, wiki pages, and youtube videos on my channel.  Or try it out and see what it does for yourself.  I think it is pretty neat.

FormerLurker, 2020-12-04:
> this is done with Octolapse enabled.

Looking better.  I think it could get even better.  I think your retraction settings and maybe your extruder temperature could use a few tweaks, at least for this filament.  There are things you can do in Octolape too, as you mentioned.  I recommend changing only one thing at a time so you can see the effect.  Taking pics after each change and noting your settings is also useful.

So, i think you are well on your way.  Post a link if you come up with some interesting timelapses!

ilker-aktuna, 2020-12-04:
btw, how can I adjust Octolapse so that the extruder is not in view while snapshot takes place ?
I don't see the extruder on some sample videos.

FormerLurker, 2020-12-04:
https://github.com/FormerLurker/Octolapse/wiki/V0.4---Profiles---Stabilization

ilker-aktuna, 2020-12-04:
[image]

According to the recommended settings on that video I found, I changed:
retraction distance to 6mm (from 6.5mm)
retraction minimum travel to 0.8 (from 1.5)
max retraction count to 90 (from 100)

Now , when compared to the previous test, I have a cleaner body in middle, but there is a thick string at the end of the print between towers. 
Any idea, why that happened with these adjustments ?
I'm sorry to bother you with this unrelated question, but since I don't understand all of these settings maybe you could tell me which of these settings could cause this thick string

FormerLurker, 2020-12-04:
Well, you changed 4 things at once, so it's hard to tell what is going on.  I don't have an ender to test on, so I'm not the best source of help here.  However, if I had to pick one of those settings that would have the biggest impact, it's the retraction distance.  I recommend doing some research to learn exactly what the settings do and how they might impact your print.  Also, I really recommend against changing more than 1 setting at a time, like I said before.  Record what you change, take a pic of the result (the print) and keep it for comparison later.  It's a process.

Here is an interesting video with a potential fix that I saw a while back:  https://www.youtube.com/watch?v=dIkjR2Ytx-g&ab_channel=Tombof3DPrintedHorrors

Requires some modifications.  Use at your own risk.

ilker-aktuna, 2020-12-05:
thanks for all your help, I achieved a good print :)
video  here: 

https://www.youtube.com/watch?v=LlTBpWGE3T4

FormerLurker, 2020-12-05:
Nice!! Snap to print FTW :)

FormerLurker, 2020-12-05:
Congratulations!  And.... Closed!

closed 2020-12-05 by FormerLurker

ilker-aktuna, 2020-12-05:
btw, is it safe to upgrade to 1.5 now ? have you adjusted Octolapse for this version of Octoprint ?

FormerLurker, 2020-12-05:
Should be.  Uograde to octolapse V0.4.1 before you upgrade OctoPrint