FormerLurker/Octolapse #636: Verbose Is not getting compiled with G-code

bangor-flyer, 2020-11-30:
Tried compiling g-code with 2.3.55  and octolapse claims no verbose g-code.
went back to 2.3.54 and no problems ?

#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: ___REPLACE_THIS__OCTOLAPSE_VERSION_GOES_HERE

#### Version of OctoPrint

OctoPrint Version: ___REPLACE_THIS__OCTOPRINT_VERSION_GOES_HERE

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: ___REPLACE_THIS__YES_OR_NO

#### What were you doing when the problem occurred

1. ___REPLACE_THIS__STEP_ONE_GOES_HERE
2. ___REPLACE_THIS__STEP_TWO_GOES_HERE
3. ___REPLACE_THIS__STEP_...

#### What should have happened?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: ___REPLACE_THIS__OS_NAME_GOES_HERE
Os Version: ___REPLACE_THIS__OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-11-30:
Please fill out the template a bit more.  Im not sure what your problem is exactly.

FormerLurker, 2020-11-30:
To be a bit more specific, v 2.3.54 of what exactly?

FormerLurker, 2021-01-03:
Closing this since the issue hasn't been updated in a while.  Let me know if you want me to reopen it.

bangor-flyer, 2021-01-03:
Hi,
i replied with the requested information and was awaiting a response from
yourself on replicating the issue.
Thanks
Peter

bangor-flyer, 2021-01-03:
Apolliges dear friend we reopened the issue
Verbose is not appearing with G-Code for octolapse #739

FormerLurker, 2021-01-03:
> Verbose is not appearing with G-Code for octolapse #739

I'm confused.  There is no issue 739.  Can you clarify what you mean?

bangor-flyer, 2021-01-04:
Hi FormerLurker/Octolapse,
I just took a screen grab from the issues page and hope it makes sense.
And again thank you for all your hard work on superslicer
Peter
[image: image.png]

FormerLurker, 2021-01-04:
The screenshot is not appearing.  If you attempted to email it, that's probably why.  Please double check to ensure the image gets uploaded and I will take a look.

bangor-flyer, 2021-01-04:
Donkey alert sorry buddy, my 1st day back at work and my head has been
scrambled.
 I attached it this time to make sure
Thanks
Pete

FormerLurker, 2021-01-04:
Try installing this new build from the plugin manager via get more->from url.. and paste this url in that box:

```https://github.com/FormerLurker/Octolapse/archive/698ef46ddbd6612173bd8e17eb32549b87c7a756.zip````

Let me know if that solves this issue.

FYI, I did NOT fix detection for the skirt, since there is currently a typo in superslicer.  If that gets fixed, then that will start working too :)