FormerLurker/Octolapse #637: After upgrade to Octoprint 1.5 rendering dialog does not open

nanuit, 2020-12-01:
Octolapse Version: 0.4.0

OctoPrint Version: 1.5.0

#### When you ran into the problem, did you have diagnostic logging enabled?
Diagnostic Logging was Enabled: YES
No corresponding entry in logfile at error occurance

1. Activating octolapse tab in Octoprint
2. Klick on button "Renderings ..."

#### What should have happened?
renderings dialog should open
Before update to Octoprint 1.5.0 everything worked fine

#### What happened instead?
error in chrome console

#### Operating System running OctoPrint and Octolapse
OS Name: Octopi
Os Version: 10

#### Browser and version of browser, operating system running browser
Browser: Chrome
Browser OS: Windows 10

#### Link to contents of Javascript console in the browser

Link to javascript console output: https://gist.github.com/nanuit/e3ee09186217fc2139916ae3e759dbaa

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: https://user-images.githubusercontent.com/503026/100744413-14d21280-33de-11eb-808e-0230ed95b94b.jpg

FormerLurker, 2020-12-01:
There is a release candidate that fixes this already.  If you wout like to help fix this, install from the plugin manager with this link in the From Url... Box: https://github.com/FormerLurker/Octolapse/archive/v0.4.1rc1.zip

FormerLurker, 2020-12-01:
Also, please report back if it works (or not).  I need more confirmation before I release.  Thanks!!

nanuit, 2020-12-01:
Yes, with the release candidate it works fine again

FormerLurker, 2020-12-01:
Cool, that's 6 confirmations.  Probably going to drop a release tonight.  Can you let me know if you get an update notification?  I had to change that too, and if it's not working I want to pull it ASAP.  I will message you when it drops, and maybe you can manually run the update check?  

Thanks for getting back to me!

rickgode, 2020-12-01:
Yeah, I think the OctoPrint 1.5 update broke something.  I can verify that this release candidate seems to have fixed it.
[image]
[image]

FormerLurker, 2020-12-01:
Great!  Can anyone here tell me if the rc works on 1.4.x??  Most users are still on that, so I need some confirmation that it's working there too (besides just me) before I drop this update.

Sorry folks, never rushed a release like this and I'm super nervous.

rickgode, 2020-12-02:
Sorry... I can't help with that.  On a related note... Have you heard of Octolapse taking OctoPrint offline?  I'm sitting here watching it (it's actually still printing) but I saw the control screen go black and "server is offline" message.  I just restarted it before applying the release candidate to see if it would fix it before I came on this thread.

[image]

FormerLurker, 2020-12-02:
That is... Unusual.  Never seen it.  Wait for the print to end before messing with it... Could it have lost network?  Can you ssh (should be safe even while printing)

FormerLurker, 2020-12-02:
Also, could be octolapse..  just because I have never seen it doesn't mean it is not real.  I take all these reports seriously.

Sembazuru, 2020-12-02:
I have similar experience with 0.4.1rc1 (installed by changing the release channel in the software update, so that also works) in OctoPrint 1.5.0 as @rickgode .
I've also had server offline issues, but never attributed to this plugin because I've gotten server offline issues with this plugin turned off. When I have the server offline issues I can't even SSH into my RPi4. I just (re)learned how to switch virtual terminals on my RPi (last time I used that sort of feature was on a VT320 terminal) so the next time I see this problem I can log into my RPi locally (around OctoDash) to see what wlan0 is up to...

nanuit, 2020-12-02:
I had also sometimes offline issues, but not related to octolapse. The wifi connection on the octopi was sometimes down. First i corrected it by reseting the wifi connection to octopi on my router, but now i have a python wlan0 restart script an had no more problems since then.

rickgode, 2020-12-02:
Thanks everyone, I don't think my wireless network dropping was octolapse either (I had the plugin disabled).  

No I am not able to ssh into my RPi when it happens.  It's happened before and seems to happen once a month or so, I'll have to unplug/replug the RPi when it does because it just seems to lose the wireless, but this was the first time I was watching the control camera feed and it just went black then went to server offline.  (At least it continues to print) 

I don't even think it's a OctoPrint thing?

Thanks again FormerLurker for a FANTASTIC plugin!

Lennaert78, 2020-12-02:
The tab button for the videos and images is not working. 
Looks like near the same error.
octoprint 1.5.0 Python 2.7.16 Octopi 0.17.0

Error from Cons

packed_plugins.js?2c72d0e7:22114 No dialog has been found to open.  Is the dialog id correct?
self.show @ packed_plugins.js?2c72d0e7:22114
self.open @ packed_plugins.js?2c72d0e7:23946
self.openTimelapseFilesDialog @ packed_plugins.js?2c72d0e7:18245
(anonymous) @ packed_libs.js?ab739628:5
dispatch @ packed_libs.js?ab739628:1
v.handle @ packed_libs.js?ab739628:1
packed_core.js?9ab43570:352 

___________________________________________________
After this I constantly get :

packed_core.js?9ab43570:352 Terminal: Client speed recovered, enabling fancy terminal functionality
packed_core.js?9ab43570:350 Terminal: Detected slow client (needed 703ms for processing new log data), disabling fancy terminal functionality
TerminalViewModel.self._disableFancy @ packed_core.js?9ab43570:350
TerminalViewModel.self.fromCurrentData @ packed_core.js?9ab43570:354
callViewModelIf @ packed_core.js?9ab43570:964
(anonymous) @ packed_core.js?9ab43570:958
Pn @ packed_libs.js?ab739628:1
(anonymous) @ packed_libs.js?ab739628:1
callViewModelsIf @ packed_core.js?9ab43570:958
callViewModels @ packed_core.js?9ab43570:957
(anonymous) @ packed_core.js?9ab43570:845
DataUpdater.self._ifInitialized @ packed_core.js?9ab43570:859
DataUpdater.self._onCurrentData @ packed_core.js?9ab43570:845
(anonymous) @ packed_client.js?8f609640:14
Pn @ packed_libs.js?ab739628:1
(anonymous) @ packed_libs.js?ab739628:1
OctoPrintSocketClient.propagateMessage @ packed_client.js?8f609640:14
(anonymous) @ packed_client.js?8f609640:21
(anonymous) @ packed_libs.js?ab739628:1
_t @ packed_libs.js?ab739628:1
(anonymous) @ packed_libs.js?ab739628:1
(anonymous) @ packed_libs.js?ab739628:1
onMessage @ packed_client.js?8f609640:21
r.dispatchEvent @ packed_libs.js?ab739628:1232
(anonymous) @ packed_libs.js?ab739628:1232
w._transportMessage @ packed_libs.js?ab739628:1232
i.emit @ packed_libs.js?ab739628:1232
ws.onmessage @ packed_libs.js?ab739628:1232
packed_core.js?9ab43570:352 Terminal: Client speed recovered, enabling fancy terminal functionality

_________
not sure what that is about

FormerLurker, 2020-12-02:
@Lennaert78, and you installed the new version 0.4.1rc1?  If you have installed it, have you rebooted?

Lennaert78, 2020-12-02:
@FormerLurker,  installed that one mentioned moments ago.
The plugin manager installer asked  me to restart when done.

Button is now working again.

current output from console:

Starting dependency resolution...
packed_core.js?9ab43570:1002 ... dependency resolution done
packed_core.js?9ab43570:1025 Initial application setup done, connecting to server...
packed_core.js?9ab43570:838 Connected to the server
packed_core.js?9ab43570:843 Triggering connect on all view models
packed_core.js?9ab43570:845 Server (re)connect processed
packed_core.js?9ab43570:1025 Finalizing application startup
packed_core.js?9ab43570:1006 Going to bind 50 view models...
packed_plugins.js?6c999834:20897 octolapse.help.js - Binding help links to #octolapse_tab a.octolapse_help[data-help-url]
packed_core.js?9ab43570:1018 Did not bind view model TimelapseViewModel to target #timelapse since it does not exist
packed_core.js?9ab43570:1018 Did not bind view model TimelapseViewModel to target #timelapse_link since it does not exist
packed_core.js?9ab43570:1018 Did not bind view model BackupViewModel to target #wizard_plugin_backup since it does not exist
packed_core.js?9ab43570:1018 Did not bind view model UsageViewModel to target #wizard_plugin_tracking since it does not exist
packed_core.js?9ab43570:1018 Did not bind view model SoftwareUpdateViewModel to target #softwareupdate_confirmation_dialog since it does not exist
packed_core.js?9ab43570:1018 Did not bind view model SoftwareUpdateViewModel to target #wizard_plugin_softwareupdate since it does not exist
packed_core.js?9ab43570:1018 Did not bind view model BLTouchViewModel to target settings_plugin_BLTouch_form since it does not exist
packed_core.js?9ab43570:1018 Did not bind view model bedlevelvisualizerViewModel to target #wizard_plugin_bedlevelvisualizer since it does not exist
packed_core.js?9ab43570:159 User lennaert logged in
packed_core.js?9ab43570:1023 ... binding done
packed_core.js?9ab43570:1023 Application startup complete
6The resource <URL> was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.
2packed_plugins.js?6c999834:20897 octolapse.help.js - Binding help links to .octolapse .webcam_settings a.octolapse_help[data-help-url]
packed_plugins.js?6c999834:20897 octolapse.help.js - Binding help links to #octolapse_timelapse_files_dialog a.octolapse_help[data-help-url]
DevTools failed to load SourceMap: Could not load content for http://192.168.178.12/static/webassets/bootstrap-colorpicker.css.map: HTTP error: status code 404, net::ERR_HTTP_RESPONSE_CODE_FAILURE

______________________________
looks good, some minors there.

Also, a preview video popup  player with for the timelaps video's list would be awesome ^_~

Nice work
Thanks

FormerLurker, 2020-12-02:
Hmmm..  some of those errors are no good.  Will do some testing.  Looks like the colorpicker in the rendering overlay wont work either..

FormerLurker, 2020-12-02:
It looks like these are the errors that apply to Octolapse:
```
2packed_plugins.js?6c999834:20897 octolapse.help.js - Binding help links to .octolapse .webcam_settings a.octolapse_help[data-help-url]
```
This might be easy to fix.  Will double check tonight.
```
packed_plugins.js?6c999834:20897 octolapse.help.js - Binding help links to #octolapse_timelapse_files_dialog a.octolapse_help[data-help-url]
```

that shouldn't be too bad either.  Probably another self closing tag in there that I missed.

```
DevTools failed to load SourceMap: Could not load content for http://192.168.178.12/static/webassets/bootstrap-colorpicker.css.map: HTTP error: status code 404, net::ERR_HTTP_RESPONSE_CODE_FAILURE
```

That was probably caused by some recent javascript updates.  I'm pretty sure that colorpicker js is from Octolapse (need to double check), but I'm not sure how I'm going to fix that in a backwards compatible way.  Will think about it.

dualznz, 2020-12-03:
i too am getting dialogs and formatting issues throughout octolapse plugin on 0.15.0

FormerLurker, 2020-12-03:
Check the first comment @dualznz, there is a fix

thijsh, 2020-12-03:
Octolapse v0.4.1rc1 fix confirmed. Thanks! :)

dualznz, 2020-12-03:
Indeed confirmed on rc1 build cheers

homerisback, 2020-12-04:
Hi FormerLurker,
Octoprint/Octolapse beginner, I had the same screenshots rickgode shown and nothing started nor happened when enabling Octolapse. The plugin prevented any print to start, even in Octolapse's test mode.

I started with Octoprint 1.5.0 and Octolapse 0.4.0, no update from previous version involved in my issue.

The RC1 you released earlier, solved it instantaneously, great job ! I now can enjoy my Prusa Mini+ working, many thanks !

J0nnyBr4v0, 2020-12-04:
> Hi FormerLurker,
> Octoprint/Octolapse beginner, I had the same screenshots rickgode shown and nothing started nor happened when enabling Octolapse. The plugin prevented any print to start, even in [...]

Have the same issue on OctoPrint 1.5.1

KenwoodFox, 2020-12-04:
This fix seems to work

FormerLurker, 2020-12-04:
Awesome folks!  I finally found someone to test this in OctoPrint 1.4.x, and it seems to be OK there too.  Prepping the release for deployment, thanks to everyone who reported back!

FormerLurker, 2020-12-05:
The release has dropped. I might have goofed the version number if you installed one of the pre-release versions, which will prevent the software update from notifying you. Not much has changed aside from some formatting fixes, an added wiki link on the tab, and an about page update, but if you want to update to the official stable release, just install from this release URL and everything will be good: https://github.com/FormerLurker/Octolapse/archive/v0.4.1.zip

Otherwise updates will pick back up starting from 0.4.2.

Thanks!

Closed :)

closed 2020-12-05 by FormerLurker

nanuit, 2020-12-05:
I had the 0.4.1 rc1. I did get the update notification and the update went well. so everything is ok on my side.

Thanks