FormerLurker/Octolapse #639: Error when cancel a print

hashoom8, 2020-12-02:
OctoPrint Version: 0.4.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. Enabled
2. edit Settings and Printer Profile
3. Stabilization Disabled
4. Trigger : Classic - Every Layer
5. start print 
6. error in printing , need to stop the printer  

#### What should have happened?
 I press stop printing in display menu and the extruder start pushing melted 
Filament

#### What happened instead?
stop printing

#### Operating System running OctoPrint and Octolapse

Printer Model:  ender 5 plus
Printer Firmware Version:  marlin 2.0.5

Firmware Settings
The following settings depend on your printer's firmware.

X/Y/Z Axis Mode
Default To Absolute
 
E Axis Mode
Default To Absolute
G90/G91 Influences Extruder
False
 
Default Units
Millimeters
 
i test it 4 times and its from the plugin

 its look like its extruding the same amount as the printed part , 
i think its from one of those Settings

FormerLurker, 2020-12-02:
This: G90/G91 Influences Extruder

must be true in Marlin 2.  Looks like the firmware got an upgrade.  That is a beta profile for a reason :)

So, it's pushing plastic after a cancel because it's probably still extruding from the previous command.  If that setting is wrong, all hell can break loose.  FYI, I'm working on a 'firmware checker' tool for my other plugin, Arc Welder, and it's got a library containing the correct setting for G90/G91 influences extruder in there.  There is no good way to test for it universally unfortunately, but in general it's true for Marlin 2+ and smoothieware.

FormerLurker, 2020-12-02:
Here is a snapshot of the firmware checker, which I hope to add to Octolapse at some point to auto-configure some stuff:

[image]

FormerLurker, 2020-12-02:
OH, also, I noticed your running the classic trigger with stabilization disabled.  If you don't care that your timelapse is stabilized, you could use this new experimental option in the stabilization advanced settings to disable M400:

[image]

If you do that there should be no delay whatsoever, just like the stock OctoPrint timelapse.

hashoom8, 2020-12-02:
you are right ,   it's probably still extruding from the previous command ,

so G90/G91 Influences Extruder true , should fix the problem , i thought its m600 but i don't have runout sensor

FormerLurker, 2020-12-03:
Hard to tell, but if you test and it works, then we will know ;)

hashoom8, 2020-12-03:
for example , this is a part of gcode

G0 X131.189 Y139.651
;TYPE:WALL-OUTER
G1 F3000 E862.03305
G1 F1350 X131.576 Y139.504 E862.04544
G1 X131.923 Y139.342 E862.05691
G1 X132.996 Y138.86 E862.09212
stop --->  G1 X134.091 Y138.424 E862.1274
G1 X135.204 Y138.036 E862.16268 <---

if i stop printing at that line , the printer will extrude ( E862.16268 ) that amount 

i'm not sure before or after end gcode

end gcode : 

G91 ;Relative positioning
G1 E-0.8 F2700 ;Retract a bit
G1 E-0.8 Z0.2 F2400 ;Retract and raise Z
G1 X5 Y5 F3000 ;Wipe out
G1 Z20 ;Raise Z more
G90 ;Absolute positionning

G1 X280 Y310 ;Present print
M106 S0 ;Turn-off fan
M104 S0 ;Turn-off hotend
M140 S0 ;Turn-off bed

M84 X Y E ;Disable all steppers but Z

M300 S440 P200 ; Play tone
M300 S660 P250 ; Play tone
M300 S880 P300 ; Play tone
M82 ;absolute extrusion mode
M104 S0
;End of Gcode

( copied from gcode file )

FormerLurker, 2020-12-03:
Hmm.  Do you have a cancel script in octoprint?

FormerLurker, 2020-12-03:
Also, the terminal output before and after you cancel would help

hashoom8, 2020-12-03:
Hmm. Do you have a cancel script in octoprint?
yes 

G91 ;Relative positioning
G1 E-1 F2700 ;Retract a bit
G1 E-1 Z0.2 F2400 ;Retract and raise Z
G1 X5 Y5 F3000 ;Wipe out
G1 Z20 ;Raise Z more
G90 ;Absolute positionning

G1 X300 Y310 ;Present print
M106 S0 ;Turn-off fan
M104 S0 ;Turn-off hotend
M140 S0 ;Turn-off bed
 
M84 X Y E ;Disable all steppers but Z

Also, the terminal output before and after you cancel would help
yes , whe i sow the extruder start pushing melted
Filament , i cut the power off

Send: N450 G1 X155.439 Y69.758 E239.4396*105
Recv: ok
Send: N451 G1 X156.781 Y69.35 E240.13939*96
Recv: ok
Send: N452 G1 X158.144 Y69.02 E240.83904*97
Recv: ok
Send: N453 G1 X159.524 Y68.767 E241.539*94
Recv: //action:cancel
Cancelling on request of the printer...
Changing monitoring state from "Printing" to "Cancelling"
Recv: //action:prompt_end
Recv: //action:prompt_begin UI Aborted
Recv: //action:prompt_button Dismiss
Recv: //action:prompt_show
Recv: //action:notification Print Aborted
Recv: ok
Send: N454 G1 X160.916 Y68.593 E242.23887*93
Recv: Not SD printing
Recv:  T:242.86 /0.00 B:84.89 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:242.86 /0.00 B:84.89 /0.00 @:0 B@:0
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N455 M108*46
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:242.26 /0.00 B:84.86 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: Not SD printing
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Recv:  T:241.67 /0.00 B:84.82 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:241.07 /0.00 B:84.71 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: Not SD printing
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Recv: ok
Send: N456 M117 Print Done*82
Recv: ok
Send: N457 M73 P100*80
Recv: ok
Send: N458 M105*46
Recv: //action:notification Print Done
Recv: ok
Send: N459 M73 P100*94
Recv: ok
Send: N460 G91*19
Recv: ok T:241.07 /0.00 B:84.64 /0.00 @:0 B@:0
[... 226 lines cut ...]
Recv:  T:229.76 /240.00 B:80.11 /0.00 @:127 B@:0
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:232.14 /240.00 B:80.00 /0.00 @:127 B@:0
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:235.71 /240.00 B:79.85 /0.00 @:32 B@:0
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:238.10 /240.00 B:79.70 /0.00 @:37 B@:0
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:239.29 /240.00 B:79.51 /0.00 @:88 B@:0
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:239.29 /240.00 B:79.39 /0.00 @:127 B@:0
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:241.67 /240.00 B:79.24 /0.00 @:4 B@:0
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:242.26 /240.00 B:79.13 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:241.67 /240.00 B:78.98 /0.00 @:9 B@:0

FormerLurker, 2020-12-03:
So your cancel script is changing axis modes, so it is definitely the cause.  Let me think about how to fix that...

hashoom8, 2020-12-03:
> So your cancel script is changing axis modes, so it is definitely the cause. Let me think about how to fix that...

i think octolaps causing one line delay 

so that line will extrude E242 ( 242 ) mm of melted
Filament

i test it 4 times and its from the plugin

maybe may octoprint is slow and octolaps plugin make it slower causing one line delay

FormerLurker, 2020-12-03:
I have an idea about what could be  happening.  Octolapse may be sending the line it was holding while taking a snapshot, if you cancelled during a snapshot..  did you?  If not I don't see how octolapse could be involved.  I will run some tests after I push out the emergency patch.  Also, I am wondering if the on print end script runs after cancel...

hashoom8, 2020-12-03:
>  if you cancelled during a snapshot.. did you? 

no

hashoom8, 2020-12-03:
i did another test and i got lucky 

terminal output : 

Send: N23434 G1 F1125 X91.748 Y224.469 E3394.64631*6
Recv: ok
Send: N23435 G0 F7200 X91.182 Y224.469*116
Recv: ok
Send: N23436 G1 F1125 X81.878 Y233.773 E3395.04018*13
Recv: ok
Send: N23437 G0 F7200 X81.878 Y233.208*124
Recv: ok
Send: N23438 G1 F1125 X90.617 Y224.469 E3395.41013*0
Recv: ok
Send: N23439 G0 F7200 X90.051 Y224.469*118
Recv:  T:230.36 /230.00 B:85.00 /85.00 @:11 B@:24
Recv: Not SD printing
Recv: ok
Send: N23440 G1 F1125 X81.878 Y232.642 E3395.75612*4
Recv: ok
Send: N23441 G0 F7200 X81.878 Y232.076*119
Recv: ok
Send: N23442 G1 F1125 X89.485 Y224.469 E3396.07815*3
Recv: Not SD printing
Recv: //action:cancel
Cancelling on request of the printer...
Changing monitoring state from "Printing" to "Cancelling"
Recv: //action:prompt_end
Recv: //action:prompt_begin UI Aborted
Recv: //action:prompt_button Dismiss
Recv: //action:prompt_show
Recv: //action:notification Print Aborted
Recv: ok
### **_Send: N23443 G0 F7200 X88.92 Y224.469*77  <---_**
Recv:  T:230.36 /230.00 B:84.96 /85.00 @:17 B@:67
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:230.36 /0.00 B:84.93 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: Not SD printing
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N23444 M108*47
Recv:  T:230.36 /0.00 B:84.93 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:227.98 /0.00 B:84.89 /0.00 @:0 B@:0
Recv: Not SD printing
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Recv: Not SD printing
Recv:  T:229.17 /0.00 B:84.86 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: Not SD printing
Recv: ok
Send: N23445 M117 Print Done*81

so  one line different will make the problem aging 

or if  i cancel 0.1 second earlier will make the problem aging 

so now i will print somthing and i will cancle but i will disable octolaps plugin

FormerLurker, 2020-12-03:
Try this as a workaround by adding these lines to your cancel script:

```
M117 Cancelling
```

I haven't tried this, but I suspect this will solve your issue for now.  It's going to take a while to figure out what's going on here.

FormerLurker, 2020-12-03:
Oh, make sure you add that to the VERY top of your script, so 

```
M117 Cancelling
G91
... rest of script here
```

hashoom8, 2020-12-03:
> Oh, make sure you add that to the VERY top of your script, so
> 
> ```
> M117 Cancelling
> G91
> ... rest of script here
> ```

where exactly ?

but octoprint cancel gcode , is way down

hashoom8, 2020-12-03:
Send: N450 G1 X155.439 Y69.758 E239.4396*105
Recv: ok
Send: N451 G1 X156.781 Y69.35 E240.13939*96
Recv: ok
Send: N452 G1 X158.144 Y69.02 E240.83904*97
Recv: ok
Send: N453 G1 X159.524 Y68.767 E241.539*94
Recv: //action:cancel
Cancelling on request of the printer...
Changing monitoring state from "Printing" to "Cancelling"
Recv: //action:prompt_end
Recv: //action:prompt_begin UI Aborted
Recv: //action:prompt_button Dismiss
Recv: //action:prompt_show
Recv: //action:notification Print Aborted
Recv: ok
### **_: N454 G1 X160.916 Y68.593 E242.23887*93_**
Recv: Not SD printing
Recv:  T:242.86 /0.00 B:84.89 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:242.86 /0.00 B:84.89 /0.00 @:0 B@:0
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N455 M108*46
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:242.26 /0.00 B:84.86 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: Not SD printing
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Recv:  T:241.67 /0.00 B:84.82 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:241.07 /0.00 B:84.71 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: Not SD printing
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Recv: ok
Send: N456 M117 Print Done*82
Recv: ok
Send: N457 M73 P100*80
Recv: ok
Send: N458 M105*46
Recv: //action:notification Print Done
Recv: ok
Send: N459 M73 P100*94
Recv: ok
### **_Send: N460 G91*19_** 
Recv: ok T:241.07 /0.00 B:84.64 /0.00 @:0 B@:0
Send: N461 G1 E-1 F2700*49
Recv: ok
Send: N462 G1 E-1 Z0.2 F2400*103
Recv: ok
Send: N463 G1 X5 Y5 F3000*77
Recv: ok
Send: N464 G1 Z20*86
Recv: ok
Send: N465 G90*23
Recv: ok
Send: N466 G1 X300 Y310*44
Recv: ok
Send: N467 M106 S0*98
Recv: ok
Recv: ok
Recv: ok
Send: N468 M104 S0*111
Recv: //action:notification Cheetah Ready.
Recv: ok
Send: N469 M140 S0*110
Recv: //action:notification Cheetah Ready.
[... 205 lines cut ...]
Recv:  T:229.76 /240.00 B:80.11 /0.00 @:127 B@:0
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:232.14 /240.00 B:80.00 /0.00 @:127 B@:0
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:235.71 /240.00 B:79.85 /0.00 @:32 B@:0
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:238.10 /240.00 B:79.70 /0.00 @:37 B@:0
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:239.29 /240.00 B:79.51 /0.00 @:88 B@:0
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:239.29 /240.00 B:79.39 /0.00 @:127 B@:0
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:241.67 /240.00 B:79.24 /0.00 @:4 B@:0
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:242.26 /240.00 B:79.13 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:241.67 /240.00 B:78.98 /0.00 @:9 B@:0

FormerLurker, 2020-12-03:
Well, in your example
```
: N454 G1 X160.916 Y68.593 E242.23887*93 the problem
```
shouldn't have caused a problem because your printer is still in absolute mode.  You should see a G91 being sent before this line, which would trigger the issue.  I only see one much later, so either that was not the problem line, or something else is going on.

Here is the problem line and the extrusion before that:

```
G1 X159.524 Y68.767 E241.53994
G1 X160.916 Y68.593 E242.23887
```

That should extrude 0.69893mm of filament UNLESS the E axis is somehow in relative mode.

Also, lots of communication errors in your terminal log, so not sure what's going on there.  

If you want you could edit your Octolapse printer profile, check the 'custom' profile box, and scroll to the bottom and uncheck this box:

[image]

That way Octolapse will NEVER change your axis mode EVER.  If your G90/G91 influences extruder setting is wrong (I believe it should be set to true) this will cause even worse problems.

hashoom8, 2020-12-03:
> If you want you could edit your Octolapse printer profile, check the 'custom' profile box, and scroll to the bottom and uncheck this box:
>
> [image]
>
> That way Octolapse will NEVER change your axis [...]

ok , i will try this tomorrow and update you , thx

hashoom8, 2020-12-04:
ok , this a test and octolaps is disabled

Recv: ok
Send: N16638 G0 X195.65 Y143.45*43
Recv:  T:222.02 /222.00 B:85.07 /85.00 @:61 B@:0
Recv: Not SD printing
Recv: Not SD printing
Recv: ok
Send: N16639 G1 F3000 E2990.43568*51
Recv: ok
Send: N16640 G0 F300 X195.65 Y143.45 Z4.3*34
Recv:  T:222.02 /222.00 B:85.03 /85.00 @:56 B@:80
Recv: Not SD printing
Recv: Not SD printing
Recv: ok
Send: N16641 G0 F9600 X155.564 Y143.836*72
Recv: ok
Send: N16642 G0 X154.194 Y145.664*39
Recv: ok
Send: N16643 G0 X152.295 Y146.395*42
Recv:  T:222.62 /222.00 B:85.03 /85.00 @:0 B@:104
Recv: Not SD printing
Recv: Not SD printing
Recv: ok
Send: N16644 G1 F3000 E2992.43568*59
Recv: ok
Send: N16645 G1 F2700 X153.462 Y147.562 E2992.52213*50
Recv:  T:222.02 /222.00 B:85.03 /85.00 @:13 B@:4
Recv: Not SD printing
Recv: ok
Send: N16646 G0 F9600 X153.461 Y147.744*67
Recv: Not SD printing
Recv: ok
Send: N16647 M117 ETL 00h43m03s*9
Recv: //action:notification ETL 00h43m03s
Recv: ok
Send: N16648 M73 P54*107
Recv: ok
Send: N16649 G1 F2700 X152.294 Y148.911 E2992.60859*50
Recv:  T:222.02 /222.00 B:85.03 /85.00 @:7 B@:94
Recv: Not SD printing
Recv: ok
Send: N16650 G0 F9600 X152.293 Y149.693*75
Recv: Not SD printing
Recv:  T:220.83 /222.00 B:85.07 /85.00 @:127 B@:7
Recv: Not SD printing
Recv: ok
Send: N16651 G1 F2700 X153.462 Y150.861 E2992.69515*54
Recv: ok
Send: N16652 G0 F9600 X153.461 Y151.043*65
Recv: Not SD printing
Recv:  T:220.83 /222.00 B:85.07 /85.00 @:89 B@:0
Recv: Not SD printing
Recv: ok
Send: N16653 G1 F2700 X152.294 Y152.211 E2992.78165*54
Recv: ok
Send: N16654 G0 F9600 X152.293 Y152.992*75
Recv: ok
Send: N16655 G1 F2700 X152.933 Y153.632 E2992.82906*58
Recv: Not SD printing
Recv: ok
Send: N16656 G0 F9600 X153.608 Y153.778*69
Recv: ok
Send: N16657 G1 F1350 X152.15 Y153.778 E2992.8727*55
Recv: ok
Send: N16658 G1 X152.15 Y146.25 E2993.09805*93
Recv: ok
Send: N16659 G1 X153.608 Y146.25 E2993.14169*104
Recv: ok
Send: N16660 G1 X153.608 Y153.778 E2993.36704*84
[... 282 lines cut ...]
Recv: Not SD printing
Recv:  T:196.59 /0.00 B:81.43 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:196.02 /0.00 B:81.29 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:195.45 /0.00 B:81.18 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:194.32 /0.00 B:81.11 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:193.75 /0.00 B:81.00 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:191.48 /0.00 B:80.82 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:192.05 /0.00 B:80.75 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:190.91 /0.00 B:80.61 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: Not SD printing

hashoom8, 2020-12-04:
now i will test it with octolaps enable and with your settings

> If you want you could edit your Octolapse printer profile, check the 'custom' profile box, and scroll to the bottom and uncheck this box:
>
> [image]
>
> That way Octolapse will NEVER change your axis [...]

hashoom8, 2020-12-05:
Recv: ok
Send: N490 G1 X132.735 Y148.019 E14.07861*86
Recv: ok
Send: N491 G1 X134.149 Y146.06 E14.13888*110
Recv: ok
Send: N492 G1 X135.581 Y143.956 E14.20237*91
Recv: ok
Send: N493 G1 X136.874 Y141.933 E14.26226*89
Recv: ok
Send: N494 G1 X138.149 Y139.809 E14.32406*81
Recv: ok
Send: N495 G1 X140.011 Y136.447 E14.41993*95
Recv: ok
Send: N496 G1 X140.087 Y136.309 E14.42386*83
Recv: ok
Send: N497 G1 X140.542 Y135.523 E14.44651*90
Recv: ok
Send: N498 G1 X141.112 Y134.697 E14.47154*89
Recv: ok
Send: N499 G1 X141.745 Y133.919 E14.49656*89
Recv: ok
Send: N500 G1 X142.148 Y133.484 E14.51136*81
Recv: ok
Send: N501 G0 F4800 X142.435 Y133.763*64
Recv:  T:76.44 /75.00 B:32.60 /30.00 @:44 B@:0
Recv: Not SD printing
Recv: ok
Send: N502 G1 F1200 X143.117 Y133.061 E14.53577*49
Recv: ok
Send: N503 G1 X143.511 Y132.67 E14.54962*111
Recv: ok
Send: N504 G1 X143.98 Y132.217 E14.56588*99
Recv: ok
Send: N505 G1 X144.718 Y131.56 E14.59053*98
Recv: ok
Send: N506 G1 X145.506 Y130.965 E14.61516*90
Recv: ok
Send: N507 G1 X146.745 Y130.211 E14.65134*85
Recv: ok
Send: N508 G1 X147.53 Y129.797 E14.67348*99
Recv: ok
Send: N509 G1 X148.011 Y129.553 E14.68693*93
Recv: ok
Send: N510 G1 X148.64 Y129.243 E14.70443*103
Recv: Not SD printing
Recv: ok
Send: N511 G1 X149.543 Y128.842 E14.72907*82
Recv: ok
Send: N512 G1 X150.475 Y128.516 E14.7537*108
Recv: ok
Send: N513 G1 X151.431 Y128.267 E14.77835*85
Recv: ok
Send: N514 G1 X151.949 Y128.165 E14.79152*87
Recv: ok
Send: N515 G1 X152.549 Y128.061 E14.80671*92
Recv: ok
Send: N516 G1 X152.904 Y128.005 E14.81567*93
Recv: ok
Send: N517 G1 X153.386 Y127.936 E14.82782*81
Recv: ok
Send: N518 G1 X154.803 Y127.817 E14.86329*92
Recv: ok
Send: N519 G1 X155.194 Y127.807 E14.87305*85
Recv: ok
Send: N520 G1 X155.506 Y127.803 E14.88083*86
Recv: ok
Send: N521 G1 X198.557 Y127.804 E15.95474*84
Recv: ok
Send: N522 G1 X199.274 Y127.815 E15.97263*82
Recv: ok
Send: N523 G1 X200.623 Y127.911 E16.00636*90
Recv: ok
Send: N524 G1 X200.946 Y127.952 E16.01449*93
Recv: ok
Send: N525 G1 X201.469 Y128.03 E16.02768*108
Recv: ok
Send: N526 G1 X202.205 Y128.156 E16.0463*107
Recv:  T:76.92 /75.00 B:32.60 /30.00 @:14 B@:0
Recv: Not SD printing
Recv: ok
Send: N527 G1 X203.171 Y128.362 E16.07094*85
Recv: ok
Send: N528 G1 X204.117 Y128.645 E16.09557*89
Recv: ok
Send: N529 G1 X205.037 Y129.004 E16.12021*86
Recv: //action:cancel
Cancelling on request of the printer...
Changing monitoring state from "Printing" to "Cancelling"
Recv: //action:prompt_end
Recv: //action:prompt_begin UI Aborted
Recv: //action:prompt_button Dismiss
Recv: //action:prompt_show
### **_Recv: //action:notification Print Aborted   no extra line_** 
Recv: ok
Send: N530 M108*44
Recv: Not SD printing
Recv:  T:77.88 /0.00 B:32.50 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:77.40 /0.00 B:32.50 /0.00 @:0 B@:0
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N531 M876 S0*110
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:77.40 /0.00 B:32.50 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:77.40 /0.00 B:32.60 /0.00 @:0 B@:0
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:77.88 /0.00 B:32.40 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: ok
Send: N532 M105*35
Recv: ok
Recv: ok
Send: N533 M117 Print Done*80
Send: N534 M73 P100*84
Recv: ok T:77.40 /0.00 B:32.50 /0.00 @:0 B@:0
Send: N535 M73 P100*85
Recv: //action:notification Print Done
Recv: ok
### **_Send: N536 M117 Cancelling*46 cancel script_** 
Recv: ok
Send: N537 G91*16
Recv: ok
Recv: //action:notification Cancelling
Recv: ok
Send: N538 G1 E-1 F2700*60
Recv: ok
Send: N539 G1 E-1 Z0.2 F2400*104
Recv: ok
Send: N540 G1 X5 Y5 F3000*77
Recv: ok
Send: N541 G1 Z20*80
Recv: ok
Send: N542 G90*19
Recv: ok
Send: N543 G1 X300 Y310*42
Recv: ok
Send: N544 M106 S0*98
Recv: ok
Send: N545 M104 S0*97
Recv: ok
Send: N546 M140 S0*98
Recv: //action:notification Cheetah Ready.
Recv: ok
Recv: //action:notification Cheetah Ready.
Send: N547 M84 X Y E*125
Recv: ok
Changing monitoring state from "Cancelling" to "Operational"
Recv: Not SD printing
Recv:  T:77.40 /0.00 B:32.50 /0.00 @:0 B@:0
Recv: Not SD printing

hashoom8, 2020-12-05:
ok it worked ,  this a test and octolaps was enabled 

i did a small test with no filament and no heat , i will do some more tests and update you , thx

hashoom8, 2020-12-05:
no , i think it did not work will yet

hashoom8, 2020-12-05:
i did the same test but i check this box 

> [image]

hashoom8, 2020-12-05:
Send: N340 G1 X225.469 Y182.715 E10.4267*101
Recv: ok
Send: N341 G1 X225.402 Y182.927 E10.43224*82
Recv: ok
Send: N342 G1 X225.05 Y184.003 E10.46048*102
Recv: ok
Send: N343 G1 X224.822 Y184.672 E10.47811*94
Recv: ok
Send: N344 G1 X224.461 Y185.608 E10.50314*86
Recv:  T:45.98 /40.00 B:36.75 /35.00 @:0 B@:0
Recv: Not SD printing
Recv: ok
Send: N345 G1 X224.026 Y186.513 E10.52819*94
Recv: ok
Send: N346 G1 X223.521 Y187.38 E10.55321*98
Recv: ok
Send: N347 G1 X222.948 Y188.204 E10.57825*94
Recv: ok
Send: N348 G1 X222.311 Y188.98 E10.60329*99
Recv: ok
Send: N349 G1 X221.615 Y189.703 E10.62833*86
Recv: ok
Send: N350 G1 X220.48 Y190.669 E10.66551*105
Recv: ok
Send: N351 G1 X219.652 Y191.291 E10.69134*85
Recv: ok
Send: N352 G1 X218.694 Y191.987 E10.72088*93
Recv: //action:cancel
Cancelling on request of the printer...
Changing monitoring state from "Printing" to "Cancelling"
### **_Recv: //action:notification Print Aborted_**
Recv: ok
Send: N353 M108*47
Recv: Not SD printing
Recv:  T:45.54 /0.00 B:36.75 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:45.98 /0.00 B:36.67 /0.00 @:0 B@:0
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N354 M105*37
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:45.54 /0.00 B:36.67 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:45.98 /0.00 B:36.58 /0.00 @:0 B@:0
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N355 M105*36
Recv: Not SD printing
Recv: ok
Recv: ok
Send: N356 M117 Print Done*85
Send: N357 M73 P100*87
Recv: ok T:45.09 /0.00 B:36.75 /0.00 @:0 B@:0
Send: N358 M73 P100*88
Recv: ok T:45.09 /0.00 B:36.75 /0.00 @:0 B@:0
Send: N359 M117 Cancelling*33
Recv: //action:notification Print Done
Recv: ok
Send: N360 G91*20
Recv: ok
Recv: ok
Send: N361 G1 E-1 F2700*54
Recv: //action:notification Cancelling
Recv: ok
Recv: ok
Send: N362 G1 E-1 Z0.2 F2400*96
Recv: ok
Recv: ok
Send: N363 G1 X5 Y5 F3000*74
Send: N364 G1 Z20*81
Recv: ok
Send: N365 G90*16
Recv: ok
Send: N366 G1 X300 Y310*43
Recv: ok
Recv: ok
Send: N367 M106 S0*101
Send: N368 M104 S0*104
Recv: ok
Send: N369 M140 S0*105
Recv: //action:notification Cheetah Ready.
Recv: ok
Recv: //action:notification Cheetah Ready.
Send: N370 M84 X Y E*127
Recv: ok
Changing monitoring state from "Cancelling" to "Operational"
Recv: Not SD printing
Recv:  T:45.09 /0.00 B:36.67 /0.00 @:0 B@:0
Recv: Not SD printing

hashoom8, 2020-12-05:
i have some prints i need to print them , and i will see if any thing filed and i will test on them .

hashoom8, 2020-12-05:
but is this safe to uncheck this box ?

> [image]

FormerLurker, 2020-12-05:
As long as your G90/G91 influences extruder setting is correct.  I haven't removed that warning yet because in general that shouldn't be necessary.  The 'checked' option has been working for 3 years, the 'unchecked' version has been around for only like 6 months in the wild, but I've been testing it for 1.5 years (it's what I use, is slightly faster).  Hopefully I didn't goof ;)  I would recommend you watch your first test print like a hawk and make sure you can cancel it in a hurry if it does anything scary.  If it's going to fail it will be somewhere between print start and the second layer, or perhaps at the end gcode.

I intended that warning to be scary, because only ppl who know what they are doing should use it.

hashoom8, 2020-12-05:
G90/G91 influences extruder is true

the box is unchecked  

and i will test it

hashoom8, 2020-12-05:
ok , there is a problem , it happened only if octolaps is enabled and when i stop from the printer lcd

Cancelling on request of the printer...

but if octolaps is disabled and i cancel from printer lcd , everything is ok ,

i dont know way , but i feel like its doing M600 command when i cancel .

hashoom8, 2020-12-05:
> G90/G91 influences extruder is true
> 
> the box is unchecked
> 
> and i will test it

Send: N114740 G1 X147.936 Y22.654 E1585.9198*108
Recv: ok
Send: N114741 G1 X148.197 Y22.722 E1585.92384*92
Recv: ok
Send: N114742 G1 X148.483 Y22.824 E1585.92839*91
Recv: ok
Send: N114743 G1 X148.76 Y22.971 E1585.93308*109
Recv: ok
Send: N114744 G1 X149.026 Y23.141 E1585.9378*96
Recv: ok
Send: N114745 G1 X149.306 Y23.388 E1585.94339*86
Recv: ok
Send: N114746 G1 X149.619 Y23.766 E1585.95074*81
Recv: ok
Send: N114747 G1 X149.824 Y24.135 E1585.95706*85
Recv: ok
Send: N114748 G1 X149.931 Y24.419 E1585.9616*97
Recv: ok
Send: N114749 G1 X150.029 Y24.845 E1585.96814*87
Recv: ok
Send: N114750 G1 X150.054 Y25.213 E1585.97366*82
Recv: ok
Send: N114751 G1 X150.06 Y30.388 E1586.05112*106
Recv: ok
Send: N114752 G1 X150.101 Y30.744 E1586.05648*83
Recv: ok
Send: N114753 G1 X150.224 Y31.098 E1586.06209*83
Recv: ok
Send: N114754 G1 X150.482 Y31.469 E1586.06885*90
Recv: Not SD printing
Recv: ok
Send: N114755 G1 X150.788 Y31.729 E1586.07486*91
Recv: ok
Send: N114756 G1 X151.303 Y32.073 E1586.08413*86
Recv: ok
Send: N114757 G1 X151.921 Y32.513 E1586.09549*81
Recv: //action:cancel
Cancelling on request of the printer...
Changing monitoring state from "Printing" to "Cancelling"
Recv: //action:prompt_end
Recv: //action:prompt_begin UI Aborted
Recv: //action:prompt_button Dismiss
Recv: //action:prompt_show
### **_Recv: //action:notification Print Aborted_**
Recv:  T:200.60 /200.00 B:79.96 /80.00 @:0 B@:32
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:200.60 /0.00 B:79.96 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: Not SD printing
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N114758 M108*20
Recv:  T:200.60 /0.00 B:79.92 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:200.00 /0.00 B:79.89 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: Not SD printing
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Recv:  T:199.43 /0.00 B:79.85 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: Not SD printing
Recv:  T:198.86 /0.00 B:79.70 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: ok
Send: N114759 M876 S0*86
Recv: ok
Send: N114760 M105*18
Recv: ok
Send: N114761 M117 Print Done*97
Recv: ok
Send: N114762 M73 P100*97
Recv: ok T:198.86 /0.00 B:79.70 /0.00 @:0 B@:0
Send: N114763 M73 P100*96
Recv: //action:notification Print Done
Recv: ok
Send: N114764 M117 Cancelling*31
Recv: ok
Send: N114765 G91*33
Recv: ok
Send: N114766 G1 E-1 F2700*1
Recv: //action:notification Cancelling
Recv: ok
Send: N114767 G1 E-1 Z0.2 F2400*85
Recv: ok
Send: N114768 G1 X5 Y5 F3000*113
Recv: ok
Recv: ok
Send: N114769 G1 Z20*108
Recv: ok
Send: N114770 G90*36
Recv: ok
Send: N114771 G1 X300 Y310*29
Recv: ok
Send: N114772 M106 S0*81
Recv: ok
Send: N114773 M104 S0*82
Send: N114774 M140 S0*85
Recv: ok
Recv: //action:notification Cheetah Ready.
Recv: ok
Send: N114775 M84 X Y E*74
Recv: //action:notification Cheetah Ready.
Recv: ok
Changing monitoring state from "Cancelling" to "Operational"
Recv: Not SD printing
Recv:  T:198.86 /0.00 B:79.66 /0.00 @:0 B@:0
Recv: Not SD printing
Recv: Not SD printing
Send: M112
Send: N114776 M112*19
Recv: Error:Printer halted. kill() called!
Changing monitoring state from "Operational" to "Error: Printer halted. kill() called!"
Send: N114777 M104 T0 S0*18
Send: N114778 M140 S0*89
Closing serial port due to emergency stop M112.
Send: M112
Send: N114779 M112*28
Send: N114780 M104 T0 S0*26
Changing monitoring state from "Error: Closing serial port due to emergency stop M112." to "Offline (Error: Closing serial port due to emergency stop M112.)"
Connection closed, closing down monitor

FormerLurker, 2021-01-03:
> Cancelling on request of the printer...

I don't think that is an issue.  This is pretty much what I would expect to happen, and I get this error if I cancel from my LCD too.