FormerLurker/Octolapse #640: Octolapse always opens helptext on any octolapse menu

KenwoodFox, 2020-12-03:
Im using Octolapse version 
Octolapse 0.4.0
Settings Version: 0.4.0

And Octoprint version Version 1.5.0 on a Pi 3 Model B Rev 1.2

Just as above, clicking anywhere in the menu will open a random helptext, tested with firefox, chrome, edge, light, internet explorer.

[image]

[image]

FormerLurker, 2020-12-03:
Please see the other issue like this that's open.  There is a resolution.

KenwoodFox, 2020-12-03:
I don't think you provided an issue number.

FormerLurker, 2020-12-03:
No, sorry, it's #637.  IT is pinned to the top of the issues now.

KenwoodFox, 2020-12-03:
Thanks so much!

closed 2020-12-03 by KenwoodFox

FormerLurker, 2020-12-03:
Anytime @KenwoodFox!

FormerLurker, 2020-12-05:
The release has dropped. I might have goofed the version number if you installed one of the pre-release versions, which will prevent the software update from notifying you. Not much has changed aside from some formatting fixes, an added wiki link on the tab, and an about page update, but if you want to update to the official stable release, just install from this release URL and everything will be good: https://github.com/FormerLurker/Octolapse/archive/v0.4.1.zip

Otherwise updates will pick back up starting from 0.4.2.

Thanks!

Closed :)