FormerLurker/Octolapse #641: Octolapse installs, but doesn't work

Jester4977, 2020-12-04:
I installed Octolapse v0.4.0 and it looks all screwy and the help boxes open when trying to enter settings.  Also when printing my prints always say "Starting" and never start.  Am I doing something wrong, or is this thing broken?

Octoprint 1.5.0
OctoPi Version 0.17.0
Raspberry Pi 4 Model B Rev 1.2 

Oh boy ....
[image]

_Originally posted by @Prutsium in https://github.com/FormerLurker/Octolapse/issues/633#issuecomment-734874803_

Jester4977, 2020-12-04:
Found on another issue that a release candidate has been posted for use with Octoprint 1.5.0.  I just loaded it and can confirm this solved the issues I was having.

https://github.com/FormerLurker/Octolapse/archive/v0.4.1rc1.zip

closed 2020-12-04 by Jester4977

FormerLurker, 2020-12-05:
The release has dropped. I might have goofed the version number if you installed one of the pre-release versions, which will prevent the software update from notifying you. Not much has changed aside from some formatting fixes, an added wiki link on the tab, and an about page update, but if you want to update to the official stable release, just install from this release URL and everything will be good: https://github.com/FormerLurker/Octolapse/archive/v0.4.1.zip

Otherwise updates will pick back up starting from 0.4.2.

Thanks!

Closed :)