FormerLurker/Octolapse #642: 0.4.1rc1 - Unable to enable Custom Image Preferences for Webcam

GJSchaller, 2020-12-04:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4.1rc1

#### Version of OctoPrint

OctoPrint Version: 1.5.1

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

1. Go into Octolapse interface
2. Edit Camera Profile
3. Click on "Enabled" for Custom Image Preferences

#### What should have happened?
Custom Preferences should have become available

#### What happened instead?
An error comes up on the right: "Unable To Enable Custom Preferences: An unexpected error was raised while testing custom image preferences for the 'Webcam - Default OctoPi 0.16.0' camera profile."

#### Operating System running OctoPrint and Octolapse

OS Name: Octopi
Os Version: 1.5.1

#### Browser and version of browser, operating system running browser

Browser: Chrome
Browser OS: macOS 11.0

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: https://pastebin.com/LsH7FEM4

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://pastebin.com/7SdXeSmB

#### Link to octoprint.log

Link to octoprint.log:  https://pastebin.com/C4y0ZPRu

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-12-04:
Please read the docs [here](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Enabling-Camera-Controls), or click on the help icon next to the custom image preferences checkbox.  You haven't performed the necessary setup.

GJSchaller, 2020-12-04:
Ooops, you're right - I did an uninstall & cleanup / reinstall when troubleshooting the layout issue, before moving to 0.4.1rc1.

My apologies!

FormerLurker, 2020-12-04:
Not a problem.  Enjoy!

closed 2020-12-04 by FormerLurker