FormerLurker/Octolapse #643: I can't download the timelapse

renegadecitizen, 2020-12-04:
I can't click on videos and images button to download the time-lapse
what am I doing wrong?

FormerLurker, 2020-12-04:
Does the tab look like this:

[image]

What version of Octoprint and Octolapse are using?  Did you recently upgrade to OctoPrint v1.5.0 or v1.5.1?

Do you have any logs?

Please fill out the issue template next time, else I'll just have to ask you for all that information anyway.

cmTera, 2020-12-04:
FYI I have the same issue, my octolapse looks like the one you linked, and I am unable to open the videos folder. Happened after upgrading to 1.5.0

FormerLurker, 2020-12-04:
Ok, @cmTera, in that case look at the stickied issue, or check out [this direct link to that issue](https://github.com/FormerLurker/Octolapse/issues/637), which contains a resolution.  Thank you for checking for duplicates before submitting an issue ;)  I've been getting so many reports on this issue that it's been difficult to get the patch ready.  I learned about this a day or two before 1.5.0 dropped, and assembled a fix immediately, but it takes time to get it tested and deployed.  There are way more people using 1.4.x at the moment, and I didn't want to accidentally break it for those folks by rushing something out.  Anyway, I wanted to explain what was going on, that it's being attended to, and that a release is imminent (in a few hours if things work out).

cmTera, 2020-12-04:
Thanks very much for the resources and info, I'll look into it right away.
Best of luck out there!

renegadecitizen, 2020-12-04:
sounds like your working on it
sorry for bothering you

FormerLurker, 2020-12-04:
@renegadecitizen, no problem.  Just check that link for the fix if you want it ASAP, else I'm working on the master release now.

renegadecitizen, 2020-12-05:
Awesome thanks

FormerLurker, 2020-12-05:
The release has dropped.  I might have goofed the version number if you installed one of the pre-release versions, which will prevent the software update from notifying you.  Not much has changed aside from some formatting fixes, an added wiki link on the tab, and an about page update, but if you want to update to the official stable release, just install from this release URL and everything will be good: ```https://github.com/FormerLurker/Octolapse/archive/v0.4.1.zip```

Thanks!

Closed :)

closed 2020-12-05 by FormerLurker