FormerLurker/Octolapse #644: Rendering failed without error being shown

leopshef, 2020-12-05:
Octolapse Version: Octolapse 0.4.1

OctoPrint Version: OctoPrint 1.5.1 

Diagnostic Logging was Enabled: No

#### What were you doing when the problem occurred

1. Started a print with octolapse enabled
2. The print completed as normal, the preview window filled as it should during the print.
3. After the print, no rendering was created and no error was shown in the window.
4.  The "incomplete renderings" button was not shown.

#### What should have happened?
Rendering should have been created as normal.

#### What happened instead?
Rendering was not created.  Incomplete renderings button was not shown.  No errors were shown.

#### Operating System running OctoPrint and Octolapse

OS Name: Raspbian
Os Version: 5.4.51-v7l+

Link to settings.json with all passwords removed: https://textuploader.com/1ewce

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://pastebin.com/0dGiYdgY

#### Link to octoprint.log

Link to octoprint.log: https://pastebin.com/P66WXYbE

mirokymac, 2020-12-09:
I have the same issue here.

Octolapse Version: Octolapse 0.4.1
OctoPrint Version: OctoPrint 1.5.0

Since no error in the web page, I captured this  in terminal:
`
2020-12-0913:12:13,478-octoprint-ERROR-Exceptionon/plugin/octolapse/loadFailedRenderings[POST]
Traceback(mostrecentcalllast):
File"/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/flask/app.py",line2447,inwsgi_app
response=self.full_dispatch_request()
File"/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/flask/app.py",line1952,infull_dispatch_request
rv=self.handle_user_exception(e)
File"/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/flask/app.py",line1821,inhandle_user_exception
reraise(exc_type,exc_value,tb)
File"/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/flask/_compat.py",line39,inreraise
raisevalue
File"/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/flask/app.py",line1950,infull_dispatch_request
rv=self.dispatch_request()
File"/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/flask/app.py",line1936,indispatch_request
returnself.view_functions[rule.endpoint](**req.view_args)
File"/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/octoprint/server/util/flask.py",line1516,indecorated_view
returnno_firstrun_access(flask_login.login_required(func))(*args,**kwargs)
File"/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/octoprint/server/util/flask.py",line1539,indecorated_view
returnfunc(*args,**kwargs)
File"/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/flask_login/utils.py",line272,indecorated_view
returnfunc(*args,**kwargs)
File"/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/octoprint_octolapse/__init__.py",line1719,inload_failed_renderings_request
failed_jobs=self._rendering_processor.get_failed()
AttributeError:'NoneType'objecthasnoattribute'get_failed'
2020-12-0913:12:13,490-tornado.access-ERROR-500POST/plugin/octolapse/loadFailedRenderings(::ffff:192.168.6.6)84.31ms
2020-12-0913:12:13,501-octoprint-ERROR-Exceptionon/plugin/octolapse/loadInProcessRenderings[POST]
Traceback(mostrecentcalllast):
File"/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/flask/app.py",line2447,inwsgi_app
response=self.full_dispatch_request()
File"/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/flask/app.py",line1952,infull_dispatch_request
rv=self.handle_user_exception(e)
File"/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/flask/app.py",line1821,inhandle_user_exception
reraise(exc_type,exc_value,tb)
File"/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/flask/_compat.py",line39,inreraise
raisevalue
File"/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/flask/app.py",line1950,infull_dispatch_request
rv=self.dispatch_request()
File"/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/flask/app.py",line1936,indispatch_request
returnself.view_functions[rule.endpoint](**req.view_args)
File"/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/octoprint/server/util/flask.py",line1516,indecorated_view
returnno_firstrun_access(flask_login.login_required(func))(*args,**kwargs)
File"/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/octoprint/server/util/flask.py",line1539,indecorated_view
returnfunc(*args,**kwargs)
File"/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/flask_login/utils.py",line272,indecorated_view
returnfunc(*args,**kwargs)
File"/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/octoprint_octolapse/__init__.py",line1731,inload_in_process_renderings_request
in_process_jobs=self._rendering_processor.get_in_process()
AttributeError:'NoneType'objecthasnoattribute'get_in_process'
2020-12-0913:12:13,514-tornado.access-ERROR-500POST/plugin/octolapse/loadInProcessRenderings(::ffff:192.168.6.6)103.97ms
2020-12-0913:13:55,732-octoprint-ERROR-Exceptionon/plugin/octolapse/rendering/font[GET]
Traceback(mostrecentcalllast):
File"/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/flask/app.py",line2447,inwsgi_app
response=self.full_dispatch_request()
File"/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/flask/app.py",line1952,infull_dispatch_request
rv=self.handle_user_exception(e)
File"/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/flask/app.py",line1821,inhandle_user_exception
reraise(exc_type,exc_value,tb)
File"/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/flask/_compat.py",line39,inreraise
raisevalue
File"/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/flask/app.py",line1950,infull_dispatch_request
rv=self.dispatch_request()
File"/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/flask/app.py",line1936,indispatch_request
returnself.view_functions[rule.endpoint](**req.view_args)
File"/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/octoprint/server/util/flask.py",line1516,indecorated_view
returnno_firstrun_access(flask_login.login_required(func))(*args,**kwargs)
File"/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/octoprint/server/util/flask.py",line1539,indecorated_view
returnfunc(*args,**kwargs)
File"/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/flask_login/utils.py",line272,indecorated_view
returnfunc(*args,**kwargs)
File"/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/octoprint_octolapse/__init__.py",line1550,inget_available_fonts
[... cut ...]
withPopen(*popenargs,**kwargs)asprocess:
File"/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/subprocess.py",line854,in__init__
self._execute_child(args,executable,preexec_fn,close_fds,
File"/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/subprocess.py",line1702,in_execute_child
raisechild_exception_type(errno_num,err_msg,err_filename)
FileNotFoundError:[Errno2]Nosuchfileordirectory:'fc-list'
2020-12-0913:13:55,748-tornado.access-ERROR-500GET/plugin/octolapse/rendering/font(::ffff:192.168.6.6)65.52ms
`

FormerLurker, 2020-12-09:
@mirokymac, that log is pretty much unreadable :)

However, it's probably a python 2 issue with the rendering overlay.  Edit your rendering profile, delete the overlay text, reboot, and things should start working again.

I'm going to try to fix this ASAP.

mirokymac, 2020-12-10:
> @mirokymac, that log is pretty much unreadable :)
> 
> However, it's probably a python 2 issue with the rendering overlay. Edit your rendering profile, delete the overlay text, reboot, and things should start working again.
> 
> I'm going to try to fix this ASAP.

`
2020-12-10 19:43:21,570 - octoprint - ERROR - Exception on /plugin/octolapse/loadFailedRenderings [POST]
Traceback (most recent call last):
  File "/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/flask/app.py", line 1952, in full_dispatch_r
equest
    rv = self.handle_user_exception(e)
  File "/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/flask/app.py", line 1821, in handle_user_exc
eption
    reraise(exc_type, exc_value, tb)
  File "/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/flask/app.py", line 1950, in full_dispatch_r
equest
    rv = self.dispatch_request()
  File "/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/flask/app.py", line 1936, in dispatch_reques
t
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/octoprint/server/util/flask.py", line 1516,
in decorated_view
    return no_firstrun_access(flask_login.login_required(func))(*args, **kwargs)
  File "/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/octoprint/server/util/flask.py", line 1539,
in decorated_view
    return func(*args, **kwargs)
  File "/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/flask_login/utils.py", line 272, in decorate
d_view
    return func(*args, **kwargs)
  File "/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/octoprint_octolapse/__init__.py", line 1719,
 in load_failed_renderings_request
    failed_jobs = self._rendering_processor.get_failed()
AttributeError: 'NoneType' object has no attribute 'get_failed'
2020-12-10 19:43:21,583 - tornado.access - ERROR - 500 POST /plugin/octolapse/loadFailedRenderings (::ffff:192.168
.6.6) 96.52ms
2020-12-10 19:43:21,593 - octoprint - ERROR - Exception on /plugin/octolapse/loadInProcessRenderings [POST]
Traceback (most recent call last):
  File "/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/flask/app.py", line 1952, in full_dispatch_r
equest
    rv = self.handle_user_exception(e)
  File "/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/flask/app.py", line 1821, in handle_user_exc
eption
    reraise(exc_type, exc_value, tb)
  File "/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/flask/app.py", line 1950, in full_dispatch_r
equest
    rv = self.dispatch_request()
  File "/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/flask/app.py", line 1936, in dispatch_reques
t
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/octoprint/server/util/flask.py", line 1516,
in decorated_view
    return no_firstrun_access(flask_login.login_required(func))(*args, **kwargs)
  File "/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/octoprint/server/util/flask.py", line 1539,
in decorated_view
    return func(*args, **kwargs)
  File "/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/flask_login/utils.py", line 272, in decorate
d_view
    return func(*args, **kwargs)
  File "/home/atlas/.pyenv/versions/3.8.2/lib/python3.8/site-packages/octoprint_octolapse/__init__.py", line 1731,
 in load_in_process_renderings_request
    in_process_jobs = self._rendering_processor.get_in_process()
AttributeError: 'NoneType' object has no attribute 'get_in_process'
`
Maybe this is better.

FormerLurker, 2021-01-03:
Excellent!  Glad it is working.  There are apparently other issues with fontconfig, but glad you aren't having those.  Marking this closed :)

closed 2021-01-03 by FormerLurker