FormerLurker/Octolapse #648: Custom Imagine Settings not working

eblieb, 2020-12-11:
When I attempt to customize the image settings of my RPi Camera v1.3 I keep getting 

Unable To Enable Custom Preferences
An unexpected error was raised while testing custom image preferences for the 'Webcam - Default OctoPi 0.16.0' camera profile.

#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: ___REPLACE_THIS__OCTOLAPSE_VERSION_GOES_HERE

#### Version of OctoPrint

OctoPrint Version: ___REPLACE_THIS__OCTOPRINT_VERSION_GOES_HERE

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: ___REPLACE_THIS__YES_OR_NO

#### What were you doing when the problem occurred

1. ___REPLACE_THIS__STEP_ONE_GOES_HERE
2. ___REPLACE_THIS__STEP_TWO_GOES_HERE
3. ___REPLACE_THIS__STEP_...

#### What should have happened?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: ___REPLACE_THIS__OS_NAME_GOES_HERE
Os Version: ___REPLACE_THIS__OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

eblieb, 2020-12-11:
I pulled the log and this is what I am getting 
2020-12-10 19:26:54,032 - octolapse.camera - ERROR - An unexpected error was raised while testing custom image preferences for the 'Webcam - Default OctoPi 0.16.0' camera profile.
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octolapse/camera.py", line 450, in _test_mjpg_streamer_control
    raise CameraError("mjpg_streamer_control_error", message)
CameraError: mjpg_streamer_control_error: The server denied access to the mjpg-streamer control.htm for the 'Webcam - Default OctoPi 0.16.0' camera profile.  <a href="https://github.com/FormerLurker/Octolapse/wiki/V0.4---Enabling-Camera-Controls" target = "_blank">Please see this link to correct this error.</a>, or disable the 'Enable And Apply Preferences at Startup' and 'Enable And Apply Preferences Before Print' options.

FormerLurker, 2020-12-11:
Are you running octoprint anywhere or the spaghetti detective?  If so, they have a compatibility mode that should fix this.  Also, did you follow the link in the log you posted?

eblieb, 2020-12-11:
So there are 2 sets of instructions that are different
https://github.com/FormerLurker/Octolapse/wiki/V0.4---Configuring-a-Raspberry-Pi-Camera
and 
https://github.com/FormerLurker/Octolapse/wiki/V0.4---Enabling-Camera-Controls

They have different instructions

FormerLurker, 2020-12-11:
Thats right.  One is for a raspberry pi cam.  One is for a usb cam.  However, the error says you aren't using mjpgstreamer.  Is that true?  How are you streaming, and what camera are you using?

eblieb, 2020-12-11:
Got it working. I am using a Raspberry Pi Camera v1.3 on the camera ribbon connection. 

"V0.4 Configuring a Raspberry Pi Camera" - This is a misleading title because it is setting up a USB Camera on a Raspberry Pi. Not a Raspberry Pi Camera.

FormerLurker, 2020-12-11:
Sorry of that was misleading, but that is what the camera is called.  I wish it were more clear.  Do you have a suggestion for improvement?

eblieb, 2020-12-11:
One problem is both of the instructions show a Logitech C920 being installed, which makes it confusing if this is actually a Raspberry Pi Camera or a USB camera. I guess having a set of instructions for the Raspberry Pi v1, v1.3 and v2.0 camera.. Or Raspberry Pi Camera via Ribbon Connector.

eblieb, 2020-12-11:
Oh I did notice another issue, not sure if you want me to open a new issue for it.
But when you check Image Stabilization it zooms into a section of the image (which is expected) but when you uncheck it, it doesnt zoom back out.

FormerLurker, 2020-12-11:
Yeah, that's a known issue with the driver.  Long story short, image stabilization is bad for a timelapse.  Don't turn it on.  

Glad you got things working!  I will try to make the docs more clear.  I've been spending most of my time recent working on the arc welder plugin, which you may want to check out, but hope to get to some octolapse stuff again soon.  One of the things I would like to do is create a custom streaming server that makes webcam controls and streaming better and easier.  It just takes time and effort.

eblieb, 2020-12-11:
Yup been using ArcWelder for the past few days and love it. Only issue with it is the conflict with PrintTimeGenius where they both analyze at the same time, then arcwelder replaces the original file and PrintTimeGenius has to reanalyze the file agian.

FormerLurker, 2020-12-11:
Interesting that you mentioned that.  I was just looking to see of stopping the analysis queue was possible.  Apparently it is!

FormerLurker, 2020-12-11:
Hey, I just implemented removing the source file from the analysis queue IF it will be overwritten within Arc Welder.  You can try it by installing via the plugin manager by clicking get more and pasting this into the 'from url...' box:  https://github.com/FormerLurker/ArcWelderPlugin/archive/8754b74bb5255eb52e67bd8e04e95404bba4b61d.zip

let me know what you think.

eblieb, 2020-12-11:
Will do! Right now I am trying to tune out the elephant foot from my prints :( Not sure what else to try to tune it out but guess I can mess with bed temp a little more

FormerLurker, 2020-12-11:
Sometimes that is caused by the nozzle being too low, or filament diameter issues.  Sometimes not.  Definitely good luck with that though, and I look forward to hearing about your results!

FormerLurker, 2020-12-11:
> Was much faster than previously

There are two reasons for this:

1.  You mentioned, no analysis running on the same thread
2.  There was a bug in the C++ logging code.  It sends info back to python when logging things, and it turns out all loggers were always enabled within C++, but they weren't getting outputted from python because the module level loggers were/might have been off.  Though c++ is really fast when creating the logging strings, calling back into python is painfully slow, so this was probably the bulk of the speedup.

eblieb, 2020-12-11:
So i replaced the RPi camera with an Amcrest USB cable. It is working but I have it mounted upside down and for some reason cant get it to rotate 180 degrees. Also I keep getting this notification everytime I open octolapse 

New Image Preferences Available
Octolapse has detected new camera preferences from your streaming server. Do you want to overwrite your current image preferences?

FormerLurker, 2020-12-11:
> It is working but I have it mounted upside down and for some reason cant get it to rotate 180 degrees.

[image]

> Octolapse has detected new camera preferences from your streaming server. Do you want to overwrite your current image preferences?

I will look into that.  It should not keep asking you.  OUt of curiosity, what happens if you click yes?

eblieb, 2020-12-11:
It changed all the settings and didn't ask me again. It turns out it rotated the video fine in the control tab, but in the octolapse preview tab it is still rotated the wrong way.

FormerLurker, 2020-12-11:
> It turns out it rotated the video fine in the control tab, but in the octolapse preview tab it is still rotated the wrong way.

The control tab is different since it just has to adjust the visuals.  Octolapse actually must physically rotates the images.  Was the timelapse in the correct orientation when you set the Octolapse Camera profile to rotate 180 degrees?

FormerLurker, 2021-01-03:
I'm closing this out since it has not been updated in a while.  Let me know if you want me to reopen it.  Thanks!

closed 2021-01-03 by FormerLurker

eblieb, 2021-02-25:
Hey can you push Arcwelder RC2 to final?

FormerLurker, 2021-02-25:
I'm in the process of doing that.  It took a while to get enough feedback so that I felt confident it was working.  Since the initial release I've printed maybe 50 hours of files with the newest version, and have gotten enough feedback from both the plugin and the console version.

Fyi, this is the Octolapse repo :)