FormerLurker/Octolapse #650: Octolapse stops after first of multiple objects finishes!

hashoom8, 2020-12-13:
When I use Cura's "One at a Time" feature, Octolapse stops taking pictures after the first object finishes printing. I would love to get a timelapse that shows each of the parts growing from the print bed.

FormerLurker, 2020-12-14:
This has been requested before, and is quite dangerous.  If I remove the z height restriction (no snapshot allowed on layers already completed), Octolapse will happily ram the print head into the parts already printed.

I have been thinking about a way that would cause Octolapse to raise Z up to the highest layer printed during a snapshot.  That would take quite a while since Z is usually really slow.   Anyway, i will think about that.

hashoom8, 2020-12-14:
first, the highest object can't go over 25 mm because the x axis  

second when i loud the objects in cura , i start from the lowest to the highest and from front to the back of the printer , so the lowest is in the front

FormerLurker, 2021-01-03:
@hashoom8, sorry it took me a while to get back.

> second when i loud the objects in cura , i start from the lowest to the highest and from front to the back of the printer , so the lowest is in the front

That may be true, the if your stabilization location is in the wrong spot, bad things would happen.  Unfortunately adding an enhancement like this would cause issues for the average user, so it would create lots of headaches for me if I allowed this.  It would be pretty cool to see though.  One could fork the repository and add this feature to a custom version of Octolapse if one were so inclined.

Happy new year, and thank you for your post!

closed 2021-01-03 by FormerLurker