FormerLurker/Octolapse #652: [Request] How to run octolapse on the TV box

dgpower, 2020-12-17:
如何在电视盒子上运行octolapse?

我已经用电视盒子安装了Octoprint、MJPG-STREAMER和octolapse插件。已经确认mjpg-streamer可以正常工作（因为我在Octoprint页面上看到了图像）。但是octolapse检测不到摄像头。

请问，我该如何配置才能让octolapse正常工作？
-----------------------------------------
I have installed Octoprint, MJPG-STREAMER and octolapse plug-ins with the TV box. It has been confirmed that mjpg-streamer can work normally (because I saw the image on the Octoprint page). But octolapse cannot detect the camera.

Excuse me, how can I configure to make octolapse work normally?

FormerLurker, 2020-12-17:
Please post any and all error messages you saw, as well as the plugin.octolapse.log file.  Thanks!

dgpower, 2020-12-27:
当我升级到最新版本的octolapse，一切正常了。非常感谢！
--------------------------------------------------------------
When I upgraded to the latest version of octolapse, everything worked fine. thank you very much!

closed 2020-12-27 by dgpower