FormerLurker/Octolapse #653: CR-10 Max Profile missing

WimVDK, 2020-12-19:
i'm hoping to use this plugin to create awesome timelapses. Now the thing is, there's plenty CR-10 profiles in the list but not one for the CR-10 Max. So which one do i pick for the optimum result?

FormerLurker, 2021-01-03:
@WimVDK,

I haven't forgotten about you, sorry for the delay!  I'm working on a few major issues, then will swing around to this.  Bug me if you don't hear back in a couple weeks please!

WimVDK, 2021-01-03:
yeah no worries. It was only shortly before the holiday season i got this thing.

WimVDK, 2021-03-29:
It's been more than a few weeks now :) Figured i'd bug you @FormerLurker

WimVDK, 2021-05-17:
bumpie?

FormerLurker, 2021-06-01:
Yes, I added this to the milestones, but i will have the profile complete well before release.  I'll create something that may work, will post it here, and you can import it and tell me if it works.  If there are issues, we will go back and forth until it is perfected.  I'll try to get to this today or tomorrow after work.  Thanks for the reminder.

WimVDK, 2021-08-08:
sorry for the late response. I don't check github much and i didn't get notified. I see there's no official 0.5.0 release  yet. Any way to access it so i can test it?

Omranello, 2021-09-09:
Hi all , same issue am facing with CR10Max as the profile is not there,  tried other profiles but didn't work properly... any help if there is a specific way to add a CR10Max profile.

Thanks in advance

FormerLurker, 2021-10-08:
Wanted to drop a quick message here to say I'm looking at adding several new profiles, including this one.  Still on the road map, and I will finally have a few weeks free to start working on this again!

WimVDK, 2021-10-31:
good to know but i'll wait for the machine specific one

silentFred, 2022-06-19:
@Omranello have you made more progress on your profile since your post? Also keen to get my CR10Max lapsed up 😂 Ender3v2 works like a dream but I'm super keen on larger pieces and my 1mm nozzle!

Omranello, 2022-06-19:
@silentFred dropped it altogether and used my phone to record,  am building my Voron 2.4 so not really giving the laps much attention.
I stopped taking laps after I started to get more issues with print quality ,,, been a while not using it so I can't pinpoint the exact issue though atm ... sorry.
Please let me know if you got anything .. would love to go back to the laps

silentFred, 2022-06-20:
@Omranello - sorry to hear than! What was your main quality concern? extra extrusion while the snap is taken?

Will keep in touch if i make progress on the CR10MAX. Perhaps using my Ender3 for social content is as far as I go. Not to keen to risk quality on a 5 day print

WimVDK, 2023-01-28:
it's been just over 2 years since my initial request. Plugin is still on 0.4.1 missing CR-10 Max support...