FormerLurker/Octolapse #654: No overlay options under rendering

cbarth3, 2020-12-23:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4.1

#### Version of OctoPrint

OctoPrint Version: 1.5.2

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: yes

#### What were you doing when the problem occurred

1. Customize rendering profile
2. Try to setup overlay options
3. If you are using Linux, you may not have font-config installed.

#### What should have happened?
available options for rendering overlay

#### What happened instead?
If you are using Linux, you may not have font-config installed

#### Operating System running OctoPrint and Octolapse

OS Name: octopi 0.17
Os Version: 1.5.2

#### Printer model & used firmware incl. version

Printer Model: Ender 3 pro
Printer Firmware Version: Big tree tech e3 v2 marlin 2.09

#### Browser and version of browser, operating system running browser

Browser: firefox
Browser OS: windows 10E

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://gist.github.com/cbarth3/7d9e7eccea7ca8bc225b0b3545a789a2

#### Link to octoprint.log

Link to octoprint.log:  https://gist.github.com/cbarth3/e473b270605e1f5cb517fd7e35d1ec34

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: [image]

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-12-28:
Just install font-config and it should be fine.  However, there should be one default font that comes with Octolapse.  I will look into this.

cbarth3, 2020-12-29:
Went through and tried to install/update font-config and got the following message:

sudo apt-get install fontconfig
Reading package lists... Done
Building dependency tree
Reading state information... Done
fontconfig is already the newest version (2.13.1-2).
0 upgraded, 0 newly installed, 0 to remove and 30 not upgraded.

FormerLurker, 2020-12-29:
Ok, so this is something else...  Give me a day or two and I will come up with some advanced logging to figure this out

FormerLurker, 2020-12-29:
Actually, it may be that your current logs are enough.  Will review those asap.

rogertetzlaff, 2021-01-03:
Replicated this issue on my install. At shell, font-config is installed and fc-list (as pi user) shows a number of fonts I installed in /usr/share/fonts/truetype/ . However even after restart, Octolapse still reports no fonts installed. Client browser: Chrome/Safari/Edge on MacOS; Server is OctoPrint Version 1.5.2 on Raspberry Pi 3 (octopi image 0.15.1), Octolapse 0.41

FormerLurker, 2021-01-03:
OK, so after a long time (sorry about that), I've attempted to replicate your setup:

[image]

I'm a bit baffled:

[image]

And there were no errors in your log unfortunately.  I have added some additional logging as well as some minor changes to the 'get_system_fonts' call (untested in python 2.7, fair warning) that you can install via the plugin manager (get more->From url...) with this url: ```https://github.com/FormerLurker/Octolapse/archive/514baf3280ad54596a4c15e3ea71cddcf0dc3e98.zip```

Once that is installed and after you reboot, please follow these steps to generate a log:

1. Edit your current logging profile and click 'clear all logs'.
2. Change your current logging profile to 'Log Everything'.
3. Open your current rendering profile.
4. Ensure the 'Customize Profile' box is checked.
5. See if any fonts are listed and close the profile.
6. Change your logging profile back to 'Log All Errors' (this is important)
7. Edit the 'Log All Errors' profile and click the 'Download Log' button and post it here.

Thanks!

FormerLurker, 2021-01-03:
OK, as I warned, there were issues with python 2.7, lol!  Here is a fix for this (tested in both 2.7 and 3.7):

```
https://github.com/FormerLurker/Octolapse/archive/8a5c0a566350427893db5a0b3859303a871b2525.zip
```

I think the issue is that the subprocess module returns different encoding if any unicode exists in the system fonts.  This is super strange to me, but I've added some code to check the type of data returned and to decode if necessary.  I tested this by manually creating a font called DejaVuSerif¢.tff.  That's probably why not everyone was having an issue with the overlay section.

cbarth3, 2021-01-07:
So on printer 1 I decided to finally upgrade to python 3 and it is working before trying this fix. Printer 2 I had not used octolapse yet so I installed it brand new and saw the same issue no fonts. Installed the fix version from above and it is working! So printer 2 is no on version v0.4.1+u.8a5c0a5 and working.

FormerLurker, 2021-01-07:
Awesome, thank you for testing both versions!  That makes me pretty confident this fix works

ptforsberg, 2021-02-10:
FWIW, I was having the same issue, but installing the referenced plugin it worked.

`pi@octopi:~ $ fc-list --format %{file}\\n
/usr/share/fonts/truetype/dejavu/DejaVuSans-Bold.ttf
/home/pi/.local/share/fonts/digital-7 (italic).ttf
/home/pi/.local/share/fonts/digital-7 (mono).ttf
/usr/share/fonts/truetype/dejavu/DejaVuSans.ttf
/home/pi/.local/share/fonts/digital-7 (mono italic).ttf
/usr/share/fonts/truetype/dejavu/DejaVuSansMono-Bold.ttf
/usr/share/fonts/truetype/dejavu/DejaVuSerif-Bold.ttf
/usr/share/fonts/truetype/dejavu/DejaVuSerif.ttf
/home/pi/.local/share/fonts/digital-7.ttf
/usr/share/fonts/truetype/dejavu/DejaVuSansMono.ttf
pi@octopi:~ $`

FormerLurker, 2021-02-10:
@ptforsberg, Excellent, thanks for reporting success!