FormerLurker/Octolapse #655: Print preview omitting Y and Z components of snapshot planned positions

tech2077, 2020-12-27:
#### Version of Octolapse
 
Octolapse Version: 0.4.1

#### Version of OctoPrint

OctoPrint Version: 1.5.2

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

1. Install OctoLapse
2. Setup printer profile
3. Select a smart trigger
4. Start print with print plan enabled
5. Print plan preview modal opens

#### What should have happened?
Coordinates for each snapshot should be displayed with X, Y, and Z components

#### What happened instead?
Only the X component of the position is displayed

#### Operating System running OctoPrint and Octolapse

OS Name: Linux (OctoPi)
Os Version: 0.17.0 (Python 3)

#### Browser and version of browser, operating system running browser

Browser: All (tested with latest Firefox, Chrome, and Safari)
Browser OS: MacOS

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: Occurs on all gcode

#### Link to settings.json

Link to settings.json with all passwords removed: https://gist.github.com/tech2077/4d4fecb59194f8bf16f652b52b5b28bb

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://gist.github.com/tech2077/4d4fecb59194f8bf16f652b52b5b28bb

#### Link to octoprint.log

Link to octoprint.log:  https://gist.github.com/tech2077/4d4fecb59194f8bf16f652b52b5b28bb

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: 
[image]

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

tech2077, 2020-12-27:
Not sure if it's rendering correctly on other systems, but the solution for me appears to be to change the span tags from self closed to a regular tag in `templates/octolapse_snapshot_plan.jinja2`. I can't find any documentation on the behavior of non-typical self closed tags specific to jinja, but I suspect it's an HTML vs XHTML spec thing.

```
<strong>Planned Positions</strong>
            <div class="row-fluid">
                <div class="span3 text-right no-min-height">
                    Initial:&nbsp;
                </div>
                <div class="span9 text-left no-min-height">
                    <strong>X</strong><span data-bind="text: SnapshotPlanState.format_coordinates(SnapshotPlanState.x_initial())" />,&nbsp;<strong>Y</strong><span data-bind="text: SnapshotPlanState.format_coordinates(SnapshotPlanState.y_initial())" />,&nbsp;<strong>Z</strong><span data-bind="text: SnapshotPlanState.format_coordinates(SnapshotPlanState.z_initial())" />
                </div>
            </div>
```

to

```
<strong>Planned Positions</strong>
            <div class="row-fluid">
                <div class="span3 text-right no-min-height">
                    Initial:&nbsp;
                </div>
                <div class="span9 text-left no-min-height">
                    <strong>X</strong><span data-bind="text: SnapshotPlanState.format_coordinates(SnapshotPlanState.x_initial())"></span>,&nbsp;<strong>Y</strong><span data-bind="text: SnapshotPlanState.format_coordinates(SnapshotPlanState.y_initial())"></span>,&nbsp;<strong>Z</strong><span data-bind="text: SnapshotPlanState.format_coordinates(SnapshotPlanState.z_initial())"></span>
                </div>
            </div>
```

Screenshot with line changed:
[image]

FormerLurker, 2021-01-03:
Ok, I think I have solved this.  Can you try installing from this link:  ```https://github.com/FormerLurker/Octolapse/archive/b83eb91437be825a1eb14a31c65fadb4d2859513.zip```

Let me know if that solves it, and thanks for reporting!

tech2077, 2021-01-03:
Looks like that fixed it. Only notes is there's an extra space before each comma on the "Initial" row:

[image]

FormerLurker, 2021-01-03:
Another good catch.  Will fix that asap

FormerLurker, 2021-01-08:
OK, this should be fixed:

[image]

Installation url here: ```https://github.com/FormerLurker/Octolapse/archive/8cae0c9d8f9b11d7a55c3d6fb04956931636429c.zip```

Hopefully I got everything this time :)

tech2077, 2021-01-22:
Sorry for the late response, but it's working great now and that fixed the formatting

closed 2021-01-22 by tech2077

FormerLurker, 2021-01-22:
Man, i have to thank you for closing this out and getting back to me.  Rare gem indeed!