FormerLurker/Octolapse #656: Error Starting Timelapse

Octoprintz, 2021-01-01:
When I start a print with Octolapse enabled it will respond with Octolapse Startup Failed.

[image]

Printer name Geeetech A10T, 3 in 1 out
Octolapse Version 0.4.1
Octoprint Version 0.18.0, running on Raspberry Pi 4 Model B Rev 1.4

Octoprintz, 2021-01-01:
### Error Details

2020-12-31 22:16:31,827 - octolapse.__init__ - ERROR - Unable to start the timelapse.
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/__init__.py", line 2427, in on_print_start
    timelapse_settings = self.get_timelapse_settings()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/__init__.py", line 2666, in get_timelapse_settings
    success, error_type, error_list = current_printer_clone.get_gcode_settings_from_file(gcode_file_path)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/settings.py", line 682, in get_gcode_settings_from_file
    new_slicer_settings.update_settings_from_gcode(new_settings, self)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/settings.py", line 2789, in update_settings_from_gcode
    num_extruders = int(settings_dict["machine_extruder_count"])
TypeError: int() argument must be a string, a bytes-like object or a number, not 'NoneType'

FormerLurker, 2021-01-01:
Post the gcode please.  Looks like some unexpected text in the slicer settings output.

Octoprintz, 2021-01-01:
Here is the gcode file I tried octolapse with,

[GA_10mmcube (1).zip](https://github.com/FormerLurker/Octolapse/files/5758575/GA_10mmcube.1.zip)

Octoprintz, 2021-01-01:
What changes do I need to make to my gcode?

FormerLurker, 2021-01-01:
Do you have 3 extruders?  If not, you added the wrong script to the start gcode.  Try the single extruder script.

Octoprintz, 2021-01-02:
The automatic configuration gcode was interfering with my gcode coloring app so I have decided not to use it, and yes I do have three extruders.

Octoprintz, 2021-01-02:
Now that I am not using the automatic configuration Octolapse is working.

Octoprintz, 2021-01-02:
And another problem, after the print head moves to take a snapshot then moves back it pauses for about 20 seconds.

Octoprintz, 2021-01-02:
Ok I have also fixed that  so I will close this issue

closed 2021-01-02 by Octoprintz