FormerLurker/Octolapse #657: Octolapse not working after upgrade to Python3

ajdavids, 2021-01-01:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4.1

#### Version of OctoPrint

OctoPrint Version: 1.5.2

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: yes

#### What were you doing when the problem occurred

1. Backup settings for Octopi v.0.15
2. Flash Octopi 0.17
3. Restore settings and plugins from backup
4. Run the [python3 update script](https://octoprint.org/blog/2020/09/10/upgrade-to-py3/)
5. Correct the ffmpeg directory from `/usr/bin/avconv` to `/usr/bin/ffmpeg` in octoprint settings
6. Try to record an octolapse
7. Receive Stabilization error and rendering error during print

#### What should have happened?
It should have correctly rendered the timelapse

#### What happened instead?
```py
2021-01-01 07:42:44,980 - octolapse.timelapse - ERROR - Failed to take a snapshot for the provided snapshot plan.
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/timelapse.py", line 409, in _take_timelapse_snapshot_precalculated
    self._trigger_profile.get_snapshot_plan_options()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/stabilization_gcode.py", line 1112, in create_gcode_for_snapshot_plan
    snapshot_plan, g90_influences_extruder, options
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/stabilization_gcode.py", line 368, in initialize_for_snapshot_plan_processing
    self.distance_to_lift = snapshot_plan.initial_position.distance_to_zlift(self.z_lift_height)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/gcode_processor.py", line 515, in distance_to_zlift
    if amount_to_lift < utility.FLOAT_MATH_EQUALITY_RANGE:
TypeError: '<' not supported between instances of 'NoneType' and 'float'
2021-01-01 08:01:07,781 - octolapse.script - ERROR - The script at path '/usr/bin/avconv' could not be found for  for Unknown.  Please check your script path and try again.
2021-01-01 08:01:07,782 - octolapse.render - ERROR - Rendering Error
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/render.py", line 1591, in _render
    return_code, stderr_text))
octoprint_octolapse.render.RenderError: return-code: Could not render movie, got return code -100: 
```

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.17

#### Printer model & used firmware incl. version

Printer Model: Prusa I3 MK3S MMU2S
Printer Firmware Version: 3.9.2

#### Browser and version of browser, operating system running browser

Browser: Chrome
Browser OS: Windows 10

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://gist.github.com/ajdavids/8d4926873c884f162d25031f387115d9

#### Link to settings.json

Link to settings.json with all passwords removed: https://gist.github.com/ajdavids/717c645cd133dda449f9bda69e3ef4b2

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://gist.github.com/ajdavids/5abbb4daa75011d0829f16b36ccf60e4

#### Link to octoprint.log

Link to octoprint.log:  https://gist.github.com/ajdavids/cb646a40b8254eccf21e6b6dc4ef4aee

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

ajdavids, 2021-01-01:
I was able to fix `/usr/bin/avconv` by creating a symlink there to ffmpeg, though octolapse should really respect the setting from the octoprint settings. Even though I'm on 0.4.1, though, I'm still getting the stabilization issue :(

FormerLurker, 2021-01-01:
@ajdavids,

>  octolapse should really respect the setting from the octoprint settings

It is supposed to :(

> I'm still getting the stabilization issue

Have you tried uninstalling and reinstalling Octolapse?  The upgrade script alone might not be enough for Octolapse, since it compiles a special python extension during installation.  If the python version is changed, it will try to use a library that is bound to python 2, which probably won't work.  Let me know if a reinstall fixes the stabilization issue, and I'll try to figure out what's going on with the ffmpeg path.

ajdavids, 2021-01-04:
Hey @FormerLurker , I'm away until the 11th but will try these steps then.

ajdavids, 2021-01-12:
Hey @FormerLurker , I uninstalled octolapse and reinstalled it. For some reason it reinstalled to 4.0 and had to then upgrade to 4.1. Same issue:

```python

2021-01-12 05:51:16,800 - octolapse.timelapse - ERROR - Failed to take a snapshot for the provided snapshot plan.
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/timelapse.py", line 409, in _take_timelapse_snapshot_precalculated
    self._trigger_profile.get_snapshot_plan_options()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/stabilization_gcode.py", line 1112, in create_gcode_for_snapshot_plan
    snapshot_plan, g90_influences_extruder, options
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/stabilization_gcode.py", line 368, in initialize_for_snapshot_plan_processing
    self.distance_to_lift = snapshot_plan.initial_position.distance_to_zlift(self.z_lift_height)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/gcode_processor.py", line 515, in distance_to_zlift
    if amount_to_lift < utility.FLOAT_MATH_EQUALITY_RANGE:
TypeError: '<' not supported between instances of 'NoneType' and 'float'
```

ajdavids, 2021-01-12:
I should remove the symlink I made as a kludge and see if it now respects the octoprint setting for ffmpeg location

FormerLurker, 2021-01-12:
OK, this looks a lot like an issue that was solved for most people via an update that was made in V0.4.1, but perhaps it exists in other places within the code.  I will dig in.  FYI, the last time this problem was fixed it took almost a year to work out, so hopefully it can be replicated in much less time now that I have a hunch what's going on.

ajdavids, 2021-01-21:
How did your hunch pan out, @FormerLurker ?

ajdavids, 2021-01-25:
If it helps, this always seems to happen for the very first snapshot in a print.

FormerLurker, 2021-01-25:
That may well help.  My initial hunch is probably not correct.  I setup a dev environment on a pi4 to try to figure this out, but so far no luck.  Will keep at it.

ajdavids, 2021-01-25:
Sure. I'm using a pi 3b+ if that matters.

FormerLurker, 2021-01-25:
It may well matter.  I'm going to start over from scratch first and re-evaluate the situation.  Will keep you posted.

ajdavids, 2021-01-29:
@FormerLurker Interestingly, this is only an issue if I do Smart - Gcode. If I do Classic - Gcode, everything seems to work.

FormerLurker, 2021-01-29:
Regarding the smart vs classic trigger, that isn't really unexpected.  My belief is this is a problem caused by different compilers/platforms (c++ code is used for the smart trigger, compiled dynamically when the plugin is installed, which is pretty tricky to do for all os variants, compilers, different updates/upgrades, other programs also trying to update/change configurations, etc...).

Anyway, going to hunker down and do some Octolapse maintenance later today or tomorrow.  Hopefully I figure something out.

ajdavids, 2021-04-25:
First snapshot plan:

[image]

ajdavids, 2021-04-25:
(this is with animated - printing, but same issue as with it stabilized to the center right like before)

matmat07, 2022-04-01:
FYI, I just had the issue. Same error message in the log. I just replaced the FFMPEG path before trying too. This is my first time in a while trying for real.

I will try reinstalling octolapse, no news is bad news.

bdleitner, 2022-05-14:
I just ran into this as well.  I only just installed Octolapse today and had previously changed the settingin Webcam & Timelapse to use `/usr/bin/ffmpeg`, but in the octolapse logs I see:

```
2022-05-15 00:06:12,789 - octolapse.script - ERROR - The script at path '/usr/bin/avconv' could not be found for  for Unknown.  Please check your script path and try again.
2022-05-15 00:06:12,790 - octolapse.render - ERROR - Rendering Error
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/render.py", line 1591, in _render
    return_code, stderr_text))
octoprint_octolapse.render.RenderError: return-code: Could not render movie, got return code -100: 
```

Octolapse: 0.4.1
Octopi: 0.18.0
OctoprintL 1.7.3

FormerLurker, 2022-05-17:
Have you tried a reboot?  Could be a problem syncing the setting.

bdleitner, 2022-05-19:
My issue persisted past a reboot as well.  I was able to solve it by creating a symlink from `/usr/bin/avconv` to `/usr/bin/ffmpeg` and that was successful and working around the problem.