FormerLurker/Octolapse #659: DSLR Camera Octolapse Inconsistencies

NathanK20, 2021-01-03:
I just took this octolapse using gphoto2 to use a dslr camera and I have turned off auto focus on the camera as well as turned off the flash to make the snapshots consitent, but I am still getting artifacts in lighting most visible in the blurry, sort of flickering layers on the print bed of my prusa mk3. How can I fix this? Video linked here: [Here](https://drive.google.com/file/d/1gErPSeX_dHs71cG5mZP4CP1qkCymz6Q6/view?usp=sharing)

FormerLurker, 2021-01-03:
Your bitrate is too low.  Check in the rendering profile and try increasing that to maybe 20000K

I'm not sure what the appropriate bitrate is since I don't know what your input resolution is.  For example, a bitrate of 35–45 Mbps is recommended for 4K images @ 30fps, but at 60FPS 53–68 Mbps is recommended.  It's a complicated subject, and I recommend doing some research on video bitrates for various resolutions, framerates, and output formats.

Thanks

NathanK20, 2021-01-03:
Thank you for the response Ill try it out.