FormerLurker/Octolapse #660: "No "

macombweare, 2021-01-03:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: ___REPLACE_THIS__OCTOLAPSE_VERSION_GOES_HERE

#### Version of OctoPrint

OctoPrint Version: ___REPLACE_THIS__OCTOPRINT_VERSION_GOES_HERE

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: ___REPLACE_THIS__YES_OR_NO

#### What were you doing when the problem occurred

1. ___REPLACE_THIS__STEP_ONE_GOES_HERE
2. ___REPLACE_THIS__STEP_TWO_GOES_HERE
3. ___REPLACE_THIS__STEP_...

#### What should have happened?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: ___REPLACE_THIS__OS_NAME_GOES_HERE
Os Version: ___REPLACE_THIS__OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2021-01-03:
You won't see any images until a snapshot is taken during a print.

micheal-hill, 2023-08-12:
I'm seeing this for the entire duration of the print.  
At the end of the print I get a video result, so it's evidently taking snapshots throughout the print. But the preview never shows.

See for example:
[image]

matiasfernandeza, 2023-08-14:
exact same problem as [micheal-hill](https://github.com/micheal-hill) here
[image]

fighterguard, 2023-08-16:
I'm also having the same problem, except I'm running Octoprint on Windows. Prints start and finish, and the timelapse videos are generated perfectly, but I never see a preview

briodan, 2023-08-17:
seeing the same issue here with a USB camera

Video generate correctly, but no snapshots are displayed

[image]

andres1995sergio, 2023-08-27:
I'm also having this issue but I'm running a rather strange setup with octoprint on WSL Ubuntu in windows and an Android Xiaomi smartphone as camera, using the ipwebcam app. I follow this guide to get things running 
https://www.reddit.com/r/ender6/comments/12t724m/how_i_installed_octoprint_in_wsl_and_connected_to/?utm_source=share&utm_medium=android_app&utm_name=androidcss&utm_term=1&utm_content=2

julianaklulo, 2023-08-30:
I have the same issue, running on a Raspberry Pi 3B with Pi Camera.
The snapshot count show that snapshots were taken, the video is rendered correctly, but still get the message error and no preview.

DiEVeXx, 2023-08-30:
I think all here can find a solution for this issue in the comments of the
issue #917

andres1995sergio, 2023-08-30:
@DiEVeXx suggestion worked for me, now I can see the timelapse preview

julianaklulo, 2023-08-31:
@DiEVeXx worked for me as well, thank you!