FormerLurker/Octolapse #661: Refuses to install - UnicodeEncodeError

jescholl, 2021-01-05:
Octolapse fails to install, giving the following error in octoprint.log (possible duplicate of https://github.com/FormerLurker/Octolapse/issues/633, though there were other issues there)
```
2021-01-05 00:48:14,862 - octoprint.plugins.pluginmanager - ERROR - Could not install plugin from https://github.com/FormerLurker/Octolapse/archive/v0.4.1.zip
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 717, in _command_install_archive
    returncode, stdout, stderr = self._call_pip(pip_args)
  File "/usr/local/lib/python3.8/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 1379, in _call_pip
    return self._pip_caller.execute(*args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/octoprint/util/pip.py", line 209, in execute
    return self.call(command, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/octoprint/util/commandline.py", line 243, in call
    all_stderr += process_stderr(p.stderr.readlines(timeout=0.5))
  File "/usr/local/lib/python3.8/site-packages/octoprint/util/commandline.py", line 239, in process_stderr
    return process_lines(lines, self._log_stderr)
  File "/usr/local/lib/python3.8/site-packages/octoprint/util/commandline.py", line 229, in process_lines
    processed = self._preprocess_lines(
  File "/usr/local/lib/python3.8/site-packages/octoprint/util/pip.py", line 511, in _preprocess_lines
    return list(map(self._preprocess, lines))
  File "/usr/local/lib/python3.8/site-packages/octoprint/util/pip.py", line 530, in _preprocess
    return to_unicode(clean_ansi(text))
  File "/usr/local/lib/python3.8/site-packages/octoprint/util/commandline.py", line 57, in clean_ansi
    return _ANSI_REGEX.sub(b"", line.encode("latin1")).decode("latin1")
UnicodeEncodeError: 'latin-1' codec can't encode character '\u2018' in position 75: ordinal not in range(256)
```

I have tried installing several different versions and always get the same results
  * Using the install button in plugin manager (it claims to instal 0.4.1 but logs show it's 0.4.0)
  * 0.4.0 by URL (https://github.com/FormerLurker/Octolapse/archive/v0.4.0.zip)
  * 0.4.1 by URL (https://github.com/FormerLurker/Octolapse/archive/v0.4.1.zip)
  * 0.4.1rc1 by URL (https://github.com/FormerLurker/Octolapse/archive/v0.4.1rc1.zip)

#### Version of Octolapse

Octolapse Version: 0.4.0, 0.4.1, 0.4.1rc1

#### Version of OctoPrint
OctoPrint Version: 1.5.2
Python Version: 3.8.7

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: N/A, octolapse wouldn't install

#### What were you doing when the problem occurred

1. Install octolapse through plugin manager

#### What should have happened?
Install successfully

#### What happened instead?
Plugin Manager UI seems to indicate success but stalls, octoprint UI stops responding, install fails

#### Operating System running OctoPrint and Octolapse
OS Name: Ubuntu
Os Version: 20.04
Octoprint container: [octoprint/octoprint:1.5.2](https://hub.docker.com/layers/octoprint/octoprint/1.5.2/images/sha256-b44bb5a2a05ac948e2fd5b7c4308a80a3909084b947ff65f817c6adae2d8962e?context=explore)

#### Link to settings.json
Link to settings.json with all passwords removed: https://gist.github.com/jescholl/aef43a2301b3802ab2dbd0229cf0d384#file-settings-json

#### Link to octoprint.log
Link to octoprint.log: https://gist.github.com/jescholl/aef43a2301b3802ab2dbd0229cf0d384#file-octoprint-log

#### Link to plugin_pluginmanager_console.log
Link to plugin_pluginmanager_console.log: https://gist.github.com/jescholl/aef43a2301b3802ab2dbd0229cf0d384#file-plugin_pluginmanager_console-log

FormerLurker, 2021-01-05:
I'm not sure if I have tested python 3.8.7.  Will try that out and see what I find.

In the meantime, can you try installing from a command prompt and see what happens?  Let me know if you need instructions.

jescholl, 2021-01-05:
I added the manual install log to that same gist (https://gist.github.com/jescholl/aef43a2301b3802ab2dbd0229cf0d384#file-manual_install), It seems to be a different error, but it's not obvious to me what's failing.  Also, one of the commands seems to reference an install record `/tmp/pip-record-p2sj6o_o/install-record.txt`, but that file doesn't exist or was deleted before it returned to the shell.

FormerLurker, 2021-01-05:
@jescholl, gcc was killed while compiling, which usually means you've run out of memory.  I'd recommend doing a full reboot of the pi (not just an octoprint reboot), and try to immediately install with a terminal.  You could run htop in a separate terminal to watch memory usage and see if that's the problem.  Or, if you have additional memory available, you could increase the amount of memory assigned to the docker container.  Let me know how that goes.

jescholl, 2021-01-05:
Ah hah! You were right!  Adding more memory fixed it, thanks for the help!

closed 2021-01-05 by jescholl

FormerLurker, 2021-01-05:
Glad to help!  I had what I thought to be a similar issue yesterday, but it turns out it was just a plain old compiler error, lol!  Enjoy the plugin!