FormerLurker/Octolapse #662: CUstom IMage PReferneces not working Octolapse 0.4.1 | Octoprint 1.5.2

Jack77777777, 2021-01-07:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4.1, settings version 0.4.0

#### Version of OctoPrint

OctoPrint Version: 1.5.2

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. Changing settigs in the ustom image preferences 
2. The above change shows no to the to image preview
3. Hitting Cancel, Defaults or Save freezes the screen for about 2 minutes and closes with out applying the changes
4. What have i tried to fix it so far: Cretaed new camera profile, disabled plugins like dashboard, print job history... which uses camera view

#### What should have happened?
Changes should apply 

#### What happened instead?
Changes not applied

#### Operating System running OctoPrint and Octolapse

OS Name: Octopi
Os Version: 0.17

#### Printer model & used firmware incl. version

Printer Model: Anycubic i3 Mega-S
Printer Firmware Version: Anycubic i3 Mega Marlin (BLTouch) 1.1.9 by davidramiro & MNieddu91
https://github.com/MNieddu91/Marlin-AI3M-BLTouch

#### Browser and version of browser, operating system running browser

Browser: Google Chrome
Browser OS: 87.0.4280.88

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: https://drive.google.com/file/d/1gYYLfsDOmvneJG4m5jggiVETN7jrcYAr/view?usp=sharing

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://pastebin.com/ivxkLsK5

#### Link to octoprint.log

Link to octoprint.log:  https://pastebin.com/A7qUP75w

#### Link to contents of Javascript console in the browser

Link to javascript console output: https://pastebin.com/dcvG0Dth

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: https://drive.google.com/file/d/1VcKxaKZkWb1azRHK-NzZmt8QA-OWvio1/view?usp=sharing

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

Jack77777777, 2021-01-07:
BTW: cobtrol.htm is showing. Removed bith camera profiles and added obe new. For the first time i could change Image Preferences and the next time same issue as before.

FormerLurker, 2021-01-07:
According to the plugin_octolapse.log file the connection to your camera was refused for some reason.  There are several possibilities, but the easiest thing to try would be to change the camera address from:  

```http://127.0.0.1:8080/```

to

```http://127.0.0.1/webcam/```

If that doesn't work, we can investigate further.  Let me know!

Jack77777777, 2021-01-07:
Changed in Camera settings within octolapse to http://127.0.0.1/webcam/ but same issue.
After trying to change something in the image prefs and save http://IPOCTOPI/webcam/ was not reachable for a while.

FormerLurker, 2021-01-07:
> Shall i re-instal octolapse?

Hmm..  couldn't hurt, but first let's get better logging of the camera module.  
Try this:

1. edit your logging profile within the Octolapse tab.
2. Click 'clear all logs' and confirm.
3. Exit the edit popup, and select the 'Log everything' option from the logging profile selection in the Octolapse tab.
4. Click the edit pen in the snapshot preview to open the custom image preference screen and attempt to change a few settings and close the popup.
5. Change the debug profile from 'log everything' to 'log all errors'.  This is important, because if you leave it set to log everything you will have performance issues.
6. Edit the 'log all errors' profile and click 'download log'
7. post the new log to gist.github.com.

Next, if you want you can try to reinstall Octolapse and see if that helps.  I suspect it won't, but it's worth a shot.  I will examine the new log, which should contain a lot more information.

Jack77777777, 2021-01-07:
Re-Install brought no improvement as you mentioned.
I followed your above instructions.

Logs:
Changed Image settings and closed popup by clicking X on the upper right corner: 
https://gist.github.com/Jack77777777/c1902c9e31ae60d66cb54e5dc566cc56
Changed Image settings and closed popup by clicking Save: 
https://gist.github.com/Jack77777777/2874928d2874ec340a13f69cd64629e0

For me the logs show no errors but the settings are not applied.

Thanks for your effort

Jack77777777, 2021-01-07:
Disbaled all 3rd Party Plugins, rebooted Pi and still same issue.
Could it be a mjpg streamer issue ? How can i update/reinstall mjpg streamer ?

Jack77777777, 2021-01-08:
Did a fresh install of Octopi 0.17 on a brand new sd card and installed octolapse. Image Prefs are working. I will install plugins, change setup step by step and after every step look if custom image prefs still working. I will send a feedback here.

FormerLurker, 2021-01-08:
@Jack77777777, I see these mysterious issues all the time.  It's probably an environment thing compounded with the fact that us plugin developers tend to test fresh OctoPi installs more than any other configuration.  I'd love to know exactly why you were having this issue, but I think the better solution is to figure out how to sidestep mpegstreamer completely and create a separate plugin that is capable of interfacing with the camera directly.  Then Octolapse could call that plugin to change the camera settings.  Other plugins could be written for different cameras or scenarios (raspberry pi camera settings plugin/IP cam plugin, etc...)

Looking forward to your results, and am very glad you got this working!

closed 2021-09-11 by Jack77777777