FormerLurker/Octolapse #663: Help people wwith

ed-french, 2021-01-08:
#### If this is a feature request describe it here

Clean Octoprint install on pi v4
Installed Octolapse
Tried to print
Warned that ffmpeg path was not set correctly

#### Version of Octolapse
 
Octolapse Version: ___REPLACE_THIS__OCTOLAPSE_VERSION_GOES_HERE

#### Version of OctoPrint

OctoPrint Version: ___REPLACE_THIS__OCTOPRINT_VERSION_GOES_HERE

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: ___REPLACE_THIS__YES_OR_NO

#### What were you doing when the problem occurred

1. ___REPLACE_THIS__STEP_ONE_GOES_HERE
2. ___REPLACE_THIS__STEP_TWO_GOES_HERE
3. ___REPLACE_THIS__STEP_...

#### What should have happened?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: ___REPLACE_THIS__OS_NAME_GOES_HERE
Os Version: ___REPLACE_THIS__OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2021-01-08:
Set the ffmpeg path here:

[image]

Then restart octoprint, because there is an error in Octolapse that can sometimes prevent the ffmpeg path from being updated.  I haven't figured that out yet.

ed-french, 2021-01-08:
Thanks, I did find it in the end, but I was expecting it to be in octolapses settings somewhere. Given how vanilla the setup is I would think it's a common problem, and it might be possible to improve the feedback to the user?

FormerLurker, 2021-01-08:
It is not a common issue because this is pre-configured within OctoPrint.  If you have a snapshot of the error you were seeing (this would just save me a bit of time) and some suggestions for how to make the problem more clear, I'm happy to add this.  I'm doing a bunch of plugin work today, so it's a good time!

FormerLurker, 2021-01-08:
EDIT:  It is pre-configured within OctoPi, not OctoPrint :)  I presume you are running on Windows, MacOs, or a custom linux install?

ed-french, 2021-01-08:
Nope, on raspbian on pi using image i downloaded a while ago and updated over Christmas

FormerLurker, 2021-01-08:
Hmm..  I can't understand then why the ffmpeg path wasn't set.  I just flashed a card yesterday and it was set.  I will try to find the error message and see if I can't clarify somewhat.

ed-french, 2021-01-08:
It was set to something like /usr/bin/(something like avideo but cant
remember exactly)

FormerLurker, 2021-01-08:
Probably avconv, which should have worked....  Anyway, I can def add a 'help' button with detailed instructions to the error popup.

ed-french, 2021-01-08:
Yep, that sounds right
The help thing would be good
Thanks

ed-french, 2021-01-08:
The rendering did fail of the print I had on this am, presumably because it
needed a restart. Is there anything you'd like me to test/record/check
whilst it is in the broken state before I restart?
Here are my settings as showing now- test button reports it's OK
[image: image.png]

fsqonbdk, 2023-05-04:
Mine was pre-configured to /usr/bin/avconv and generated an error when I clicked on test. Changing it to /usr/bin/ffmpeg worked and no more error when pressing Test.

Octolapse 0.4.2
Octoprint 1.8.7
OctoPi 0.18.0