FormerLurker/Octolapse #665: OctoLapse works throughout a print but then doesn't stop at the end.

NameOfTheDragon, 2021-01-08:
TL;DR: there does seem to be an interesting error in the OctoLapse log file (below)

#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4.1

#### Version of OctoPrint

OctoPrint Version: 1.5.2

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

1. Start a print. Observe that the Octolapse title contains the camera icon, indicating that it is active and that OctoPrint has entered `Printing` state.
2. Wait for the print to complete. Observe that OctoPrint has returned to `Operational` state.
3. Observe that OctoLapse is still active (camera icon still present, timelapse not rendered, etc)

From this point there's no recovery. You can either:
- Start a new print, in which case Octolapse will produce an error `Line XXX expected but Line 1 found` and then stop.
- Go into the OctoLapse configuration screen and click the STOP icon, in which case the session is cancelled and any recorded snapshots are deleted without rendering.

Either way, the timelapse is lost.

**Does not happen every time**. This happens on maybe 1/4 of prints.

#### What should have happened?
Expected behaviour is pretty obvious, OctoLapse should stop and render the time-lapse when the print finishes. It doesn't seem to be detecting the end of the print correctly.

#### What happened instead?
OctoLapse did not detect the end of the print and did not stop or render the time-lapse.

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.17.0

#### Printer model & used firmware incl. version

Printer Model: Creality Ender 3, heavily modified (32-bit mainboard, direct drive extruder)
Printer Firmware Version: Klipper

#### Browser and version of browser, operating system running browser

Browser: Edge
Browser Windows 10

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://gist.github.com/NameOfTheDragon/e08be6f296e2e9408063e80d8e23b2f4 

#### Link to settings.json

Link to settings.json with all passwords removed: https://gist.github.com/NameOfTheDragon/ba14e4a1a36e1cfdce62b7a028d6c44d

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://gist.github.com/NameOfTheDragon/270f46b2e4826494c86b8ad4b403ace0

#### Link to octoprint.log

Link to octoprint.log:  https://gist.github.com/NameOfTheDragon/e15ad64ddf3091d4a0034d504fc1edb1

#### Link to contents of Javascript console in the browser

Link to javascript console output: n/a

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: n/a

FormerLurker, 2021-01-08:
Can you try restarting OctoPrint and see if you see a 'Failed Rendering' button appear on the octolapse tab?  OctoLapse detects the end of the print via the 'Print Ended' event, and it's hard for me to imagine how it would fail to receive the event.  It may be necessary to capture a more detailed log to see what's going on.  I wonder if this could be triggered by running Octolapse in test mode, and trying several very small prints (maybe a couple layers of a calibration cube) until the error is encountered.  'Debug logging' should suffice for this.  If that isn't illuminating, we may need to do a serial log via OctoPrint and try the same steps.

The errors you see in the existing octolapse_log are mysterious (why can't it read those files??), but they should NOT prevent the files from being available in the 'failed rendering' dialog.

Also, have you maybe upgraded to python 3 via the upgrade script that everyone is using?  If so, have you reinstalled Octolapse since doing that?  If not, try installing Octolapse again and see if that helps.  The custom extension that is compiled needs to be recompiled for python 3, else all kinds of strange things can happen.

NameOfTheDragon, 2021-01-22:
I suspect something external to OctoLapse is going on, because sometimes when I finish a print then send a new gcode file from Prusa Slicer, OctoPrint doesn't "see" the file until I click the refresh button.