FormerLurker/Octolapse #666: Unable to install zlib error

rhetz, 2021-01-13:
[script.txt](https://github.com/FormerLurker/Octolapse/files/5805797/script.txt)

#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: ___REPLACE_THIS__OCTOLAPSE_VERSION_GOES_HERE

#### Version of OctoPrint

OctoPrint Version: 1.5.2

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. ___REPLACE_THIS__STEP_ONE_GOES_HERE
2. ___REPLACE_THIS__STEP_TWO_GOES_HERE
3. ___REPLACE_THIS__STEP_...

#### What should have happened?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: ___REPLACE_THIS__OS_NAME_GOES_HERE
Os Version: ___REPLACE_THIS__OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2021-01-13:
Post the console output from your manual install attrmpt please.  You may want to try updating your raspberry pi first too.

rhetz, 2021-01-13:
source venv/bin/activate
(venv) root@orangepilite:~# pip install https://github.com/FormerLurker/Octolapse/archive/v0.4.1.zip --no-cache-dir
Collecting https://github.com/FormerLurker/Octolapse/archive/v0.4.1.zip
  Downloading https://github.com/FormerLurker/Octolapse/archive/v0.4.1.zip
     / 1.3 MB 703 kB/s
    ERROR: Command errored out with exit status 1:
     command: /root/venv/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-o16550v1/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-o16550v1/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-spfhgznk
         cwd: /tmp/pip-req-build-o16550v1/
    Complete output (8 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-req-build-o16550v1/setup.py", line 57, in <module>
        import octoprint.server
    ModuleNotFoundError: No module named 'octoprint'
    likely tags: v0.4.1
    picking v0.4.1
    got version from expanded keyword {'version': 'v0.4.1', 'full-revisionid': 'cec243c65a979e8bbe90f09fc5aec6d0844d2d3b', 'dirty': False, 'error': None, 'date': '2020-12-04T18:35:29-0600'}
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.

rhetz, 2021-01-13:
If I try on octoprint I get:

Installing plugin...

Installing plugin from uploaded archive.../opt/venv/bin/python -m pip --disable-pip-version-check install file:///tmp/tmpwx57wygp.zip --no-cache-dirProcessing /tmp/tmpwx57wygp.zipCollecting pillowDownloading Pillow-8.1.0.tar.gz (44.9 MB)Requirement already satisfied: sarge in /opt/venv/lib/python3.8/site-packages (from Octolapse==0.3.4) (0.1.5.post0)Requirement already satisfied: six in /opt/venv/lib/python3.8/site-packages (from Octolapse==0.3.4) (1.15.0)Requirement already satisfied: OctoPrint>1.3.8 in /opt/venv/lib/python3.8/site-packages (from Octolapse==0.3.4) (1.5.2)Requirement already satisfied: psutil in /opt/venv/lib/python3.8/site-packages (from Octolapse==0.3.4) (5.7.2)Requirement already satisfied: Flask-Assets<3,>=2.0 in /opt/venv/lib/python3.8/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (2.0)Requirement already satisfied: Flask-Login<0.6,>=0.5 in /opt/venv/lib/python3.8/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.5.0)Requirement already satisfied: PyYAML<6,>=5.3.1 in /opt/venv/lib/python3.8/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (5.3.1)Requirement already satisfied: requests<3,>=2.23.0 in /opt/venv/lib/python3.8/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (2.24.0)Requirement already satisfied: Jinja2<3,>=2.11.2 in /opt/venv/lib/python3.8/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (2.11.2)Requirement already satisfied: blinker<2,>=1.4 in /opt/venv/lib/python3.8/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (1.4)Requirement already satisfied: zeroconf<0.25,>=0.24; python_version >= "3" in /opt/venv/lib/python3.8/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.24.5)Requirement already satisfied: Click<8,>=7.1.2 in /opt/venv/lib/python3.8/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (7.1.2)Requirement already satisfied: wrapt<2,>=1.12.1 in /opt/venv/lib/python3.8/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (1.12.1)Requirement already satisfied: semantic-version<3,>=2.8.5 in /opt/venv/lib/python3.8/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (2.8.5)Requirement already satisfied: cachelib<1,>=0.1 in /opt/venv/lib/python3.8/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.1.1)Requirement already satisfied: filetype<2,>=1.0.7 in /opt/venv/lib/python3.8/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (1.0.7)Requirement already satisfied: OctoPrint-FirmwareCheck>=2020.09.23 in /opt/venv/lib/python3.8/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (2020.9.23)Requirement already satisfied: OctoPrint-FileCheck>=2020.08.07 in /opt/venv/lib/python3.8/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (2020.8.7)Requirement already satisfied: websocket-client<1,>=0.57 in /opt/venv/lib/python3.8/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.57.0)Requirement already satisfied: future<1,>=0.18.2 in /opt/venv/lib/python3.8/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.18.2)Requirement already satisfied: rsa==4.0 in /opt/venv/lib/python3.8/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (4.0)Requirement already satisfied: regex!=2018.11.6 in /opt/venv/lib/python3.8/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (2020.7.14)Requirement already satisfied: watchdog==0.10.4 in /opt/venv/lib/python3.8/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.10.4)Requirement already satisfied: werkzeug<2,>=1.0.1 in /opt/venv/lib/python3.8/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (1.0.1)Requirement already satisfied: pyserial<4,>=3.4 in /opt/venv/lib/python3.8/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (3.4)Requirement already satisfied: markupsafe<2.0,>=1.1 in /opt/venv/lib/python3.8/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (1.1.1)Requirement already satisfied: pkginfo<2,>=1.5.0.1 in /opt/venv/lib/python3.8/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (1.5.0.1)Requirement already satisfied: feedparser<7,>=6.0.2; python_version >= "3" in /opt/venv/lib/python3.8/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (6.0.2)Requirement already satisfied: itsdangerous<2,>=1.1.0 in /opt/venv/lib/python3.8/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (1.1.0)Requirement already satisfied: tornado==5.1.1 in /opt/venv/lib/python3.8/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (5.1.1)Requirement already satisfied: markdown<3.2,>=3.1 in /opt/venv/lib/python3.8/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (3.1.1)Requirement already satisfied: pylru<2,>=1.2 in /opt/venv/lib/python3.8/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (1.2.0)Requirement already satisfied: unidecode<0.05,>=0.04.14 in /opt/venv/lib/python3.8/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.4.21)Requirement already satisfied: flask<2,>=1.1.2 in /opt/venv/lib/python3.8/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (1.1.2)Requirement already satisfied: netifaces<1,>=0.10.9 in /opt/venv/lib/python3.8/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.10.9)Requirement already satisfied: Flask-Babel<2,>=1.0 in /opt/venv/lib/python3.8/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (1.0.0)Requirement already satisfied: sentry-sdk<1,>=0.15.1 in /opt/venv/lib/python3.8/site-packages (from OctoPrint>1.3.8->Octolapse==0.3.4) (0.16.5)Requirement already satisfied: frozendict<2,>=1.2 in /opt/venv/lib/python3.8/si [... cut ...] f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-93gjj4oe/install-record.txt --single-version-externally-managed --compile --install-headers /opt/venv/include/site/python3.8/pillow Check the logs for full command output.Error!Could not parse output from pip, see plugin_pluginmanager_console.log for generated outputError!Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

FormerLurker, 2021-01-13:
When you tried to install from pip, it looks like you may have activated the wrong Virtual environment: No module named 'octoprint

Is it possible you upgraded to python 3 and moved your venv?

FormerLurker, 2021-01-13:
Also, mayby double check the exact prerequisites for your particular linux flavor: https://pillow.readthedocs.io/en/stable/installation.html#building-on-linux

I see you attempted to install those, but perhaps not all of them?