FormerLurker/Octolapse #667: Extruder stops after first frame of time lapse taken

nadipietro, 2021-01-14:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: ___REPLACE_THIS__OCTOLAPSE_VERSION_GOES_HERE

#### Version of OctoPrint

OctoPrint Version: ___REPLACE_THIS__OCTOPRINT_VERSION_GOES_HERE

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: ___Yes

#### What were you doing when the problem occurred

1. Enable Octolapse
2. Load Gcode file
3. Start print

#### What should have happened?
Should print and capture time lapse video.

#### What happened instead?
I'm using PrusaSlicer-2.2.0-rc5+win64, I have both Verbose G-code and Label Objects enabled.
After first image is captured, the extruder stops, heat stays on. The print continues and Octolapse continues to capture images. If I disable Octolapse, everything prints normally. This problem happens on all G-code files. 

#### Operating System running OctoPrint and Octolapse

OS Name: Linux v5.4.51-v8+ #1327
Os Version: Octoprint v1.5.2

#### Printer model & used firmware incl. version

Printer Model: Printer: Ender 3 Pro - Titan direct drive extruder
Printer Firmware Version: Marlin 2.0.6 for BigtreeTech v2

#### Browser and version of browser, operating system running browser

Browser: Microsoft Edge
Browser OS: Windows 10

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: [3DHubs_Marvin_x4-PLA0.12mm_7.11638g_0.2n_Ender3_4h32m.gcode.zip](https://github.com/FormerLurker/Octolapse/files/5815989/3DHubs_Marvin_x4-PLA0.12mm_7.11638g_0.2n_Ender3_4h32m.gcode.zip)

#### Link to settings.json

Link to settings.json with all passwords removed: 
[settings.zip](https://github.com/FormerLurker/Octolapse/files/5816024/settings.zip)

___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  
[plugin_octolapse.log](https://github.com/FormerLurker/Octolapse/files/5816036/plugin_octolapse.log)

#### Link to octoprint.log

Link to octoprint.log:  [octoprint.log](https://github.com/FormerLurker/Octolapse/files/5816044/octoprint.log)

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: https://user-images.githubusercontent.com/67333417/104627447-06297d00-5665-11eb-8a27-448657b9a0e3.mp4

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2021-01-14:
Did you flash with custom firmware (marlin 2) by chance?  If so, change the 'g90/g91 influences extruder' setting in your printer profile to true.  Should fix it in that case.

nadipietro, 2021-01-14:
I’m not sure what would be considered custom.  I did adjust variables, i.e. offsets for my system, and enabled specific Marlin features, i.e. BLTouch, etc.  I did not make changes to the Marlin code. 

 

The extruder works fine until Octolapse takes its first frame of the time lapse video.  Also, the extruder works correctly when Octolapse is disabled.  For example, I ran the same G-code file that I set you and the only change was to disabled Octolapse and everything printed correctly.

 

 

 

 

 

 

 

 

 

 

Thanks for your help.

Nick

FormerLurker, 2021-01-14:
> I’m not sure what would be considered custom.

Custom is any firmware that is not delivered by the printer manufacturer, in this case Creality.  If you downloaded and compiled marlin yourself, or if you used some third party firmware that isn't issued by the MFG, then I'm calling it custom.  The default profiles assume you are using what the printer shipped with.

> The extruder works fine until Octolapse takes its first frame of the time lapse video.  Also, the extruder works correctly when Octolapse is disabled.  For example, I ran the same G-code file that I set you and the only change was to disabled Octolapse and everything printed correctly.

This is expected if that setting I mentioned is incorrect.  It is a feature of the firmware your printer is using, which is why I wanted to know if you changed it.  If Octolapse doesn't know how your printer responds to certain gcodes (in this case G90/G91 and M82/M83), then it can't correctly track the printer's internal state.  It's likely that your extruder's mode was changed from absolute to relative or the other way around.  I can explain how this happens, and why there is no way to detect this at the moment if you are interested (let me know).  

So, in short, edit your printer profile and switch *G90/G91 Influences Extruder* to *True* here:

[image]

And see if that fixes it.

nadipietro, 2021-01-14:
Thank you. That seems to have taken care of the problem.

 

Regards,

Nick

 

 <https://user-images.githubusercontent.com/33363128/104655597-89f26200-5683-11eb-8b68-0ec03f88192b.png>

FormerLurker, 2021-01-14:
I am very happy to hear it?  Should I consider this issue closed?

In any case, enjoy the plugin!

nadipietro, 2021-01-15:
Yes, you can close this issue.

 

Nick

FormerLurker, 2021-01-15:
Thanks, closed!

closed 2021-01-15 by FormerLurker