FormerLurker/Octolapse #668: Printing starts instantly with cold bed/hotend only when octolapse is enabled

sergiocasero, 2021-01-15:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4.1

#### Version of OctoPrint

OctoPrint Version: 1.5.2

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: No

#### What were you doing when the problem occurred

1. Upload gcode
2. click on "print"

#### What should have happened?
The printer must wait to the bed/hotend heating and then, print

#### What happened instead?
Starts to print with the bed&hotend cold. I've tested with the print disabled and everything works as expected.

#### Operating System running OctoPrint and Octolapse

It's running on a PI, full specs:

```
connectivity.connection_check: 8.8.8.8:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1400
env.hardware.ram: 917016576
env.os.bits: 32
env.os.id: linux
env.os.platform: linux2
env.plugins.pi_support.model: Raspberry Pi 3 Model B Plus Rev 1.3
env.plugins.pi_support.octopi_version: 0.17.0
env.plugins.pi_support.throttle_state: 0x50000
env.python.pip: 19.3.1
env.python.version: 2.7.16
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.5.2

```

#### Printer model & used firmware incl. version

Printer Model: Anet ET4
Printer Firmware Version: Official

#### Browser and version of browser, operating system running browser

Browser: Chrome
Browser OS: macOS

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log: https://gist.github.com/sergiocasero/a54f81717db716a4ef2880b2320a74e7

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2021-01-15:
Turn off test mode in the octolapse tab and try again

FormerLurker, 2021-01-15:
> WTF I can say without ANY doubt that this is the fastest response I've received ever on github.

Lol!  I'm not always that fast.  

> checked and it's disabled, maybe is there any cache or something

Hmm..  this is new.  Does your printer go through the motions, but the extruder doesn't turn?  Also, you don't happen to have the 'cool running' plugin installed, do you?

Might need a log to figure this out.  Try this:

1. Switch to the 'classic layer' trigger.
2. Change the 'Debug' profile to 'Debug'
3. Start your print.
4. Wait till your extruder moves around a bit (maybe like 1 min) then cancel the print.
5. Switch your 'Debug' profile to 'Log all errors'.
6. Edit the logging profile and click the 'download log' button.
7. Paste the log into gist.github.com and drop a link here.

After that, maybe try toggling test mode on and off?  You could also try rebooting.  I've never heard of an issue with test mode before, but it's not impossible there is some edge case I've never seen.

Thanks!

sergiocasero, 2021-01-16:
Looks like after a complete fresh reboot for both (printer&PI) the issue is solved!! Let me close the issue

closed 2021-01-16 by sergiocasero

FormerLurker, 2021-01-16:
@sergiocasero, It's possible you're printer was in test mode.  There is a gcode that works the same way for some firmwares.  Not sure how that would have gotten enabled, but it would make sense that a reboot would fix it.  Either that, or somehow Octolapse's test mode setting got stuck in place.