FormerLurker/Octolapse #670: Animated Printing

TWIISTED-STUDIOS, 2021-01-17:
I have noticed I have the ender 3 pro with a BLTouch when ever I use the animated printing option on octolapse, the preview shows that it's going to be taking an image in the center the left and then the right, but what actually is happening is that the printhead moves all the way to the endstop on the left and then it takes the image. I can't get this animated printing to work how it claims to be no matter how much I try, I have even defaulted all the setting of octolapse and still nothing.

TWIISTED-STUDIOS, 2021-01-18:
[Animated printing at the front instead of middle ](https://cdn.discordapp.com/attachments/712407050956832829/800517548994461696/CE3PRO_Orc_Skull_-_Mug_-_Threaded_20210117230920.mp4)

FormerLurker, 2021-01-18:
Can you post your settings.json:

[image]

Upload to gist.github.com and put a link in here.

My guess is your have this setting incorrect in your Octolapse printer profile:

[image]

but I can't be sure.

TWIISTED-STUDIOS, 2021-01-18:
Hello thank you for your message I checked my printer profile and I belive the center is set as the current center for the origin type 
[Settings.json](https://gist.githubusercontent.com/TWIISTED-STUDIOS/1aa21e114572523a0fcbc731c99cd0f7/raw/5410cd9800bffba92c129cc8f069455fc30b9329/settings.json)

FormerLurker, 2021-01-19:
Looks like your origin is incorrect.  The origin of the Ender 3 pro is at the front left.  You can test this by first homing XY, and then sending the carriage to the origin with this gcode:

```
G28 X Y; home X and Y
G0 X0 Y0; Move X and Y to the Origin
```

If the carriage goes to the front left, you will need to change the origin to the front left as I mentioned in my previous post.

TWIISTED-STUDIOS, 2021-01-19:
I have the Bl touch installed so when I use a g28 command the printer moves to the center of the X carage and Y caragae. 

But I shall try that and see that happens but the g28 command on its own is the center.

FormerLurker, 2021-01-19:
G28 does not necessarily move the carriage to the origin.  Typically it would move X, Y, and Z to the end stops, then it will move to a destination coordinate.  In your case I believe the destination coordinates to be X110 Y110.  You can also verify this by running G28, then M114 to report the current position.

Looking forward to hearing your results!

FormerLurker, 2021-01-19:
Oh, I forgot to mention, it's possible you have an offset applied, so be sure to restart your printer (just turning it off and on should suffice) before running those commands.

TWIISTED-STUDIOS, 2021-01-19:
Awsome cheers I'll get to that after this print finishes thank your for that bit of information, it will become helpful in the future, you learn something new everyday it seems. 

I'll get back to you with the findings and of that was the issue then I'm a clown.

FormerLurker, 2021-01-19:
> you learn something new everyday it seems
Ain't that the truth!

> I'll get back to you with the findings and of that was the issue then I'm a clown.

Lol, no you're not!  Most of what I know about 3d printers I learned from users like you reporting issues like this.  Go back a couple years, and I wouldn't have known this either :)

TWIISTED-STUDIOS, 2021-01-20:
Hello so from following this guide you my friend are 100 percent correct g28 followed by m114 moves to the endstops and then to the location of x161 y127.

Doing G28 X Y moves the the front left and then GO X0 Y0 stays in that location so I do believe my origin of the profile is incorrect and that is a mistake on my behalf not realising that the G28 did not actually change the origin location

FormerLurker, 2021-01-20:
Excellent!  Let me know when you verify that Octolapse is working with the new origin location and I will close this out.

TWIISTED-STUDIOS, 2021-01-20:
That may be some time yet I do apologize I have a print going that was qued up to go. And it's a 67 or so hour print. So it may take some time to check that after this print I will definitely give it a try and see if it's working correctly now

FormerLurker, 2021-01-20:
No rush, just report back when you can , and good luck with your epic print!

TWIISTED-STUDIOS, 2021-02-01:
Ok so after adding the change to the origin and printing the animated printing setting it now prints correctly with the head staying around the center of the X carriage and the Y carriage 

Thank you for taking your time out and explaining these to me

FormerLurker, 2021-02-01:
Of course, it was my pleasure!  Am very glad to close out an issue (they seem to be opening faster than I can solve them, lol)!

closed 2021-02-01 by FormerLurker