FormerLurker/Octolapse #671: Octolapse breaks SpoolManager plugin from accounting

deejay2, 2021-01-18:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: ___REPLACE_THIS__OCTOLAPSE_VERSION_GOES_HERE

#### Version of OctoPrint

OctoPrint Version: 0.4.1

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: No

#### What were you doing when the problem occurred

1. Start a print with Octolapse enabled and Spool Manager plugin enabled too.
2. At the end of the print, notice that SpoolManager hasn't accounted for used filament... 

#### What should have happened?
SpoolManager should be able to account filament while octoprint is enabled.

#### What happened instead?
SpoolManager couldn't account for filament used in the print.

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.17.0

#### Printer model & used firmware incl. version

Printer Model: Prusa i3 MK3S
Printer Firmware Version: 3.9.3

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2021-01-18:
I suppose I could install the SpoolManager plugin and see what happens.  At the moment I have no idea what it could be.

deejay2, 2021-01-19:
It only happens when octolapse is enabled. I'm really not sure but could it be because both octolapse and spoolmanager are raising a pop-up for confirmation before starting the print?

Issue also opened on SpoolManager: https://github.com/OllisGit/OctoPrint-SpoolManager/issues/93

FormerLurker, 2021-01-19:
> I'm really not sure but could it be because both octolapse and spoolmanager are raising a pop-up for confirmation before starting the print?

You could confirm this by editing the Octolapse Main Settings (gear icon on octolapse tab-> man settings) and disabling the snapshot plan preview:

[image]

Let me know if you try this out, thanks!

shogun168, 2021-02-25:
Hmmm, I just noticed this as well .. wasn't tracking a bunch of prints so now a couple of spools are way off ..

brianantonelli, 2021-05-30:
I disabled snapshot plan preview and spool manager still didn't run..

FormerLurker, 2021-06-01:
Thanks for trying that.  I will have to take some time to test this out.  Adding to milestones.