FormerLurker/Octolapse #672: New install can not be configured

Cyber-One, 2021-01-20:
Octolapse version 0.4.0
Settings 0.4.0

Fresh install on a new install of Octopi 0.18.0 running on a Raspberry Pi 3 Model B Plus Rev 1.3
OctoPrint version 1.5.2

Attempting to setup for a Cocoon Create Touch, a re-badged Wanhao - Duplicator I3 Plus

When attempting to configure the setting, most of the configuration fields are not present, however the help question marks are present.
When trying to set the few fields that are present, a popup window appears warning the 1st extruder setting will be used preventing of setting the retraction and another 5 mandatory fields.

Refer to screen capture below.

[image]

Viewing using Chrome version 87.0.4280.141 64bit running on a Windows 10 PC
If you have any more question or need other additional information, let me know :-)

I used a previous version on a previous version of octopi (before the SD card failed) and found it to be great.
Ray

FormerLurker, 2021-01-20:
The install link in the plugin repository is broken (I just learned about this).  Check the software update plugin and install v0.4.1, and it should be fixed.

Cyber-One, 2021-01-20:
Just completed an update, 
Now the settings can be configured :-)
Starting a print with Octolapse enabled, I'll know later if all worked, but early indications are good :-)

Ray

closed 2021-01-20 by Cyber-One