FormerLurker/Octolapse #673: Can't Set Resolution or Framerate in /boot/octopi.txt

RAMilewski, 2021-01-22:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4.1

#### Version of OctoPrint

OctoPrint Version: 1.5.3

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: Yes

#### What were you doing when the problem occurred

1. Uncomment  camera_usb_options="-r 640x480  -f 10" in  /boot/octopi.txt
2. Save the file
3. Reboot the Raspberry Pi
4. Click on the OctoLapse tab
5. Select Logitech C920 in the select camera dropdown
6. Click on the pencil to the right of the dropdown

#### What should have happened?
The camera image should have appeared.

#### What happened instead?
The dreaded pink Error Loading Webcam Settings dialog box appeared.

#### Operating System running OctoPrint and Octolapse
Python 2.7.16 running OctoPi 0.17.0
OS Name:Raspian
Os Version:PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
                  NAME="Raspbian GNU/Linux"
                 VERSION_ID="10"
                 VERSION="10 (buster)"
                 VERSION_CODENAME=buster
                 ID=raspbian
                 ID_LIKE=debian

sudo apt update and sudo apt upgrade run earlier the day of the test.

#### Printer model & used firmware incl. version

Printer Model: Prusa Mini
Printer Firmware Version: 4.2.1

#### Browser and version of browser, operating system running browser

Browser: Identical results with:  Firefox Developer Edition 85.0b9, Firefox 84, and Google Chrome Version 88.0.4324.96 (Official Build) (x86_64)
Browser OS: MacOS Mojave 10.14.6

#### Link to the gcode file you were printing when the problem occurred
       Problem occurs even before a gcode file is loaded for printing.

#### Link to settings.json

Link to settings.json with all passwords removed: _https://pastebin.com/jxFnER4k
#### Link to plugin_octolapse.log

Link to plugin_octolapse.log: https://pastebin.com/TKyqe3jzd

#### Link to octoprint.log

Link to octoprint.log: https://pastebin.com/aU3iEus6

#### Link to contents of Javascript console in the browser

Link to javascript console output: https://pastebin.com/rF5Kc5fD

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2021-01-22:
Octopi.txt is not part of Octolapse, but rather is a part of OctoPrint.  Did you follow [this official guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Enabling-Camera-Controls) for getting image preferences to work with Octolapse?  You may notice that the first image in the wiki is a link to a youtube tutorial that follows the guide.

Also, the C290 is the webcam I used to develop Octolapse, so you definitely can get it working eventually!  Let me know how it goes.

Finally, I'm getting a 404 error for the link to your plugin_octolapse.log file.  Can you please double check that?

RAMilewski, 2021-01-24:
Apologies for the pastebin error.  I must not have set the duration properly.  Try this: https://pastebin.com/EcRekNp3

Yes I followed the official guide.  In Step 2 that guide instructs editing octopi.txt.   5 minutes and 10 seconds into the video it says to make the edits to octopi.txt.    Should I be changing the resolution and framerate somewhere else?    

Simply uncommenting the line breaks Octolapse, regardless of whether it's  camera_usb_options="-r 640x480 -f 10"  or  camera_usb_options="-r 1280x720 -f 30" 

For reasons I don't quite understand, octopi is using python 2.7.16.  Gina has been urging moving everything to python3.  3.7.3 is on the pi, but apparently not being used.  I'll dig into why and if that doesn't help try imaging a fresh octopi instance on another sd card and see if there's something borked elsewhere in the system.

FormerLurker, 2021-01-24:
Cool, i understand.  Will take a look at pastebin asap.

Regarding python 2.7, it is because there is a community of plugins that need to be upgraded.  Octolapse and arcwelder (my plugins) work, but it took a lot of effort.  If all the plugins you want to use work in python 3, by all means upgrade!!  The more ppl using it, the faster the cutover!  I perst hate maintaining backwards compatibility.  It is just too much work.

FormerLurker, 2021-01-24:
Ok, paste in your octopi.txt file please.  Looks like the PI can't access the necessary control.htm page.  It is probably something simple.

RAMilewski, 2021-01-27:
Octopi.txt is at https://pastebin.com/E8BDY6bA.  

OctoPi 0.18.0 just dropped.  The changelog mentions Python3 and some USB camera fixes.  I'll load that up and see if anything changes and report back.

FormerLurker, 2021-01-27:
Hmm, the issue seems to be a problem connecting to your camera from this address:

http://127.0.0.1:8080/input.json

That's the loopback address.  See if you can access input.json this was:

http://{IP_OF_PI}:8080/input.json

or

http://{IP_OF_PI}/webcam/input.json

let me know if either of those work.

RAMilewski, 2021-01-29:
http://{IP_OF_PI}:8080/input.json returns the camera settings as a JSON file.     http://{IP_OF_PI}:8080/webcam/input.json comes up 404.

Where do I need to set that address?

FormerLurker, 2021-01-29:
The second url there is wrong, no need for :8080.

However, it can get to the other url, so that really answers my question.

I need to review your settings, which i will try tomorrow.

RAMilewski, 2021-01-29:
Many thanks for all the help!