FormerLurker/Octolapse #675: Octolapse and Pi Cam v2

krepitus, 2021-01-25:
I followed the Wiki and Ocrolapse isn't recognizing the Pi Cam V2.   When I click the edit button under Time Lapse Preview it doesn't recognize the camera.  There is no Switch to Default Page tab and selecting the Switch to Custom Page tab produces an error message that says 

Unknown webcam
There is no custom control for your webcam model at the moment!

Thank you for any assistance.

#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: ___REPLACE_THIS__OCTOLAPSE_VERSION_GOES_HERE

#### Version of OctoPrint

OctoPrint Version: ___REPLACE_THIS__OCTOPRINT_VERSION_GOES_HERE

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: ___REPLACE_THIS__YES_OR_NO

#### What were you doing when the problem occurred

1. ___REPLACE_THIS__STEP_ONE_GOES_HERE
2. ___REPLACE_THIS__STEP_TWO_GOES_HERE
3. ___REPLACE_THIS__STEP_...

#### What should have happened?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: ___REPLACE_THIS__OS_NAME_GOES_HERE
Os Version: ___REPLACE_THIS__OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

jsdokos, 2021-01-26:
Can you see the webcam on the control page? I have the same camera and I am using the Webcam - Default OctoPi 0.16.0. I can share the exact specs if you would like.

krepitus, 2021-01-26:
I can see the image but the quality is pretty bad no matter what settings I
try.  If you don't mind.  I'd give yours a try.

jsdokos, 2021-01-26:
Here is the profile:

[Camera_Profile.zip](https://github.com/FormerLurker/Octolapse/files/5875702/Camera_Profile.zip)

FormerLurker, 2021-01-26:
Can you take a snapshot of the image preference screen in Octolapse?

Next, go to the following url, save as a file, then attach:

http://{ip_of_pi}:8080/input.json

or maybe 

http://{ip_of_pi}/webcam/input.json

krepitus, 2021-01-27:
I'm working on your request.

krepitus, 2021-01-28:
Sorry to keep bothering you.  After an Octoprint update I get this message
when I try to enable Custom Image Preferances

An unknown webcam server type 'Werkzeug/1.0.1
Python/3.7.3' is being used for the 'Raspberry Pi
Camera V2' camera profile. Currently only
MJPEGStreamer is supported. Unable to apply custom
image preferences.

FormerLurker, 2021-01-28:
Ok, so you are using either octoprint anywhere or the spaghetti detective probably, right?  If so, go to the settings for those plugins and enable compatibility mode, then reboot.

krepitus, 2021-01-28:
I'm using Spaghetti Detective.  I'll try that.  I can't thank you enough
for your help.

krepitus, 2021-01-28:
With compatibility mode turned on I can use Octolapse and access camera
settings but I don't have a default settings tab, just a Switch to Custom
Page tab that gives me an error message when I try to use it.

There is no custom control for your webcam model at the moment!

With compatibility mode set to auto I get this error message.

There was a problem retrieving webcam settings from the server.

FormerLurker, 2021-01-28:
There may well be no custom control page for your camera.  I basically added 'extra nice' pages for a few popular cameras.  If ANY features or options for the camera you are using changes (perhaps v2 -> v2.1 or some such nonsense), the custom page will not be loaded.  Attach your input.json file (as a link please, try gist.github.com for that), and I can see what the difference is.  Occasionally I can modify the existing page to work depending on what is different.  Otherwise I'll have to create an entirely new page, which usually requires that I actually have the camera in question.

FYI, you will NOT be missing any controls, even if the custom page isn't available.  You just won't have as nice organization and validation.  The 'default' page is built dynamically based on the controls reported by input.json.

FormerLurker, 2021-01-28:
Oh, FYI, I have no idea how TSD determines if compatibility mode should be used.  It is absolutely necessary for Octolapse.  The author of TSD and I had some discussions about it a while back, and determined that there is no reasonable way to make the custom image controls (at least as they are now written) work with his custom streaming server.  Also, the custom server has much lower bandwidth usage, but requires significantly more time to acquire a snapshot (meaning the camera delay needs to be increased a lot), so if you are using octolapse in any way, you will want compatibility mode on.  If you aren't using Octolapse, you can turn that off and save some bandwidth.

krepitus, 2021-01-28:
OK.  Thank you.  I started down this road so I could try to get a better
image with the Pi cam.  I print some things for my nephew's special ed
class and I try to include a timelapse with it so the kids can see their
model get printed.  I'll leave compatibility mode on and I may have to buy
a USB cam when I can set some money aside.  Thank you very much for the
help.

FormerLurker, 2021-01-28:
Attach an image of your image preferences.  You can dramatically improve the quality by increasing the bitrate in the preferences screen.

krepitus, 2021-01-28:
[image: image.png]

FormerLurker, 2021-01-28:
You may need to acutally sign into github to do that.

krepitus, 2021-01-28:
[image]

krepitus, 2021-01-28:
I think I've done it properly this time.

FormerLurker, 2021-01-28:
So you are following the guide for improving the image quality [here](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Enabling-Camera-Controls#step-7---dialing-in-your-camera)?  I'm interested in any feedback you might have.

krepitus, 2021-01-28:
> Yes.  That's the one I'm using.

The guide was great.  I was only confused about why I was setting the Pi
cam up as a USB camera since it uses a different type of connection.  I
don't know if the explanation would be worth getting into but I went back a
few times to make sure I was using the correct guide.

krepitus, 2021-02-16:
I just wanted to take a minute to thank you for all the help you gave me.
Octoprint is an amazingly useful program.