FormerLurker/Octolapse #678: [Request] Installation error, apt-get g++?

Eibwen, 2021-01-31:
#### If this is a feature request describe it here

I tried to install Octolapse on an old computer with a i686 processor. And the installation failed with the error `i686-linux-gnu-gcc: error trying to exec 'cc1plus': execvp: No such file or directory`

It took a minute to re-read the installation guide, then find and sort through the very long error log to find that error.  But once I found that, found the solution quickly:

Solution that worked was to `sudo apt-get install g++` due to missing compiler stuff

So my request is that that command is listed as a requirement on some of the Getting Started or Installation documentation pages, **if** it would be happening commonly enough. I'm not certain if I ran into this because I'm using an odd processor family, or because I used a slimmed down linux, but I already had many other things compile fine on the system.  However often that this occurs, it could be helped by a simple documentation addition

An addition such as:

> Troubleshooting: If any compiler errors occur, this package may be required: `sudo apt-get install g++`

Or if you don't imagine it happens enough to be helpful, feel free to close this request

#### Version of Octolapse
 
Octolapse Version: 0.4.1

#### Version of OctoPrint

OctoPrint Version: 1.5.2

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: ___REPLACE_THIS__YES_OR_NO

#### What were you doing when the problem occurred

1. ___REPLACE_THIS__STEP_ONE_GOES_HERE
2. ___REPLACE_THIS__STEP_TWO_GOES_HERE
3. ___REPLACE_THIS__STEP_...

#### What should have happened?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: ___REPLACE_THIS__OS_NAME_GOES_HERE
Os Version: ___REPLACE_THIS__OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2021-01-31:
> Or if you don't imagine it happens enough to be helpful, feel free to close this request

No, this is an excellent issue.  I've been plagued with multi-platform support issues since including custom extensions, and anything I can do to make that better would be great.  I'd love to solve this in the setup script, but access is limited (like installing packages and such).  I will keep trying to improve this.

FormerLurker, 2021-01-31:
Check out [this addition to readme.md](https://github.com/FormerLurker/ArcWelderPlugin/tree/rc/maintenance#troubleshooting) (it is item 6 in the list), and see if you think it solves your issue:

Eibwen, 2021-02-01:
I would have thought it would apply under item 3?  My issue was that this was required before Octoprint's Plugin Manager could get octolapse installed.  The failure messages which popped up on the Announcement errors of OctoPrint were very unhelpful (something like "Unknown plugin: error")

I'm not aware of what the _ArchWelder_ plugin is, so wouldn't have thought that item was relevant

Here is the `tail` of the `plugin_pluginmanager_console.log` from Octoprint:

```
2021-01-31 01:55:46,024 ! set build/lib.linux-i686-3.6/octoprint_octolapse/_version.py to 'v0.4.1'
2021-01-31 01:55:46,024 ! running build_ext
2021-01-31 01:55:46,025 ! Compiling Octolapse Parser Extension with <distutils.unixccompiler.UnixCCompiler object at 0xb6e6f72c>.
2021-01-31 01:55:46,025 ! building 'GcodePositionProcessor' extension
2021-01-31 01:55:46,026 ! creating build/temp.linux-i686-3.6
2021-01-31 01:55:46,026 ! creating build/temp.linux-i686-3.6/octoprint_octolapse
2021-01-31 01:55:46,027 ! creating build/temp.linux-i686-3.6/octoprint_octolapse/data
2021-01-31 01:55:46,027 ! creating build/temp.linux-i686-3.6/octoprint_octolapse/data/lib
2021-01-31 01:55:46,028 ! creating build/temp.linux-i686-3.6/octoprint_octolapse/data/lib/c
2021-01-31 01:55:46,028 ! i686-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/home/octo/OctoPrint/include -I/usr/include/python3.6m -c octoprint_octolapse/data/lib/c/gcode_position_processor.cpp -o build/temp.linux-i686-3.6/octoprint_octolapse/data/lib/c/gcode_position_processor.o -O3 -std=c++11
2021-01-31 01:55:46,028 ! i686-linux-gnu-gcc: error trying to exec 'cc1plus': execvp: No such file or directory
2021-01-31 01:55:46,029 ! error: command 'i686-linux-gnu-gcc' failed with exit status 1
2021-01-31 01:55:46,029 ! ----------------------------------------
2021-01-31 01:55:46,030 ! ERROR: Command errored out with exit status 1: /home/octo/OctoPrint/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-igdtjyrp/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-igdtjyrp/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-nuzutl6n/install-record.txt --single-version-externally-managed --compile --install-headers /home/octo/OctoPrint/include/site/python3.6/Octolapse Check the logs for full command output.
2021-01-31 01:55:46,033 > Running setup.py install for Octolapse: finished with status 'error'

```

FormerLurker, 2021-02-01:
Whoops, I was mass handling issues and for whatever reason I just missed that this was Octolapse.  ArcWelder (another one of my plugins) also requires a c++ compiler.  

I will alter that ASAP and will post back.  It will probably be a couple days.  Sorry about that!!!

Eibwen, 2021-02-01:
Ah that makes a lot more sense, definitely can understand that sort of mistake

ArcWelder seems very cool (and _perfectly_ named), almost wish I had a use case for it since it seems like a very elegant way to avoid gcode streaming issues

FormerLurker, 2021-02-01:
@Eibwen, I wish I had come up with the name.  A patron suggested it to me, and I too thought it was perfect.  Also, just FYI, recent testing is showing that even SD prints benefit, especially at high speed or high resolution.  This was even shown on a 32 bit board, though the speeds were pretty high before substantial effects were shown in this case.  For 8 bit boards like I have, it's pretty substantial across the board.  I didn't realize I was encountering planner starvation at all, because I had nothing to compare it to.  When the planner starts to run dry on my firmware, it automatically starts slowing down, and I believe this is where I see the benefits.  I'm seeing improvements on all my prints, especially the artifacts that appeared to be totally random.

I guess what I'm saying is that it's worth a shot :)

Eibwen, 2021-02-01:
@FormerLurker hah, yeah I said no use case because I'm not using a Raspberry Pi on the Octoprint side, and I recently put a BigTreeTech board into my Ender3, so I assumed the 32bit would be handling it easily.

But hey, it doesn't hurt, and re-compiling the firmware with arcs enabled is pretty fun.  Do arcs on 3d printers make any different noise, or is the planner already optimizing that to a degree and so the actual signal to the motor is about the same?  That would be a selling point for me, if it makes different interesting noises at all :)

FormerLurker, 2021-02-01:
> Do arcs on 3d printers make any different noise
They do!  Or, more precisely, they can.  It depends on how arcs are configured.  Using smaller interpolated segments will definitely change things.  Try a setting for mm_per_arc_segment of 0.1 (your board may not be able to handle this, or you might be able to go smaller.  The lower the value, the more power you will need).  Leave min_arc_segments alone.  I'm interested to hear what happens :)