FormerLurker/Octolapse #679: Cura 4.7.0 - Octolapse Automatic configuration: Adding the configuration commands crash cura

ldiegos, 2021-01-31:
#### If this is a feature request describe it here

On Cura 4.7.0 when try to add the configuration commands to the printer start g-code and restart cura, it starts to raise errors on loading and fail to load correctly the print profile. It seems that it detect that the configuration parameter is being assigned over there without all the print profile loaded before. See attached pictures.

I wonder if you could change this "Current Slicer Settings" "Automatic" to search at the end of the g-code, because cura always add the print profile to all the g.codes:

>;SETTING_3 {"extruder_quality": ["[general]\\nversion = 4\\nname = PETG-Layer0.2
>;SETTING_3 -InitialLayer0.12\\ndefinition = creality_ender3pro\\n\\n[metadata]\\
>;SETTING_3 nquality_type = standard\\ntype = quality_changes\\nintent_category =
>;SETTING_3  default\\nposition = 0\\nsetting_version = 14\\n\\n[values]\\naccele

#### Version of Octolapse
 
Octolapse Version: Octolapse: 0.4.1 

#### Version of OctoPrint

OctoPrint Version: OctoPrint: 1.4.0 

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

1. Follow instructions from here: https://github.com/FormerLurker/Octolapse/wiki/V0.4---Automatic-Slicer-Configuration#install-the-cura-settings-script
2. Add the commands to the machine_start_gcode
3. Restart cura.
4. Cura crash and can't load all the stuff.

#### What should have happened?
Slice and print without any issue.

#### What happened instead?
Versión 4.7.0 from cura, try to load and assign the configuration parameters from the profile before the profile is loaded, so it crash.

#### Operating System running OctoPrint and Octolapse

OS Name: ¿raspbian?
Os Version: Whatever is with the OctoPrint: 1.4.0 

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: Can't slice due to the Cura Crashes.

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: 

Cura ok: [image]
With the parameters added to the printer:
[image]
After restarting, the problems raise, with print profile not loaded correctly.
[image]
The configuration is ok, because the cura itself crash after loading the printer configuration.
[image]

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2021-01-31:
The section you are pointing out only shows the differences between your current settings and your original profile, so unfortunately this will not work.  The only way to reliably get all the cura settings is to add the script from the wiki you followed.

Regarding the crash, my install does not do this (I just tried it again).  This seems like a cura bug.  Give me a little time to review, because this is a bit complex.

ldiegos, 2021-01-31:
> The section you are pointing out only shows the differences between your current settings and your original profile, so unfortunately this will not work. The only way to reliably get all the cura [...]

Which version of cura do you have? Just for test with it. I'm not very confortable updating to 4.8, but i will give a try if you think so.

FormerLurker, 2021-02-01:
So, I have 4.5, 4.7 and 4.8.  I did notice that I have to scroll down before it shows the start gcode.  it appears to be hidden until I do that.  Maybe double check?  I think something else is going on.

ldiegos, 2021-02-01:
> So, I have 4.5, 4.7 and 4.8. I did notice that I have to scroll down before it shows the start gcode. it appears to be hidden until I do that. Maybe double check? I think something else is going on.

Before opened the issue i try everything, double click, add command by commmand and so on.

ldiegos, 2021-02-01:
Tomorrow i will update to cura 4.7.x and i will come back again with my results.
Thank you so muy in advance for your time

ldiegos, 2021-02-01:
You can close the error/issue/bug I found the reason:

Changes on cura:
4.7.1
For an overview of the new features in Cura 4.7, please see this video: Change log overview
Bug fixes
- Fixed a crash when duplicating a built-in profile.
- **Having an equals symbol in your start or end g-code would cause part of that g-code to disappear and could cause a crash when loading a model. This is fixed now.**
- ...

closed 2021-02-01 by ldiegos

FormerLurker, 2021-02-01:
Wow, that's an interesting find!  Feel free to post another issue if you encounter any more problems.  Thanks for the update, and enjoy Octolapse!