FormerLurker/Octolapse #680: Camera setting values should allow increment of 1

bgie, 2021-02-01:
#### If this is a feature request describe it here

Older version(s) allowed setting of camera focus in increments of 1.
I remember 27 being a good value for my camera.
I haven't used octoprint in a while. Now, the current version only has a slider, and I can move only in increments of 5.
I am having trouble getting perfect focus, the ideal focus value is somewhere between 25 and 30 but it is no longer possible to input this.

#### Version of Octolapse
 
Octolapse Version: 0.4.1

#### Version of OctoPrint

OctoPrint Version: 1.5.3

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: ___REPLACE_THIS__YES_OR_NO

#### What were you doing when the problem occurred

1. ___REPLACE_THIS__STEP_ONE_GOES_HERE
2. ___REPLACE_THIS__STEP_TWO_GOES_HERE
3. ___REPLACE_THIS__STEP_...

#### What should have happened?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: ___REPLACE_THIS__OS_NAME_GOES_HERE
Os Version: ___REPLACE_THIS__OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2021-02-01:
So, each control with a slider has a min, max, and step associated with it.  This information is queried from the camera, and the resulting data is used to dynamically generate a control page.  There are a few custom camera control pages in Octolapse, and it's possible I made a mistake in the min, max, or step values, but I don't know what camera you are using exactly, so I can't rule that out.

You can double check to see if the value you remember (27) is a valid value for your camera by entering the following URL:  ```http://{IP_OF_PI}:8080/input.json```

That should result in a text file (.json) containing every control your camera supports.  Find the name of the control you are trying to change and look for the min, max, and step value to verify that Octolapse is handling it correctly, or to show that there is an error in the values Octolapse is using.

You can also use the  ```http://{IP_OF_PI}:8080/control.htm``` page to manually send values.  That page has no data validation at all, so sometimes it's a bit tricky to see what's happening.  However, you can try different values there and see if they have any noticeable effect.

Another option would be just to allow any value or increment.  If you look through the closed issues, you'll find several that are very similar to this one.  I've experimented with this, and have decided that it's far better to restrict the input for the following reasons:

1. If given an invalid value, some cameras will simply do nothing.  This is very confusing for users.  Let's say your camera only allows focus values in increments of 5, so you first enter 25 and see the focus change.  Then you might enter 27 and see no effect.  Later 37, then 47, etc, but the focus never changes.  I'm guessing you would be surprised to see a dramatic change if you entered 50.  However, my camera seems to focus on the closest valid value, so that wasn't an issue for me.
2. Some cameras will have corrupted images or even crash if incorrect values are sent to them.  This is obviously the biggest problem.

I know that is a lot of info, but I wanted to give you enough to decide if this is a problem with Octolapse (maybe a typo on a custom page), and incorrect input.json report (maybe it's saying increments of 5, but increments of 1 are really allowed), or maybe it's something else entirely.

In any case, thanks for the report.  we'll figure this out!

bgie, 2021-02-01:
Looks like the step size is correctly handled by octolapse.
The json file contains a "step": "5" in the "name": "Focus (absolute)" category.

I guess I might have been mistaken about the 27 value actually being different from 25 or 30.
Anyway I think your decision to restrict the input is the best one. 

Guess the issue can be closed. 
Thank you for the quick response, 
I'll give a little support on patreon.

closed 2021-02-01 by bgie

FormerLurker, 2021-02-01:
Hey, @bgie, thank you for your support!  I am really glad you posted, and am more than happy to discuss and reexamine my thought process.  I'm a learn-as-you-go kind of person, and one thing I've learned over the last few years is good to re-examine everything occasionally.  I'm glad we were able to do this, and to come to an agreement.  I would still love to come up with a way to show this info to users so they are not surprised at the unexpected behavior.  I haven't really come up with a good way to do this, so if you have any ideas I'm all ears :)