FormerLurker/Octolapse #683: No videos to download after finished printing

gilbertlucas46, 2021-02-05:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: Octolapse 0.4.1

#### Version of OctoPrint

OctoPrint Version: Version 1.5.3

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. Printed small file to test
2. Clicked the "Videos and Images"  button in octolaps after the printer finished printing
3. Modal opens and the message is "There are no files available." it's confusing because I can see the camera snapping some pictures.

#### What should have happened?
I should be able to download the video
after its done printing,

#### What happened instead?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: UBUNTU
Os Version: Ubuntu 20.04.1 LTS

#### Printer model & used firmware incl. version

Printer Model: Creality Ender 5 Plus
Printer Firmware Version: Ver 1.70.1 BL

#### Browser and version of browser, operating system running browser

Browser: Chrome
Browser OS: Version 87.0.4280.88 (Official Build) (64-bit)

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://drive.google.com/file/d/1nFh-Vqqqivsvxhkwc8RdqEYmT5wNt1qV/view?usp=sharing

#### Link to settings.json

Link to settings.json with all passwords removed: https://drive.google.com/file/d/1q7w2QA1RJiGbesEjbMWRgP5L-2o1n7gc/view?usp=sharing

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: 
https://prnt.sc/yd44z4
https://prnt.sc/yd48gl

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2021-02-06:
Hmm.. That shouldn't happen.  Let's capture a log.  Let's do it in test mode to save you some time and filament;

1. Enable test mode on the octolapse tab.
2. Select the 'Debug' profile in your 'Logging' profiles on the Octolapse tab.
3. Print 3 layers of any fast file, like a calibration cube.  Note that your printer will not warm up and will not extrude any filament.  This is very convenient for us :)
4. Stop the print after Octolapse captures 3 snapshots (somewhere on layer 3 this will happen)
5. Select the 'Log all errors' Logging profile in the Octolapse tab (no reason to keep logging and drain your SD and CPU).  
6. Click the edit pencil to the right of the logging profile.
7. Click the 'Download Log' button.
8. Post the log to gist.github.com (you already have an account there since you have a github account) and put a link to that gist here.

That will hopefully give me enough info to figure out what's going on.  Also, you might try rebooting after this to see if you notice a new button on the Octolapse tab called 'Failed Renderings'  In some cases this might not show up right away if there was some kind of exception in the rendering thread, so it's worth a shot.

FormerLurker, 2021-02-06:
Oh, I forgot one more important step:

9.  Disable test mode

You will probably want to actually print something at some point, so it's important to turn that off when not using it, lol

gilbertlucas46, 2021-02-06:
Thank you for the reply
https://gist.github.com/gilbertlucas46/f2cbd1036ffb2af34a69465741a156c0

gilbertlucas46, 2021-02-06:
I changed camera settings

Stabilization: from `center` to `back right`
Rendering: From `MP4 30FPS` to `MP4 60FPS`

[image]

Now I have video preview

[image]

I will try in a real print

FormerLurker, 2021-02-06:
I haven't looked at your settings yet, but maybe rendering was disabled?  The logs looked ok.

closed 2021-02-06 by gilbertlucas46