FormerLurker/Octolapse #684: Octolapse trying to render with wrong configured FFMPEG path

jwvanderbeck-Unity, 2021-02-05:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: v0.4.1

#### Version of OctoPrint

OctoPrint Version: 1.5.3

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: Error Only logging was enabled.

#### What were you doing when the problem occurred

1. Set custom FFMPEG path in settings, as the location changed from the default in the latest OctoPi
2. Did a print, Octolapse recorded snapshots
3. After print finished, received an error that rendering failed
4. Tried to manually initiate render, same fail
5. Looked at log, log indicates Octolapse is trying to use FFMPEG from the previous default path, not the newly configured one.

#### What should have happened?
Video should have been rendered using the FFMPEG in the configured path.

#### What happened instead?
Octolapse tried to use FFMPEG from a different path.  The path it tried to use is the old default path.

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.17.0

#### Browser and version of browser, operating system running browser

Browser: Opera 73.0.3856.344
Browser OS: Windows

#### Link to settings.json

Link to settings.json with all passwords removed: [settings.zip](https://github.com/FormerLurker/Octolapse/files/5934294/settings.zip)

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://gist.github.com/jwvanderbeck-Unity/cb258a24ed7a2c8bdce9c22ede579ec5

#### Link to octoprint.log

Link to octoprint.log:  https://gist.github.com/jwvanderbeck-Unity/c46c85d2a0a9b7f56904bc7be4260f8d

#### Link to contents of Javascript console in the browser

Link to javascript console output: https://gist.github.com/jwvanderbeck-Unity/677957a9f34f61eba8b31c8b8ceddf96

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: 
[image]

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2021-02-05:
I have seen this issue before, and it is definitely a bug.  Will fix as soon as i can find some time.  Will update when i do.  Reboot, and it should start working.  You can re-render the failed timelapse after you reboot.

jwvanderbeck-Unity, 2021-02-05:
Will try a reboot, thanks.