FormerLurker/Octolapse #685: Cura "Pause at Height" kills timelapse

mpaw, 2021-02-07:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4.1

#### Version of OctoPrint

OctoPrint Version: 1.5.2

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

1. Use Cura `Post Processing` plugin to add `Pause at Height` script
2. Select `Marlin (M0)` method
3. Slice your model and upload to OctoPrint
4. Begin printing the model
5. When it reaches the layer you added the pause at, use Octoprint to resume the print
6. When print finishes, Octolapse video is only up until the point where the pause happened. Nothing after

It looks like this was reported before but the issue was closed: https://github.com/FormerLurker/Octolapse/issues/341

I tried finding a solution before posting this issue but couldn't find one.

#### What should have happened?
The timelampse should have been of the full print, not just until the pause

#### What happened instead?
The video stopped at the layer pause

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.18.0

#### Printer model & used firmware incl. version

Printer Model: Ender 3 Pro
Printer Firmware Version: Marlin 2.0.6.1 custom build

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: Easily reproducible on any print

#### Link to settings.json

Link to settings.json with all passwords removed: https://gist.github.com/mpaw/b2804a507ee66cce5e340cfc680c84e8

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2021-02-07:
Yeah, that issue was closed before I had a chance to look at it.  If you can capture a log, I will be able to take a look much sooner.  To do this: 

1. switch the Octolapse logging profile to debug
2. start a print (something V small like a calibration cube), and let it run a few layers
3. do your filament switch (guessing you have to set this up in cura beforehand)
4. run a few more layers
5. cancel the print
6. very important - set your logging profile back to 'Log all Errors' and turn Test Mode off!
7. download the log (edit debug profile, click 'download log')
8. post the log to gist.github.com and drop a link in here.

Thanks!

mpaw, 2021-02-08:
@FormerLurker So ran it again with the debug logging on and now can't reproduce the issue. Just in case it might be informative anyway, here is the log:

https://gist.github.com/mpaw/cf28e3fc9f8cec30720a697161f81afd

Here is the resulting timelapse which looks fine
https://www.dropbox.com/s/um8k63iuirtqb20/Calibration_Cube_20210207221007.mp4?dl=0

I'll try a couple more prints with the debug logging turned on to see if I can reproduce it again.

Thanks for your help, appreciate it.

FormerLurker, 2021-02-08:
Ok, if it is hard to reproduce, that's probably why I haven't seen it yet.  I've got little doubt that what you've encountered is a real issue, just don't know where it is coming from without catching it in the act so to speak.

> I'll try a couple more prints with the debug logging turned on to see if I can reproduce it again.

That is awesome, but keep in mind you will be getting a crazy big log, and your performance will suffer somewhat.  I would disable logging if you do any critical prints.  I would advise you to only turn on specific loggers, but unfortunately the problem is likely to be in the bits of the code that do the most logging.

Also, you may want to clear your log between prints.  This will reduce the file size and make it more manageable.  Just edit your octolapse logging profile and click the 'clear all logs' button to do this.

Thanks!