FormerLurker/Octolapse #688: Cannot Access Pi Camera Additional Settings

johnbozuk, 2021-02-11:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4.1

#### Version of OctoPrint

OctoPrint Version: 1.5.3

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. Followed guide to modify octopi.txt file and updated as per https://github.com/FormerLurker/Octolapse/wiki/V0.4---Configuring-a-Raspberry-Pi-Camera
2. Updated /etc/modules to include bcm2835-v4l2
3. Go to Octolapse settings for pi camera and click custom image preferences and get error

#### What should have happened?
Should see the additional custom image settings

#### What happened instead?

Error Below.

Unable To Enable Custom Preferences
An unexpected error was raised while testing custom image preferences for the 'Pi Camera v2.1' camera profile.

#### Operating System running OctoPrint and Octolapse

OS Name: Octopi
Os Version: 0.18.0

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://pastebin.com/GnDQ3P2H

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

GeekDad63, 2021-02-17:
I am getting the same error with my raspberry pi camera and trying to click custom image preferences. 

_Unable To Enable Custom Preferences
An unexpected error was raised while testing custom image preferences for the 'Webcam - Default OctoPi 0.16.0' camera profile._

johnbozuk, 2021-02-17:
Remove octolapse and try worked for.me, but I have now gone to usb camera
as can't get both working

FormerLurker, 2021-02-21:
Hey, sorry it took me a while to get to this, I've been getting hammered this week :)

So, @johnbozuk, there were lots of errors in there, but it looks like you switched your port from 8888 to 8080 at some point and started to get a different error.  You also mention you switched to a USB camera and can't get both working.  What do you mean by 'both'?  Are you running a raspberry pi camera and a usb camera?  In any case, let's do one thing at a time.  I can help you get the picam working, but it will take a bit of back and forth.  The same applies to you @GeekDad63.

I will say the steps are pretty complicated to get the picam working.  I'm not sure why everything is so non-standard with those cameras.  However, please provide the following:

1. Your entire octopi.txt file
2. the /etc/modules file
3. Your octolapse settings.json file (you can export this from the Octolapse tab by clicking the 'Gear' icon, then clicking the export settings button).
4. Try navigating to http://{ip_of_pi}:8080/control.htm (raplace {ip_of_pi} with the appropriate ip address) and take a screen shot.  Note, your camera may be on a different port.  You can also try http://{ip_of_pi}/webcam/control.htm
5. Try navigating to http://{ip_of_pi}:8080/input.json.  Just like item 4, you can try the http://{ip_of_pi}/webcam/input.json version.

Post all items to gist.github.com and drop a link to them here please.  The exception is item 4, which you can paste right into the issue (but not via email, it doesn't work).

It's possible that I won't be able to tell what's wrong even with all of that info.  In that case I'll ask you to create a special log file, but we'll deal with that step only if it is necessary.

Thanks for your patience!

GeekDad63, 2021-02-21:
I cannot find the /etc/modules file. It’s not on the boot disk.

> On Feb 20, 2021, at 7:41 PM, FormerLurker <notifications@github.com> wrote:
> 
> the /etc/modules file

johnbozuk, 2021-02-21:
Hi, I managed to get the pi camera working in the end, had to uninstall octolapse, then reinstall after doing all the modifications in the octopi.txt file, the USB camera would not work if I had the pi camera activated in the raspiconfig options, So at the moment its one or the other, not sure why, but don’t really need two cameras working so problem fixed for me, but I do get the advanced options for the camera now. Thanks John Sent from Mail for Windows 10 From: John BickfordSent: 21 February 2021 21:28To: FormerLurker/OctolapseCc: johnbozuk; MentionSubject: Re: [FormerLurker/Octolapse] Cannot Access Pi Camera Additional Settings (#688) I cannot find the /etc/modules file. It’s not on the boot disk.>

GeekDad63, 2021-02-21:
Problem solved. I switched back to camera:(usb)

FormerLurker, 2021-02-22:
@geekdad63, excellent!!!  Post a vid next time you take a timelapse.  Always interested in those.

@johnbozuk, you may want to walk through the wiki once more.  I suspect you missed a step

ArcAIN6, 2021-03-02:
I just ran into a similar situation on the Pi3 with a raspicam v1.6
Attempting to enable the additional settings throws an error popup
Hitting the ? next to the error, says to load bcm2835-v4l2 into modules, doing so and rebooting, camera appears to no longer work at all. 

Tryign to hunt down the issue, i've come across a multitude of posts both on, and off of the official raspberry pi forums, detailing an issue with the firmware, and calling specific modules. 

Seems there's a problem with the module in raspberry pi firmware that specifically affects the pi 3, and pi zero (and as far as i'm aware, hasn't been addressed)

When you load the bcm2835-v4l2 module and reboot, all functionality of the camera ends. 
The errors below are shown in dmsg once booted up:

```
] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
] usbcore: registered new interface driver brcmfmac
] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.raspberrypi,3-model-b.txt failed with error -2
] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Oct 22 2019 01:59:28 version 7.45.98.94 (r723000 CY) FWID 01-3b33decd
] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled

```

Commenting out the entry in modules returns camera image. 

Unsure where to proceed from here

GeekDad63, 2021-03-02:
> Did you change your camera type to USB in octopi.txt?

Yes I was having issues with it set to USB but then I switched it back and all is working as it should. I may have changed some other things like forgetting to remove some hash marks.

John

FormerLurker, 2021-03-02:
@GeekDad63, ok, I just flashed a fresh version of OctoPi, and will update the environment and try the driver myself to see if something has changed.  This will take a few days for me to get to (work is busy). I really hope there were no breaking changes, but it's possible.

ArcAIN6, 2021-03-03:
Sorry for the late reply, looks as if i did miss the step of setting it to usb, once i did that, everything's working . Thanks :)

FormerLurker, 2021-03-03:
Whew 😅