FormerLurker/Octolapse #689: Octolapse 0.4.1 shows is not primed under print state and not taking any snapshots.

anivdas, 2021-02-14:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4.1

#### Version of OctoPrint

OctoPrint Version: 1.5.3

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: No

#### What were you doing when the problem occurred

1.  Created a camera profile as per the raspi cam v2 settings in wiki, able to see the camera image under control and also in customize panel. Also created the profile for Qidi X-Max printer to the best of my knowledge and set priming height to 1.0mm as it didn't work with 0.75mm default either. Trigger is set to classic gcode , (P1 42 on layer change).
2. print is successful but "is primed" under print state continues to remain in the red. 
3. Tried overriding the octoprint printer profile but issue persists.

#### What should have happened?
Should have taken snapshots to create timelapse, 0 snapshots taken for a 1499 layer print.

#### What happened instead?
Print was successful but timelapse was not created, no preview either.

#### Operating System running OctoPrint and Octolapse

OS Name: Octopi
Os Version: 0.18.0

#### Printer model & used firmware incl. version

Printer Model: Qidi X-MAX
Printer Firmware Version: 4.2.6.5

#### Browser and version of browser, operating system running browser

Browser:Google Chrome Version 88.0.4324.150 (Official Build) (64-bit)
Browser OS: Windows 10 Professional 20H2

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://gist.github.com/anivdas/b1e928a53cbc0416475ac5c3878e3c28

#### Link to settings.json

Link to settings.json with all passwords removed: https://gist.github.com/anivdas/ac59a6edf8ec2efbc33d6974167ab0af

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

I just did as Early Supporter - Level 2

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2021-02-14:
Hi there!  It is late, but i wanted to suggest changing the priming height to 5 or 10mm.  If you post your start gcode from your slicer, i will be able to tell you for sure.

FormerLurker, 2021-02-14:
Oh, I will be around tomorrow AM and will follow up then.

anivdas, 2021-02-14:
Thanks so much for the quick response and no rush at all, , I have a print in progress for another 18 hours or so and wont be able to try out the recommendation until it's done.

Here is the start script under Simplify 3D

G92 A0 B0
G28 X0 Y0
G0 X-1000 Y-1000 F1800
G28 Z0
G1 X0 Y0 Z50 F3600
G1 X0 Y0 F3600
G92 E0
G92 A0 B0
G0 X0 Y4 Z0.3 F3600
G1 X160 Y5 F2400
G1 X5 A13 F2400
G92 A0 B0
G1 X5 F2400

Thanks!

FormerLurker, 2021-02-14:
Ok, your printer has no priming routine at all (which is pretty odd).  Set priming height to 0 to disable priming detection, and you should be good.

anivdas, 2021-02-14:
Thanks so much for looking into this. I will give it a shot in few hours after the current print is done. I am also thinking of using an optimized start/end gcode scripts from someone else.

it does a 2mm prime so I will play with the number after I test it works fine with 0/prime detection disabled.

START GCODE

G92 A0 B0 ; Not sure what they use A and B for
M140 S{material_bed_temperature_layer_0}
M104 S{material_print_temperature_layer_0} T0
G28 X0 Y0 Z0 ; Home XYZ
G0 X250 Y4 F1800 ; Move head to front
G1 X250 Y4 Z150 F3600 ; Move bed to midway
M190 S{material_bed_temperature_layer_0} ; Wait for temps
M109 S{material_print_temperature_layer_0} T0
G92 E0
G1 E2 F300 ; Move filament to melt zone
G92 E0 ; Set extruder position to 0
G92 A0 B0 ; Not sure why they are repeating this
G0 X300 Y4 Z{layer_height_0} F3600
G1 X150 A13 F2400
G92 A0 B0 ; And again not sure why they are repeating this, but left it from default

M201 X1000 Y1000 Z200 E1000 ; sets maximum accelerations, mm/sec^2
M203 X200 Y200 Z10 E120 ; sets maximum feedrates, mm/sec
M204 P1000 R1000 T1000 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
M205 Z0.20 E5.00 ; sets the jerk limits, mm/sec

M201 X1000 Y1000 Z200 E1000 ; sets maximum accelerations, mm/sec^2
M203 X200 Y200 Z10 E120 ; sets maximum feedrates, mm/sec
M204 R1000 ; sets retract acceleration (R), mm/sec^2
M205 Z0.20 E5.00 ; sets the jerk limits, mm/sec

;END GCODE

G92 Z0 E4 ; Set z and e positions
G0 Z10 F600 ; Move print down 10mm from nozzle
M104 S0 ; Turn off heater
M140 S0 ; Turn off bed heater
G1 E-3 F300 ; Retract filament to keep from cooking in melt zone
G0 Z300 F600 ; Lower the bed
M84

FormerLurker, 2021-02-15:
That gcode looks a bit sketchy to me.  Not sure what ```G92 A0 B0``` is supposed to be doing.  As far as I know, A and B are not valid parameters, at least for any firmware I'm aware of.  Also, are you certain the feedrates/accelerations are correct for your printer?  That section looks like it was taken right out of the Prusa's MK2/MK3 start gcode, and modified slightly (which is fine if the values are right).  Anyway, I'd figure out what is going on with the G92 A0 B0.  Generally one would NOT set any axis with G92 except E before homing.