FormerLurker/Octolapse #690: [BUG] Cannot save profile

numpty9989, 2021-02-16:
#### Version of Octolapse
 
Octolapse Version: 0.4.1

#### Version of OctoPrint

OctoPrint Version: 1.5.3

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: yes

#### What were you doing when the problem occurred

1. trying to save printed r profile

#### What should have happened?
saved the profile

#### What happened instead?
nothing

#### Operating System running OctoPrint and Octolapse

OS Name: Ubuntu
Os Version: 18.04

#### Printer model & used firmware incl. version

Printer Model: Creality Ender 3
Printer Firmware Version: 2.41 TH3D

#### Browser and version of browser, operating system running browser

Browser: Firefox
Browser OS: Windows

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: https://youtu.be/NVzllt2IKLk

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2021-02-17:
Your video is marked as private.  Also, please check the javascript console in your browser (f12) and check for any errors.  In the end, I might need some log files, but we'll see.

numpty9989, 2021-02-17:
Sorry my thing said unlisted haha  fixed the vid now.  I’ll see if can get some log files for you.

FormerLurker, 2021-02-17:
OK, definitely check the js console.  My initial guess would be there is some other plugin interfering somehow, but who knows.  

Maybe you can check the 'customize' check box and see if there are any validation issues?  maybe i messed up the default profile somehow. 

Also, just FYI, I primarily develop on Windows in Firefox, so it's pretty odd that there would be a problem :(

numpty9989, 2021-02-18:
this is what i get

https://i.imgur.com/G0X3FTf.png
on pressing load more I get this https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Errors/Not_a_function
i have tried in edge also

FormerLurker, 2021-02-18:
That's a strange error considering I don't use that function.  Can you try disabling other plugins you installed, reboot, and see if you still have the issue? Also, did you try clicking the 'customize profile' box in your printer profile?  Look down at the snapshot command box and see what is in there.

numpty9989, 2021-02-18:
No plugins dnt use them xD, and I load profile press ok to load it and then try the save button no customization or anything

FormerLurker, 2021-02-18:
> and then try the save button no customization or anything

Yeah, I saw your video.  Please look at the printer profile and click on the 'customize profile' check box, then scroll down and look at the 'Snapshot Command'.  Click save and let me know what you see then please.

numpty9989, 2021-02-18:
G4 P1

FormerLurker, 2021-02-18:
Hmm..  I did a code search and the only place i find the code that's throwing the error is in a minified javascript library.  Did you happen to enable minification in config.yaml or something?

Sorry, this is a tough one, because tons of people use that profile, and this is the first time I've heard about this.  Something unusual must be going on.  I can't replicate the issue on my end, and I'm using the same OS/Browser you have.  The fact that you have no other plugins other than Octolapse installed makes this extremely mysterious.

numpty9989, 2021-02-18:
dont think so .. where is config.yaml don't think I've touched it

FormerLurker, 2021-02-18:
Ok, can you download octoprint.log (OctoPrint settings->Logs) and post to gist.github.com, then drop a link in here?

numpty9989, 2021-02-18:
[octoprint.log](https://github.com/FormerLurker/Octolapse/files/6005627/octoprint.log)

numpty9989, 2021-02-18:
[octoprint.log](https://github.com/FormerLurker/Octolapse/files/6005628/octoprint.log)

FormerLurker, 2021-02-18:
> No plugins dnt use them xD,

According to your octoprint.log file, you have a ton of plugins installed.  Try disabling 'PrettyGcode', reboot, and see if it works.  I need to figure out what's going on there, but I've seen this issue before.

numpty9989, 2021-02-18:
wen u sed plugins i thought u meant browser lol my bad :)

https://i.imgur.com/YgNRQqT.png

i was pretty gcode it was kind of buggy but i liked to wathch the thing printing lol

FormerLurker, 2021-02-18:
> wen u sed plugins i thought u meant browser lol my bad :)

Not a problem.

I will see what I can do to fix PrettyGcode, as I like using it too.  Some of my code (allow G2/G3 commands to be displayed) is actually in that plugin, so I'm keen to get it working.  It's probably a javascript conflict.  

Can I mark this closed?

numpty9989, 2021-02-18:
yeah, thanks good luck fixing it :) id love to write my own plugin but dunno where to start LOL

FormerLurker, 2021-02-18:
Cool, closed!

FYI, if you want to write a plugin, just follow [this guide](https://docs.octoprint.org/en/master/plugins/gettingstarted.html).  Don't expect things to go quickly right away, and be sure to read through everything first (you will follow windows instructions).  Once you figure out what's going on, it's just a matter of learning python, Javascript, and Knockout, though you don't really need to know a whole lot about any of those to get something working.  Good luck!

closed 2021-02-18 by FormerLurker