FormerLurker/Octolapse #691: Only 1st photo is running... :/   error for next photos

nano2sillery, 2021-02-17:
Hello Octolapsers 🙂
I 'like Octolapse ...i use it with my webcam ! All is ok !
Now i I wish to produce better quality timelapses and use my DSLR CANON 1300D... I found and apply this tutorial : https://github.com/.../Oct.../wiki/V0.4---Configuring-a-DSLR

All is ok when i test from the ssh terminal :
pi@octopi:~ $ gphoto2 --auto-detect --capture-image-and-download

Model                          Port                                            
----------------------------------------------------------
Canon EOS 1300D                usb:001,004     
New file is in location /store_00020001/DCIM/105ND800/IMG_5110.JPG on the camera
Saving file as IMG_5110.JPG
Deleting file /store_00020001/DCIM/105ND800/IMG_5110.JPG on the camera
....
but when i test the script in the settings of octolapse, an error message is displayed. 
Here after the log file :

2021-02-17 11:06:36,928 - octolapse.script - DEBUG - Executing CANON 1300D - Snapshot Camera Script: "/home/pi/scripts/take-snapshot.sh" "0" "125" "/tmp/tmpe83_hu3p" "/tmp/tmpe83_hu3p/bb10a44c-084a-463c-8f7a-a3f7d85aad79/78c55645-6873-4b8c-8548-6094c162f644" "test_snapshot000000.jpg" "/tmp/tmpe83_hu3p/bb10a44c-084a-463c-8f7a-a3f7d85aad79/78c55645-6873-4b8c-8548-6094c162f644/test_snapshot000000.jpg" "CANON 1300D"
2021-02-17 11:06:38,675 - octolapse.script - ERROR - An error occurred while executing 'CANON 1300D - Snapshot Camera Script'
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/script.py", line 248, in execute_process
    self.proc = subprocess.Popen(args, stdout=subprocess.PIPE, stderr=subprocess.PIPE)
  File "/usr/lib/python3.7/subprocess.py", line 775, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.7/subprocess.py", line 1522, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
OSError: [Errno 8] Exec format error: '/home/pi/scripts/take-snapshot.sh'

after some hours of test, i see that only the first photo taken is running good.... at the second photo = ERROR... if  i shutdown and restart the DSLR it is ok for the 1st photo and 2nd is ERROR again 😕  Like the DSLR stay BUSY after 1st Photo taken.

COULD SOMEONE HELP ME PLEASE ? Thanks

FormerLurker, 2021-02-21:
Hey there @nano2sillery, sorry I didn't respond sooner.  If you closed this issue because you solved it, excellent!  I just wanted you to know i wasn't ignoring you, I've just been pretty behind on issues lately due to an influx from another plugin.  Anyway, if you need help still, let me know.

nano2sillery, 2021-02-21:
Thanks... i don't solve it. 😕

Do you know why my LAMETRIC display NOT FOUND and only sometime display %
and Temperature ?

I work with the latest versiin of octoprint and latest version of your
plugin. My printer is Creality CR6SE.

Thanks

Cyrille LEFRANC - studiopixl.fr

FormerLurker, 2021-02-21:
Not sure what is up with the Lametric display.

Regarding your camera, can you take a screenshot of your visudo setup, and your take-snapshot.sh file?  I know your camera can work, because I've had other users saying it worked for them.  You may want to check your gphoto2 version as well, and let me know how you installed it.

nano2sillery, 2021-02-22:
HI,

Thanks for your Help

DSLR is detected...but only ONE PHOTO could be taken... Next photo is in
error.

I have use this tutorial :
https://github.com/FormerLurker/Octolapse/wiki/V0.4---Configuring-a-DSLR

I have install GPHOTO with this tutorial :
https://github.com/FormerLurker/Octolapse/wiki/V0.4---GPhoto2-Installation
with this command :

cd /home/pi/scripts

nano install-gphoto2.sh

Please find in attached files screenshots asked.

Thanks

*Cyrille LEFRANC*
*-----------------------------------------*
PHOTOGRAPHIE, VIDEO & GRAPHISME.

06.03.92.04.27
*www.studiopixl.fr <http://www.studiopixl.fr/>*