FormerLurker/Octolapse #692: [Request] Option to use ffmpeg video filters

gos-fr, 2021-02-17:
#### If this is a feature request describe it here

I would like to be able to use ffmpeg video filter during rendiring of the timelapse. Most importantly the deflicker filter against changing light conditioin during the print.
Would it be possible to add a text field where one could enter the video filter string to be passed to the ffmpeg -vf option ?

#### Version of Octolapse
 
Octolapse Version: 0.4.1

#### Version of OctoPrint

OctoPrint Version: 1.5.3

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: ___REPLACE_THIS__YES_OR_NO

#### What were you doing when the problem occurred

1. ___REPLACE_THIS__STEP_ONE_GOES_HERE
2. ___REPLACE_THIS__STEP_TWO_GOES_HERE
3. ___REPLACE_THIS__STEP_...

#### What should have happened?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: ___REPLACE_THIS__OS_NAME_GOES_HERE
Os Version: ___REPLACE_THIS__OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2021-02-18:
I have gotten this request before, and it is a good idea.  However, it will be very difficult to validate the parameters due to the way they must be structured, making it easy to break the rendering process.  That doesn't mean it isn't possible, just that i need to find some reasonable way to test the parameters before they are used.  I was thinking maybe I could include 3 test photos and render them into a simple movie, and throw a validation message if there are any rendering errors.  Will think about it.  In the meanwhile, you can ask octolapse to save your images, download them, and render them any way you like.  That is obviously just a workaround.  See the archive option in the rendering profile.

FormerLurker, 2021-02-18:
Actually, it would be easier to add checkboxes for various filters.  There are always multiple ways to solve a problem.  Let me know what you think.

gos-fr, 2021-02-21:
Both options are good.
I personally like the first one more for being more flexible, and allowing the use of every ffmpeg filter, and more so allowing to build complete filter graphs. It is however more targeted to users with good ffmpeg knowledge, like myself. The idea of a small render to test the filter is great.
The second option with check boxes and some fields for each filter's parameters is a good option for users with no ffmpeg knowledge. But that will mean limited filter availability and a lot of work for you building a UI for the filter and their parameters, and no filter graph.
I guess the decision lies in what kind of user you are targeting.

Of course one can always postprocess the video, or render the capture image sequence, but the ease of automatic processing is too attractive :)

FormerLurker, 2021-02-22:
Agreed, on all points.  I need to do some research on the available filters and parameters.  I'll let you  know how it goes.