FormerLurker/Octolapse #693: Combine @OCTOLAPSE commands with Smart - Snap To Print triggers

gritstub, 2021-02-19:
Feature request to combine @OCTOLAPSE commands to manually take a screen shot while using Smart - Snap To Print triggers. 

The use case is to use the Smart - Snap To Print trigger, and then add End Gcode to my file to move the nozzle up and away, and take photos.

The reason is that I have Stabilization Disabled, but want to take a shot of the completed print with the nozzle out of the way.

eg:

```
;END GCODE
G91 ;Relative positioning
G1 Z10 ;Raise Z more
M400 ;Wait
@OCTOLAPSE TAKE-SNAPSHOT
G1 Z10 ;Raise Z more
M400 ;Wait
@OCTOLAPSE TAKE-SNAPSHOT
```

#### Version of Octolapse
Octolapse Version: 0.4.1

#### Version of OctoPrint
OctoPrint Version: 1.5.3

gritstub, 2021-02-19:
Is there an alternate way to do this by tricking "Smart - Snap To Print" to calculate that I have some extra layers at the end and should take a snapshot?

gritstub, 2021-02-24:
@FormerLurker : Hi, I just wanted to make sure you hadn't missed this, since you're very active with the issue tracker! If you think it's not a worthwhile feature request, that is acceptable as well.

FormerLurker, 2021-02-24:
Sorry, i had a comment that i failed to submit.  Thanks for reaching out!

This is on the agenda.  I plan to allow gcode triggering as an option, and am adding some other useful commands.

gritstub, 2021-02-24:
No need to apologize, I'm just glad to hear it's on your radar! I'm looking forward to this, and the other commands you're adding. Thanks for being so involved with your project and users.

AubreyKilian, 2021-06-20:
I attempted this behaviour by adding the end gcode into Cura's "End gcode" section, but that didn't work.
Having some kind of "final list of actions before starting rendering" such as gcode would be extremely useful.