FormerLurker/Octolapse #695: Position offset after snapshot

0xBeFa, 2021-02-23:
Hi, I have a problem with Octolapse. If it is enabled my printer returns to the wrong position after taking a snapshot which is a very dangerous situation and could damage the printer. I use Cura for slicing and have added following G-code after each layer change:

`;CustomTimelapse Begin
; STEP 1 : retraction
G1 F1800 E-2;Retraction
; STEP 2 : Move the head up a bit
G0 Z2.5;Move Z axis up a bit
; STEP 3 : Move the head to "display" position and wait
G0 Y70;GCODE for the first position(display position)
G0 X110;GCODE for the second position(trigger position)
M400;Wait for moves to finish
G4 P700;Wait for camera
; STEP 4 : send photo trigger command if set
@OCTOLAPSE TAKE-SNAPSHOT ;Snap Photo ;Snap Photo
; STEP 5 : Move the head back in its original place
G0 X-3.829 Y26.447
; STEP 6 : Move the head height back down
G0 Z1.5;Move Z axis back down
; STEP 7 : undo retract
G1 F1800 E2;Retraction
;CustomTimelapse End`

If I disable Octolapse this G-code prints without problems, the same is true if I print from SD-Card. Since the movement of the head is already included in the G-code I expect Octolapse to not send any command to the printer, it just needs to listen to the @OCTOLAPSE TAKE-SNAPSHOT command and take the picture.

For that purpose I have set the trigger to Classic-Gcode and stabilization to disabled. I tried to fill in all the printer and slicer settings as good as possible but I don't really understand why Octolapse needs this settings anyway. I have disabled layer change detection and edited a lot of options because I was getting many errors at the start of a print job which resulted in the printer not doing anything. I don't want to repeat myself but from my point of view all these printer settings are not necessary because I tell Octolapse exactly when to take a snapshot in the G-code and expect it to do nothing else that taking the snapshot and combine it to a video.

#### Version of Octolapse
Octolapse Version: 0.4.1

#### Version of OctoPrint
OctoPrint Version: 1.5.3

#### When you ran into the problem, did you have diagnostic logging enabled?
Diagnostic Logging was Enabled: yes

#### What were you doing when the problem occurred
The print job was started. Startup code and first layer was ok. After the snapshot the 2nd layer is printed at a wrong position.

#### What happened instead?
Position was wrong, offset of the 2nd layer to the first layer is about X:-90mm and Y:-70mm.

#### Operating System running OctoPrint and Octolapse
OS Name: OctoPi
Os Version: 0.18.0

#### Printer model & used firmware incl. version
Printer Model: Qidi Tech 1
Printer Firmware Version: Sailfish 7.8

#### Browser and version of browser, operating system running browser
Browser: Google Chrome
Browser OS: Windows 10

#### Link to the gcode file you were printing when the problem occurred
Link to Gcode File: https://gist.github.com/0xBeFa/eee6972d9a528b8a0b4d695630cd1047

#### Link to settings.json
https://gist.github.com/0xBeFa/f16f1c8688121c3fe7fd2cd7e2324705

#### Link to plugin_octolapse.log
Link to plugin_octolapse.log:  https://gist.github.com/0xBeFa/6e38263eb4034d9fc5b2fb355ba4ac81

#### Link to octoprint.log
Link to octoprint.log:  https://gist.github.com/0xBeFa/5a5b707fc3aab7ff8ec816d6de5e6d48

#### Screenshots and/or videos of the problem:
Screenshot Links: https://drive.google.com/file/d/1oKNu2a_m5WDKHZXGqruNyqj1H1kCJIYq/view?usp=sharing

FormerLurker, 2021-02-23:
I'm guessing this is a makerbot clone?  I think there are a few culprits here, the biggest being the 

```M135 T1 ; left or right tool```

command.  My guess is your print is off by the tool separation, right?  is the Z-Offset being handled correctly, or is the Z height off too after layer 1?  I'm not sure how to interpret this code:

```
G161 Z F1200 ; home Z minimum
G92 Z-5 ; set Z to -5
G1 Z0 ; move Z to 0
G161 Z F100 ; home Z slowly
```

It appears to set an offset after homing (height 0 is now considered -5), then it moves up 5 mm (G1 Z0), then it re-homes?  I'm not sure if that clears the Z offset or not.  I would expect it should, but if so, why set it in the first place?  If you can explain what this is supposed to do, and why, that may be helpful.

I also don't understand this gcode:

```
M132 X Y Z A B
```

I'm guessing that configures tool offsets as are stored in the firmware.  If so, what is A and B?  Are those extruders for the left and right extruder?  If so, that's a bit unusual too.

Also, here:

```
G92 A0 B0 ; zero extruders
```
I've never seen code like this for handling extruder offsets.  Typically one would see:

```
T0 ; toolchange
G92 E0; Reset extruder for tool 0
```

 Can you tell me if T0 = tool1 or if T1 = tool1?  I'm guessing tool 1 is on the left too.  If you have the extruder offsets, that would be useful too.

Fyi, I'm very glad to finally receive a multi-tool support issue, since it's been available and in beta for a LONG time:

[image]

I've tested shared extruder support extensively, but have never owned (or even seen) a dual extruder machine.  I've run some theoretical tests, but they rely on having all the appropriate gcodes implemented, having done all the offset math correctly, and me understanding exactly how it all works.  Odds are I missed a few of those items.  Specifically I see several gcodes in your file that are unknown to me: 

* M135 - which is PID sample interval in reprap firmware and change tool in makerbot (why not use T0/T1?).  In your start gcode you have conflicting comments indicating what the purpose of this gcode is:

```
M135 T0 ; load right extruder offsets
M135 T1 ; left or right tool
```

* G162 and G161 - I MAY have these gcodes implemented (need to double check)
```
G162 X Y F3000 ; home XY maximum
G161 Z F1200 ; home Z minimum
```

But I recall an open issue about multiple behaviors for this command.  Will G28 not work?  Not to say I shouldn't get that gcode correct (I will try), but since I see it so rarely I'm thinking a temporary workaround would be just to G28 XY, then G28 Z (or just plain old G28).  If there is a reason this code is used, I'd love to know so I am better equipped to handle this code.

* G92 with A and B parameters (what about C, D, E, F?? How do I know which extruder is associated with A and B?)

Anyway, this could take extensive modifications to handle all of these differences.  I will take a look and see if I can find other examples of multi-extruder code in different flavors.

I'll look at your settings as soon as I have some time to see how you've configured your printer.  I have to warn you that I've been really busy the last couple months with work and an influx of issues (new plugin + ppl getting printers for christmas), but I promise I'll take a closer look ASAP.

FormerLurker, 2021-02-23:
I did take a quick peak at your printer profile settings, and I think there are some issues:

[image]

Is your first extruder really T0 (see the question mark)?  Also, it is unlikely that both your extruders have 0mm offsets.

Since I didn't see any G10 or M218 gcodes, I don't see how this could work.  It's still 100% possible that the offsets won't work correctly anyway, since this is the first feedback I've ever gotten on this feature to my knowledge.

FormerLurker, 2021-02-23:
One more thing:  I think your gcode hack is brilliant :)  I never would have thought to do that, so Kudos for working with what you have in extremely creative ways!

FormerLurker, 2021-02-23:
Simplify3D has trouble visualizing your gcode:

[image]

[image]

Do you have a photo or stl of what this piece should actually look like?

0xBeFa, 2021-02-23:
I don't really understand all of the start G-code. It was part of a Cura profile I found online. The Z axis is homed twice. The first time fast the 2nd time slow. I guess it has something to do with precision.

T0 selects the first (right) extruder and T1 the second (left) extruder. The offset between the extruders is handled with the tool change command, in Cura I have set it also to X:0.0 and Y:0.0.

I will have a look at your other G-code questions tomorrow.

As you already noticed there is some other problem with my G-code file. It is an Z offset on some layers. Immediately after creating my initial post I started printing a different file and saw the same Z offset at layer 2. I will fix this first and post some updated G-Code and Log file. Cure doesn't show this Z layer shift in the preview so I have to search the problem in the G-Code.
Probably it has something to do with my search for the absolute z position of the next layer because I wanted to avoid the switching to relative movement for lifting the nozzle.

0xBeFa, 2021-02-23:
The G-code file I posted is only a singe extrusion file. I will test dual extrusion after single is working.

FormerLurker, 2021-02-23:
> T0 selects the first (right) extruder and T1 the second (left) extruder. The offset between the extruders is handled with the tool change command, in Cura I have set it also to X:0.0 and Y:0.0.

So, cura won't need to know where the nozzle is after a toolchange, and can basically just issue a G0 and move to whatever position.  Octolapse potentially DOES need to know the offset, because a 'Take Snapshot' command could appear right after a toolchange, in which case Octolapse would return to the wrong position.  Also, to stabilize a dual extruder print, the offset has to be known to correctly position the hotend.  Without the offsets, even if everything else worked perfectly, the hotend would appear to jump around by the offset.

> T0 selects the first (right) extruder and T1 the second (left) extruder.

Ok, that makes sense.  

> I don't really understand all of the start G-code. It was part of a Cura profile I found online. The Z axis is homed twice. The first time fast the 2nd time slow. I guess it has something to do with precision.

I'm wondering what's going on with the Z offset after the home is complete.  Is it still active?  Maybe send this sequence (with a clear buildplate!) and let me know what the M114 response is:

```
G161 Z F1200 ; home Z minimum
G92 Z-5 ; set Z to -5
G1 Z0 ; move Z to 0
G161 Z F100 ; home Z slowly
```

Understanding that G92 is pretty important, though.  I'll start looking into what I can soon.

0xBeFa, 2021-02-26:
I fixed the z offset from my gcode and run another test. Here are the new log and gcode files:
gcode: https://gist.github.com/0xBeFa/a50ea65d225257a1b07305a91e54833c
octolapse log: https://gist.github.com/0xBeFa/9c381deb464bc08fcbe067718cfbe5a7
octoprint log: https://gist.github.com/0xBeFa/83c0506a02b7b73a2791eccd6d73a0fd

I noticed that after the first layer the print head is moving to the snapshot position in the top left corner. Then it is crashing against the X axis homing switch and then moving to the wrong position.

I compared the    octolapse.timelapse - DEBUG - Sent:    commands in the log file to the original gcode file and it is matching exactly. However I found also this entries in the log file:
`2021-02-26 19:35:21,425 - octolapse.stabilization_gcode - INFO - Triggered at line: 578 by gcode: @OCTOLAPSE TAKE-SNAPSHOT
2021-02-26 19:35:21,429 - octolapse.stabilization_gcode - INFO - Snapshot Gcode (Tool: 2):
	Snapshot       - G0 X210.000 Y140.000 F4200.000
	Take Snapshot  - @OCTOLAPSE TAKE-SNAPSHOT
	Return         - G0 X110.000 Y70.000
	End            - G1 F1800.000`

I strongly suspect that this G-Code - if it is send to the printer -  is responsible for the offset. (X210,Y140) is way out of my build volume. My build plate is approximately X -110 to +110 and Y -70 to +70. If the head is moving to X210 but is physically limited at 120 it will have an additional travel back distance of 90mm which matches the observed X offset. For the Y axis it is the same.

I am using the "disabled" profile for stabilisation but there is stabilization gcode added nevertheless. Could it be that my dual extruder printer needs a different disabled profile?

FormerLurker, 2021-02-28:
This is pretty complex, but I think I have an idea what's going on.  It may be a while before I can get to this, unfortunately, as I'm in the middle of a custom printer build, and it's sapping a lot of time.  However, when it's finished, I should be able to test with various boards and firmware, so hopefully it will speed up future issues.  As soon as it's finished I'm going to start tackling some of these more difficult issues.