FormerLurker/Octolapse #697: delay after snapshot with gphoto2 and sony alpha6300

yvesgarnier, 2021-02-25:
Octolapse Version: 0.4.1

OctoPrint Version: 1.5.3

Diagnostic Logging was Enabled: YES

**What i did:**
I installed Octolapse and set it up with gphoto2 for external Camera use. 

**What should have happened?**
It should take a picture before moving to the next layer and then move on without a 5s break. Snapshot is working and transfering on the raspberry pi.

**What happened instead?**
It takes a long break after the snapshot and then moves on. Nothing really when i changed the delay parameters in the camera settings.

**Operating System running OctoPrint and Octolapse**
OS Name: OctoPi
Os Version: 0.18.0

**Printer model & used firmware incl. version**
Printer Model: Original Prusa i3 MK3s+
Printer Firmware Version: 3.9.3-3556

Browser: Safari
Browser OS: MacOS

FormerLurker, 2021-02-25:
What is your snapshot delay:

[image]

Also, please include the requested files, in particular settings.json.  Otherwise I have to go back and forth to get info that is contained there, so it just speeds things along.

yvesgarnier, 2021-02-25:
hi

i tried different setting here but the delays did not really change. Currently:
Snapshot Delay: 0 ms
Snapshot Timeout: 6000 ms

when i use less than 6000 ms and test the script in the “edit camera” settings in octolapse it returns:

 Errors were detected - The 'DSLR - Snapshot Camera Script' timed out in 1.0 seconds
 
here settings.json as txt:

[settings-json.txt](https://github.com/FormerLurker/Octolapse/files/6046222/settings-json.txt)

thanks for your help, yves

FormerLurker, 2021-02-25:
Ok, that looks right.  Can you time the gphoto2 call in the terminal and see how long it takes there?

yvesgarnier, 2021-02-25:
i entered: `/home/pi/scripts/take-snapshot.sh 1 1 "" "/home/pi/scripts" "" "/home/pi/scripts/test.jpg"` via ssh, after 2 seconds the file was created. The camera snaps instantly after i hit enter.

FormerLurker, 2021-02-25:
Does every layer have this extra 4 second delay, including the first layer?  I'm asking because Octolapse just waits for the script to return, and then moves on, so I'm having a hard time figuring out what's going on.  We may need to create some logs, but maybe you could post your take-snapshot.sh script?

yvesgarnier, 2021-02-25:
my take-snapshot.sh is an exact copy of yours ;-) from: https://github.com/FormerLurker/Octolapse/wiki/Configuring-an-External-Camera

the log from yesterday when i tried several times. dont know if it helps.
[plugin_octolapse-2.log](https://github.com/FormerLurker/Octolapse/files/6046319/plugin_octolapse-2.log)

yvesgarnier, 2021-02-25:
i first thought that its because of the delay from creating/copying the image from the camera. But then i tried manually in the terminal and there was less delay. i also changed the file size/resolution on the camera, but nothing changed.

FormerLurker, 2021-02-25:
That was helpful, but it is an exception only log.  This indicates some permission error:
```
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/script.py", line 103, in kill
    proc.kill()
  File "/home/pi/oprint/lib/python3.7/site-packages/psutil/__init__.py", line 272, in wrapper
    return fun(self, *args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/psutil/__init__.py", line 1239, in kill
    self._send_signal(signal.SIGKILL)
  File "/home/pi/oprint/lib/python3.7/site-packages/psutil/__init__.py", line 1186, in _send_signal
    raise AccessDenied(self.pid, self._name)
psutil.AccessDenied: psutil.AccessDenied (pid=29420)
2021-02-25 03:45:00,277 - octolapse.__init__ - ERROR - An unexpected error occurred while executing the snapshot script.
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/psutil/__init__.py", line 1176, in _send_signal
    os.kill(self.pid, sig)
PermissionError: [Errno 1] Operation not permitted
```

It does look like I have some kind of error in the 'kill' script, so I'll look into that, but I think this can be remedied.

Did you configure sudoers correctly?  Double check [this](https://github.com/FormerLurker/Octolapse/wiki/Configuring-an-External-Camera#step-3---configure-etcsudoers-file) step, and be sure to use the ```whereis gphoto2``` command to determine the location of gphoto2.  It's possible you have multiple installs too.

yvesgarnier, 2021-02-26:
Here whats in my visudo:

> `
#
# This file MUST be edited with the 'visudo' command as root.
#
# Please consider adding local content in /etc/sudoers.d/ instead of
# directly modifying this file.
#
# See the man page for details on how to write a sudoers file.
#
Defaults        env_reset
Defaults        mail_badpass
Defaults        secure_path="/root/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin"

# Host alias specification

# User alias specification

# Cmnd alias specification

# User privilege specification
root    ALL=(ALL:ALL) ALL

# Allow members of group sudo to execute any command
%sudo   ALL=(ALL:ALL) ALL

# See sudoers(5) for more information on "#include" directives:

#includedir /etc/sudoers.d

# allow 'sudo gphoto2' to run without supplying a password
pi ALL = (root) NOPASSWD: /usr/bin/gphoto2

`

whereis gphoto2 return:

> `gphoto2: /usr/bin/gphoto2  /usr/share/man/man1/gphoto2.1.gz`

FormerLurker, 2021-02-26:
Maybe those log entries were old?  Can you change the logging profile to debug, edit the profile and delete all logs, then switch on test mode and print a few layer, cancel the print, turn off test mode, then download and post the log?

yvesgarnier, 2021-02-26:
Snapshot before printing plus one layer was printed (icluding the beginning of the second layer).

[plugin_octolapse-3.log](https://github.com/FormerLurker/Octolapse/files/6046694/plugin_octolapse-3.log)

FormerLurker, 2021-02-26:
Awesome.  Will take a look in yge AM, thanks!

FormerLurker, 2021-02-26:
So, here are the relevant log entries:

```
2021-02-26 00:41:56,409 - octolapse.snapshot - INFO - Starting snapshot acquisition
2021-02-26 00:41:56,410 - octolapse.snapshot - INFO - Starting 1 snapshot threads.
2021-02-26 00:41:56,411 - octolapse.snapshot - INFO - Snapshot - running snapshot script for the DSLR camera.
2021-02-26 00:41:56,412 - octolapse.snapshot - DEBUG - Snapshot Delay - No pre snapshot delay configured for the DSLR camera.
2021-02-26 00:41:56,413 - octolapse.script - DEBUG - Executing DSLR - Snapshot Camera Script with no timeout: "/home/pi/scripts/take-snapshot.sh" "1" "0.0" "/home/pi/.octoprint/data/octolapse/tmp" "/home/pi/.octoprint/data/octolapse/tmp/octolapse_snapshots_tmp/595f1f67-ec27-45d8-9bda-6f4c264cdf27/6c0ae244-5ead-4284-90d3-5b3760c33fcc" "camera_rear_0.2mm_PLA_MK3S_18m000001.jpg" "/home/pi/.octoprint/data/octolapse/tmp/octolapse_snapshots_tmp/595f1f67-ec27-45d8-9bda-6f4c264cdf27/6c0ae244-5ead-4284-90d3-5b3760c33fcc/camera_rear_0.2mm_PLA_MK3S_18m000001.jpg" "DSLR"
2021-02-26 00:42:01,550 - octolapse.script - DEBUG - Console output (stdout) for 'DSLR - Snapshot Camera Script':New file is in location /capt0000.jpg on the camera
	Saving file as /home/pi/.octoprint/data/octolapse/tmp/octolapse_snapshots_tmp/595f1f67-ec27-45d8-9bda-6f4c264cdf27/6c0ae244-5ead-4284-90d3-5b3760c33fcc/camera_rear_0.2mm_PLA_MK3S_18m000001.jpg
	Deleting file /capt0000.jpg on the camera
2021-02-26 00:42:01,553 - octolapse.snapshot - DEBUG - Post-processing snapshot for the DSLR camera.
2021-02-26 00:42:01,553 - octolapse.snapshot - INFO - The snapshot script job completed, signaling task queue.
2021-02-26 00:42:01,557 - octolapse.snapshot - INFO - Snapshot threads complete, but may be post-processing.
2021-02-26 00:42:01,558 - octolapse.snapshot - INFO - Snapshot acquisition completed in 5.148 seconds.
```

As you can see, the delay occurs almost entirely within the gphoto2 script call.  Something is taking longer than expected there for whatever reason.  Try altering your gphoto2 call to this:

```
sudo gphoto2 --auto-detect --capture-image-and-download --filename "${SNAPSHOT_FULL_PATH}" --debug
```

That should produce detailed output.  Test this on the command line first and make sure you get a lot more info, then repeat the steps for creating a debug log that I posted before (make sure you clear the previous log first please), and we should be able to tell what is taking so long.

Thanks!

yvesgarnier, 2021-02-26:
Hey another day another try. Thanks for staying with me and helping me out!

so i did what you said and here the log file:
[plugin_octolapse-4.log](https://github.com/FormerLurker/Octolapse/files/6052355/plugin_octolapse-4.log)

also note that now after every shot taken this notification apears:

[image]

here is what i did to the take-snapshot.sh file:

```
#!/bin/sh
# Put the arguments sent by Octolapse into variables for easy use
SNAPSHOT_NUMBER=$1
DELAY_SECONDS=$2
DATA_DIRECTORY=$3
SNAPSHOT_DIRECTORY=$4
SNAPSHOT_FILENAME=$5
SNAPSHOT_FULL_PATH=$6

# Check to see if the snapshot directory exists
if [ ! -d "${SNAPSHOT_DIRECTORY}" ];
then
  echo "Creating directory: ${SNAPSHOT_DIRECTORY}"
  mkdir -p "${SNAPSHOT_DIRECTORY}"
fi

# IMPORTANT - You must add gphoto2 to your /etc/sudoers file in order to execute gphoto2 without sudo
# otherwise the following line will fail.
sudo gphoto2 --auto-detect --capture-image-and-download --filename "${SNAPSHOT_FULL_PATH}" --debug

if [ ! -f "${SNAPSHOT_FULL_PATH}" ];
then
  echo "The snapshot was not found in the expected directory: '${SNAPSHOT_FULL_PATH}'." >&2
  exit 1
fi
```

FormerLurker, 2021-02-28:
Holy cow that log is insane.  It looks to be doing a lot, and I can see the timestamps slowly increment all the way to the total time that Octolapse reports.  The only thing I can think of is that when running from the command line the script exists to the terminal before the process is actually completed, but octolapse waits until it is finished.  The 'Trigger Capture' option is supposed to circumvent this, but I'm not sure if the Sony supports that option.  Maybe you could try that from the command line?

yvesgarnier, 2021-03-08:
hey, sorry was busy that days. What do you mean with the "Trigger Capture" option?

FormerLurker, 2021-03-10:
[Here](http://gphoto.org/doc/manual/ref-gphoto2-cli.html) is the definitive manual that explains these options.  Keep in mind you will need to test the commands, as not all cameras support all commands.

kot0005, 2021-03-12:
I am having similar issues on my A6000 sony camera, These cameras are a Pain.. 

Sometimes test command ### gphoto2 --auto-detect --capture-image works and sometimes The camera takes photos but i get error could not capture image
Error could not Capture

I also have the same red box error when trying to test the script in octolapse.

[octoprint (4).log](https://github.com/FormerLurker/Octolapse/files/6130967/octoprint.4.log)

kot0005, 2021-03-12:
> Octolapse Version: 0.4.1
> 
> OctoPrint Version: 1.5.3
> 
> Diagnostic Logging was Enabled: YES
> 
> **What i did:**
> I installed Octolapse and set it up with gphoto2 for external Camera use.
> 
> **What should have happened?**
> It should take a picture before moving to the next layer and then move on without a 5s break. Snapshot is working and transfering on the raspberry pi.
> 
> **What happened instead?**
> It takes a long break after the snapshot and then moves on. Nothing really happens when i change the delay parameters in the camera settings.
> 
> **Operating System running OctoPrint and Octolapse**
> OS Name: OctoPi
> Os Version: 0.18.0
> 
> **Printer model & used firmware incl. version**
> Printer Model: Original Prusa i3 MK3s+
> Printer Firmware Version: 3.9.3-3556
> 
> Browser: Safari
> Browser OS: MacOS
> 
> **Settings.json (as txt)**
> [settings-json.txt](https://github.com/FormerLurker/Octolapse/files/6046225/settings-json.txt)

How is your A6300 connected to the Pi ? I am using PC Remote but not sure if its the right one, I am also using manual mode on the wheel and USB LUN setting is set to multi. ( i dunno what the USB LUN setting does)

FormerLurker, 2021-03-13:
@kot0005, thanks for chiming in!

I wanted to mention again that these issues are not really Octolapse problems, but rather the firmware and/or gphoto2.  I would strongly recommend doing some google-fu regarding the specific camera model and gphoto2.  It's quite possible that there are workarounds or known limitations.  Things are changing with gphoto2 all the time, so it's possible some of them have been addressed.  I do provide a guide for using an advanced installation script to install the latest (including development versions) of gphoto2 in the 'Configuring a DSLR guide'.  If any fixes have been added that are not available via apt-get (updated much less frequently), you can install them that way.

FormerLurker, 2021-03-13:
Oh, and you may want to check for firmware updates for your camera too.  Be sure to read the change log so you know what bugs have been fixed or features that have been added.  It's just possible one of them may address these issues.  Not a high chance, but might be worth a look.

kot0005, 2021-03-14:
ok so update, I reset my camera and using Superior auto, USB  (PC remote) and USB LUN set to Multi , Turned on Airplane mode and octolapse is working, i guess some of the manual settings i had was breaking something.

FormerLurker, 2021-03-15:
@kot0005, great!!  DSLR timelapses are a pain, but they do look great :)