FormerLurker/Octolapse #699: [Request] Print Without Octolapse button on startup error dialog

lukemaurer, 2021-03-07:
#### If this is a feature request describe it here

Sometimes Octolapse reports an error on startup and I'd like to print anyway, just with Octolapse temporarily disabled. Having to dismiss the dialog, disable Octolapse, and click Print again is inconvenient, and I'm always afraid I'll forget to enable it again afterward.

#### Version of Octolapse
 
Octolapse Version: v0.4.1

#### Version of OctoPrint

OctoPrint Version: 1.5.3

FormerLurker, 2021-03-10:
Not a bad idea, I'll look into it.  Marking this as an enhancement.

garyt72, 2021-03-17:
I would like to add to this request - not sure how similar my idea is, but could you add a "Skip this print" option to the Snapshot Plan review screen?  I understand at that point the gcode has already been generated/modified but would it be possible to revert to the unmodified gcode?

I like leaving Octolapse enabled, but would like to skip the occasional item for one reason or another.

FormerLurker, 2021-03-17:
> I like leaving Octolapse enabled, but would like to skip the occasional item for one reason or another.

This is a good idea too.  I will think about it.  Turns out it's actually not so easy to do this in every case, but I think something can be worked out with enough thought.

>  I understand at that point the gcode has already been generated/modified but would it be possible to revert to the unmodified gcode?

The gcode is not modified.  Octolapse just builds a lookup of when snapshots should take place by line number, as well as the steps necessary to perform the snapshots, so that won't be a problem.  The new gcode is injected in real time when it is needed, so cancelling would just prevent that.

AnduriI, 2021-03-23:
is it possible to add some gcode to disable octolapse for that specific print? I would like to deactivate it for 'boring' prints that I print multiple times.

FormerLurker, 2021-03-23:
There currently isn't any gcode for this, but you can click the 'disable' button on the Octolapse tab.

AnduriI, 2021-03-24:
I know that I can manually disable octolapse for a running print or completely (but I often forget to reactivate later). But I have a few prints that I do occasonally and that a not worth filming (so tags, temptowers for new filament,...). Maybe we could add a specific line in a gcode file that makes octolapse ignore this print but be ready for the next one...

AnduriI, 2021-03-26:
well I found another post by you talking about ```@Octolapse stop-snapshots``` - wouldn't that work? Simply adding this to the start gcode of the 'boring' files...

FormerLurker, 2021-03-28:
Well, it WOULD, but Octolapse will complain if there aren't enough snapshots to create a timelapse (at least two).

traviselam, 2022-12-28:
I would like to throw in a +1 for this feature. I would love this feature to skip my usual calibration prints. A calibration cube and temperature towers for new filaments aren't really that interesting to see a timelapse for haha

AdmiralbadBeef, 2023-12-04:
throwing in my +1 for this.

there are occasions that you dont want a timelapse of a print - disabling octolapse is currently the only real option, this is painful since you inevitably forget before you start and have to cancel and restart OR you forget to re-enable and dont get a timelapse of something you do want.....

Summing up:
   * ability to "skip this print" during snapshot plan
   * adding specific gcode (comment or other) to skip specific prints
   * skip and print anyway on octolapse startup error