FormerLurker/Octolapse #700: No Print Features Detected even if Verbose is enabled in Superslicer

DaStivi, 2021-03-09:
Even if STL is sliced with verbose gcode enabled in Superslicer octolapse does not find the gcode

#### Version of Octolapse
Octolapse Version:  v0.4.1

#### When you ran into the problem, did you have diagnostic logging enabled?
Diagnostic Logging was Enabled: yes (log everything)?

#### What were you doing when the problem occurred
1. slice file, upload it
2. start print job -> seeing the gcode processing window
3. Snapshot Plan Preview window opens, with the "no print features detected" warning

#### Operating System running OctoPrint and Octolapse
OS Name: OctoPi 0.18.0
OctoPrint Version: 1.5.3

#### Link to the gcode file you were printing when the problem occurred
Link to Gcode File: https://1drv.ms/u/s!AiIZqPOtqD4XqNJrbAhWLJxteygzlA?e=gu5BEx

#### Link to settings.json
https://1drv.ms/u/s!AiIZqPOtqD4XqNJsJnPvTS_V6Klfdg?e=S3JjUX

#### Link to plugin_octolapse.log
Link to plugin_octolapse.log:  https://1drv.ms/u/s!AiIZqPOtqD4XqNJuwG7BLhPRx1KtjQ?e=hXhEKs

#### Link to octoprint.log
Link to octoprint.log:  https://1drv.ms/u/s!AiIZqPOtqD4XqNJvTV5ArOWmnJE0MA?e=rs3Ou6

#### Screenshots and/or videos of the problem:
[image]
[image]
[image]

FormerLurker, 2021-03-10:
The current release version does not support features from SuperSlicer yet.  I have added some basic support in the latest development branch though.  I will figure out which branch later, as soon as I have some time to dedicate to issues.  In the middle of a new printer build, and it's kind of urgent because my old one is starting to break down and will need to be rebuilt, but i need it to finish the new one.  Talk about chicken or egg, lol!

FormerLurker, 2021-03-10:
They identify print features differently :)  Someone changed the wording and didn't think it would impact anything, but it means that Octolapse can't utilize feature detection until it is updated.  I just got a list of all of the print features sent today in fact, and they are subtly different, but enough to break detection.

DaStivi, 2021-03-11:
ok... that explains it 🤦‍♂️ thats also what you get with opensource i guess 🙈🙊  so at least thank for the clearification and i just gonna wait for a release to fix this... if you need someone to test, just tell me... really use the superslicer instead of cura now...

SturmB, 2021-11-18:
+1 to this. I hope the 'print quality reduction' isn't much, but until this is implemented, I'll probably hold off on using Octolapse for a while.

Hustenbonbon, 2021-12-16:
Hey all,
would this be resolved by https://github.com/FormerLurker/Octolapse/pull/782 ? If yes, I would be willing to test the PR if I find out how to use custom builds of plugins on Octoprint

FormerLurker, 2021-12-16:
Guys, I got several other issues reported about this, and I forgot to respond to this one.  Newer versions of superslicer output gcode feature comment blocks, which do NOT require verbose mode, are faster, and provide more information.

That pull request was missing some things, so I went ahead and implemented it myself.  If you want to try it out, install from this link via the plugin manager:

```https://github.com/FormerLurker/Octolapse/archive/7d9479091232ed88e8789754cce6ce9f5a03a398.zip```

There are a few other fixes in there:  % signs in the gcode file now no longer cause rendering to fail (introduced by some recent changes to Octoprint.  This used to be stripped out).  I also fixed smart triggers operating on files with unicode characters in windows (I hope).

I'd appreciate any feedback.

Hustenbonbon, 2021-12-19:
Tested with Superslicer without verbose mode, 4 different prints between 1 hour and 6 hours, works like a charm

KantinHoll, 2022-06-29:
Hi,

I've tested your zip with SuperSlicer without verbose and it's working very well !! Awesome !

lettore, 2022-09-11:
Any chance that this issue will be resolved?