FormerLurker/Octolapse #704: Octolapse Startup Failed Error Starting Timelapse   An unexpected error occurred while starting the timelapse. See plugin_octolapse.log for details.

BinkyBooBar, 2021-03-13:
Octolapse Startup Failed
Error Starting Timelapse

An unexpected error occurred while starting the timelapse. See plugin_octolapse.log for details.

!--
READ THE FOLLOWING FIRST:

I like to keep things clean.  This is a place for reporting bugs and
requesting new features, but is not a good place to ask questions about
or to learn how to use OctoPrint, Octolapse, or any third party programs
and tools like OctoPi, mjpg-streamer, or GPhoto2. Unfortunately there is
currently no user form dedicated to Octolapse, but you can find lots of
information directly within Octolapse (click on the blue question marks or
the help buttons within the interface), and in the Octolapse Wiki
(https://github.com/FormerLurker/Octolapse/wiki).  You should also
consider submitting an issue in Octoprint Community Forms within the
plugin category (https://community.octoprint.org/c/plugins/), or via the
official OctoPrint Discord server within #support-plugins channel.
Basically, please exhaust every other option before positing an issue so I
can focus as much as possible on improving Octolapse, and fixing known
bugs.  However, if you get stuck and can't find help, and you are sure your
problem isn't actually a bug, please send an email to: FormerLurker@pm.me.

For feature requests, add the prefix "[Request]" in the title (without the
quotes).

Do not delete anything from the template except where noted or the
bot will complain.

Thanks!
-->

#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: ___REPLACE_THIS__OCTOLAPSE_VERSION_GOES_HERE

#### Version of OctoPrint

OctoPrint Version: ___REPLACE_THIS__OCTOPRINT_VERSION_GOES_HERE

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: ___REPLACE_THIS__YES_OR_NO

#### What were you doing when the problem occurred

1. ___REPLACE_THIS__STEP_ONE_GOES_HERE
2. ___REPLACE_THIS__STEP_TWO_GOES_HERE
3. ___REPLACE_THIS__STEP_...

#### What should have happened?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: ___REPLACE_THIS__OS_NAME_GOES_HERE
Os Version: ___REPLACE_THIS__OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

closed 2021-03-13 by BinkyBooBar

FormerLurker, 2021-03-15:
Hey, I hope you got your issue solved.  Sorry I didn't get to this until now, but if you did solve your problem, thanks for closing the issue!

BinkyBooBar, 2021-03-15:
Hi FormerLurker,

Thank you for getting in touch.  My issues are not resolved however I
couldn’t cope with filling in all those fields at the time I was after help
so backed out of it 🤣

I just can’t get Octolapse to work with my Raspberry Pi 4 and FlashForge
Finder and Logitech webcam…any tips would be greatly appreciated.

Many thanks,

Hannah

FormerLurker, 2021-03-15:
I don't think a profile exists for your printer.  However, [read through this guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Getting-Started), which will walk you through the rest of the steps.  Be sure to follow the link for [Creating and Configuring a Custom Printer Profile](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Creating-And-Configuring-Your-Printer-Profile).  If you have questions about the custom profile, please let me know and be as specific as you can, and I'll help you where possible.  Please include as much info about your printer as you can (technical specs, bed size, firmware version and a URL to the source if it is available) with your requests to speed things along.